Koenkk/zigbee2mqtt #3529: _multiple_ Aqara Opple / WXCJKG13LM switches?

prankousky, 2020-05-11:
Hi,

I have been using a [WXCJKG13LM](https://www.zigbee2mqtt.io/devices/WXCJKG13LM.html) for a couple of months now. It works great, so I bought two more.

They were delivered, and I tried pairing them. They will pair, but that's it. Neither will they report any button presses, nor status information (link quality, battery, etc.) - all empty / unavailable (or "unknown" in `Home Assistant`).

zigbee2mqtt info:

```
{"version":"1.13.0","commit":"daecf45","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"info","permit_join":true}
```

I am using the [CC1352p-2](https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc1352p-2) with no additional dedicated routers (but own `Hue` and`innr` lights acting as routers). Just mentioning this for your information, I assume this doesn't matter as pairing is taking place ~2 meters from the coordinator in my office.

`zigbee2mqtt` ought to be a sufficient version to work with these devices, as I have successfully been using one for long enough to verify that it works ;) Might there be something wrong with these switches, for example, do they block one another when adding more than one?

This is the output I get when trying to (**re-**)pair one of them, same with the other one.

```
zigbee2mqtt/bridge/log {"type":"device_announced","message":"announce","meta":{"friendly_name":"Button/Multiswitch_zwei"}}
```

It is displaying a friendly_name, because I had paired this device before and changed the configuration file accordingly. When I remove the switch from my config, then restart, and try pairing again, I'll get

```
zigbee2mqtt/bridge/log {"type":"device_announced","message":"announce","meta":{"friendly_name":"0x04cf8cdf3c77b45f"}}
zigbee2mqtt/bridge/log {"type":"device_announced","message":"announce","meta":{"friendly_name":"0x04cf8cdf3c77b45f"}}
zigbee2mqtt/bridge/log {"type":"device_announced","message":"announce","meta":{"friendly_name":"0x04cf8cdf3c77b45f"}}
```

Three times, because I tried different methods:

* hold **(C)** button on the back for 5 seconds (led will blink)
* press **(C)** button 5 times (no visual indication)
* hold button for 5 seconds, then push button once every couple of seconds (trying  keep the device awake while pairing)
* hold button for 5 seconds, then push button(s) on the front every now and then (trying to keep awake as well)
* hold button for 5 seconds, then pushing button 5 times a few seconds later (trying to keep alive)

Of course now that I am trying to recreate yesterdays initial pairing process, this will not happen, but I also got two other outputs (**similar** to this, just from memory)

1. "interview completed" and
2. "pairing failed"

Neither of these can be recreated now; I tried deleting the devices from my config, restarting, pairing again, etc.; but all they will do is `announce`, then nothing else (waiting multiple minutes in between, which is way more than it took when pairing the first switch)

Are any of you using multiple of these buttons, and can help me pair the new ones I got? I have been searching for a solution like this and now that I got them already have use cases (just been waiting for the long delivery to finally put them to action).

Thank you for your help :)

IVI053, 2020-05-11:
I have three of these currently in use and can not confirm any issue like this. I'm running 1.12.2 and a CC26X2R1 `Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200312}}'`

Pressing the **[C]** button for 5 seconds is the correct way.

prankousky, 2020-05-11:
Thank you. What channel are you running on? I had to switch to `15` because of the [Konke Motion Sensor](https://www.zigbee2mqtt.io/devices/2AJZ4KPBS.html). This is the last thing I can think off why those switches wouldn't work - but then again, **one does work**, so these two new ones would have to have different firmware or similar.

Also, are you on the `dev` branch? I often run the `update.sh` in `/opt/zigbee2mqtt/`, but your revision is much newer than mine - while your version is older, which confuses me.

IVI053, 2020-05-11:
I'm on Ch 25.
You mix up coordinator firmware and z2m. `update.sh` updates z2m not coordinator. Regarding coordinator you are correct, I'm on dev currently. I did not update to 1.13.0 yet because it has no changes regarding my environment and simply had no time to update yet.

prankousky, 2020-07-11:
I totally forgot to reply, sorry!! I ended up flashing the newer coordinator firmware, and now the switches are connected. However, there is a different issue: while I have not yet programmed the switches to do anything at all, they both automatically controll **all lights** controlled via zigbee2mqtt.

One button will turn off all lights in the house, others will increase/decrease brightness for all lights! This is kinda weird, as the switches should currently not be able to control anything at all. But they both do this. Is this a bug? Or am I missing something in my config? I was hoping to map each button to individual actions.

closed 2020-08-12 by prankousky

sti0, 2020-10-14:
> One button will turn off all lights in the house, others will increase/decrease brightness for all lights! This is kinda weird, as the switches should currently not be able to control anything at [...]

@prankousky i have the same issue with one of two Opple switches. How did you solve this issue?

prankousky, 2020-10-14:
> How did you solve this issue?

Once I flashed the newest firmware onto my controller and re-paired the buttons, they worked properly. Good luck!

sti0, 2020-10-14:
Ok, thanks will give this a try.