Koenkk/zigbee2mqtt #3535: Documentation missing Ikea Sensor

MORD0R, 2020-05-11:
https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/E1525_E1745.md

I do not have Home Assistant. These 2 are missing.
requested_brightness_level
requested_brightness_percent

[image]

Koenkk, 2020-05-12:
Added, thanks for reporting.

Documentation will be automatically updated on next zigbee2mqtt release. Assuming this can be closed.

closed 2020-05-12 by Koenkk

MarcR2000, 2020-06-09:
Hi, just added the motion sensor to my setup and the payload only includes occupancy and linkquality. Not even battery status, no update info and no brightness levels. Are you sure the documentation is correct or is there any issue in zigbee2mqtt? The Hardware is the newer model of the motion sensor. Thanks for any advice!

MORD0R, 2020-06-09:
MarcR200 what if you use mqtt explorer? Are you sure you have the new version?

MarcR2000, 2020-06-09:
Just updated to the newest version. Still the same in MQTT.fx

MarcR2000, 2020-06-10:
I have now also tested another Tradfri Motion sensor (latest model) and it's the same. Newest software at all ends. Apparently brightness is not working for the newest model.

MORD0R, 2020-06-10:
I have an E1745
mqtt and got this. (no update available)
[image]

What if you press the buttons on the back a few times?

MarcR2000, 2020-06-10:
Hi Mordor,

still this is all I get as a payload:

"occupancy":true,"linkquality":84,"battery":87,"update_available":false,"device":{"friendlyName":"Tradfri_Motion_Treppe","model":"E1525/E1745",....

MORD0R, 2020-06-10:
Thats very strange i have no battery... I only have one tradifi sensor to test. 

I have attribute output on.

MORD0R, 2020-06-11:
its related too https://github.com/Koenkk/zigbee-herdsman-converters/issues/519