Koenkk/zigbee2mqtt #3539: No events from Ikea Tradfri E1810

przeqpiciel, 2020-05-12:
# Bug Report
## What happened
Ikea Tradfri E1810 has connect to zigbee2mqtt but no sends events while I pressing buttons

## What did you expect to happen
I expect to see json logs what I pressed

## How to reproduce it (minimal and precise)
1. Pair Ikea Tradfri E1810 to zigbee
2. press any buttons on controller

## Debug Info
Zigbee2mqtt version:  Starting zigbee2mqtt version 1.13.0 (commit #unknown)
Adapter hardware: CC2531
Adapter firmware version: have no idea how to check it

Koenkk, 2020-05-12:
This device requires a recent firmware version, otherwise it won't work (firmware version is shown in the startup log). https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

przeqpiciel, 2020-05-12:
Thank you for your respond. I restarted zigbee2mqtt and I have something like this
May 12 18:39:14 prox-01 npm[11315]: zigbee2mqtt:info  2020-05-12 18:39:14: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20180815}}'
May 12 18:39:14 prox-01 npm[11315]: zigbee2mqtt:debug 2020-05-12 18:39:14: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}

There is firmware version ?

przeqpiciel, 2020-05-13:
i uploaded new firmware via RPi and I get that
`zigbee2mqtt:info  2020-05-13 17:10:42: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-05-13.17-10-42' filename: log.txt
zigbee2mqtt:info  2020-05-13 17:10:43: Starting zigbee2mqtt version 1.13.0 (commit #55e6283)
zigbee2mqtt:info  2020-05-13 17:10:43: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-13 17:10:51: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-13 17:10:51: Failed to start zigbee
zigbee2mqtt:error 2020-05-13 17:10:51: Exiting...
zigbee2mqtt:error 2020-05-13 17:10:51: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-05-13T16_10_56_744Z-debug.log
`

Koenkk, 2020-05-14:
Please check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

closed 2020-07-20 by stale[bot]