Koenkk/zigbee2mqtt #3540: Zigbee2mqtt wont start.

aroe0546, 2020-05-12:
I have followed the manual closely via https://www.zigbee2mqtt.io/
but when i start npm i get the following error

```
pi@raspberrypi:~ $ npm start
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/pi/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-05-12T11_10_37_705Z-debug.log
```
i am using a raspberry pi 3 with buster lite and an zigbee2mqtt stick

Koenkk, 2020-05-12:
You need to execute this command in the directory with Zigbee2mqtt is installed. So probably `cd /opt/zigbee2mqtt && npm start`.

aroe0546, 2020-05-12:
also tried gives the same error

Koenkk, 2020-05-13:
What is the output of `ls /opt/zigbee2mqtt`?

aroe0546, 2020-05-13:
> What is the output of `ls /opt/zigbee2mqtt`?

‘’’
pi@raspberrypi:~ $ ls /opt/zigbee2mqtt
azure-pipelines.yaml  node_modules
data                  npm-shrinkwrap.json
docker                package.json
images                README.md
index.js              scripts
lib                   test
LICENSE               update.sh

‘’’

Koenkk, 2020-05-14:
Looks good, to be sure can you post the output when running with: `cd /opt/zigbee2mqtt && npm start`

aroe0546, 2020-05-15:
```
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt && npm start

> zigbee2mqtt@1.13.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-05-15 09:12:23: Logging to console and directory: '/opt/z                                                                                                                     igbee2mqtt/data/log/2020-05-15.09-12-23' filename: log.txt
zigbee2mqtt:info  2020-05-15 09:12:24: Starting zigbee2mqtt version 1.13.0 (comm                                                                                                                     it #55e6283)
zigbee2mqtt:info  2020-05-15 09:12:24: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-15 09:12:24: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-15 09:12:24: Failed to start zigbee
zigbee2mqtt:error 2020-05-15 09:12:24: Exiting...
zigbee2mqtt:error 2020-05-15 09:12:24: Error: Error while opening serialport 'Er                                                                                                                     ror: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapt                                                                                                                     er/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/zn                                                                                                                     p.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-s                                                                                                                     tack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dis                                                                                                                     t/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_mod                                                                                                                     ules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/st                                                                                                                     ream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional log                                                                                                                     ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-05-15T07_12_24_541Z-debug.log

```

Koenkk, 2020-05-15:
There is a different error: `Error while opening serialport`.

Make sure to closely follow this part of the guide: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#1-determine-location-of-cc2531-usb-sniffer-and-checking-user-permissions

aroe0546, 2020-05-15:
I have followed the instructions step by step several times.  fresh installation done this morning at the first startup is going well.  After a reboot it no longer works 

I also flashed the stick again but it doesn't matter

Sorry for my English, I’m Dutch

Koenkk, 2020-05-17:
After the reboot, what is the output of `ls -l /dev/serial/by-id` ?

aroe0546, 2020-05-17:
pi@raspberrypi:~ $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 May 17 15:04 usb-FTDI_FT232R_USB_UART_AO3ZSW72-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root 13 May 17 15:04 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E27B05-if00 -> ../../ttyACM0

I was thinking.  how do i check the status of zigbee2mqtt after the reboot.  I always do npm start myself but you can't start something that is running then you will definitely get an error

Koenkk, 2020-05-17:
This command should show the logging `sudo journalctl -u zigbee2mqtt.service -f`.

njordan77, 2020-05-20:
hi,
i'm not sure if this is realated....used zigbee@mqtt for a while now.....running the latest official version, without a warning...suddenly your service seems to have stopped working.....no more mqtt anymore....
Restart of the service does only bring error messages....reboot of the Raspi....works again....after a while same problem.....

The strange thing, this Raspi has not been touched for weeks now.....

Could you please check! Thanks a lot Norbert

```
zigbee2mqtt:info  2020-05-20 15:10:57: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-05-20.15-10-57' filename: log.txt
zigbee2mqtt:info  2020-05-20 15:10:58: Starting zigbee2mqtt version 1.13.0 (commit #55e6283)
zigbee2mqtt:info  2020-05-20 15:10:58: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-20 15:10:58: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-20 15:10:58: Failed to start zigbee
zigbee2mqtt:error 2020-05-20 15:10:58: Exiting...
zigbee2mqtt:error 2020-05-20 15:10:58: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.serialPort.open (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at binding.open.then.err (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-05-20T13_10_58_715Z-debug.log

```

Koenkk, 2020-05-21:
Your adapter is being used by another application: `Error: Error Resource temporarily unavailable Cannot lock port'`. You should be able to find out which with `ls -l /proc/[0-9]/fd/ | grep [PATH_TO_PORT] `

closed 2020-07-26 by stale[bot]