Koenkk/zigbee2mqtt #3541: [feature request] Possibility to remove auto-created sensors

to4ko, 2020-05-12:
# Feature request
## What happened
It' will be good to have possibility to switch off additional entitys auto-created in Home Assistant - for Xiaomi Plug there is 3  entities created -  switch, power and link quality. But last 2 of them are available as attributes in switch. It leads to additional DB recordings etc.
Yes they can be excluded from being recorded. But this is easy when 2 plug...and not so easy when 130+ devices .... simple option in config might solve this one very easily.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13
Adapter hardware: CC2538

to4ko, 2020-05-12:
as of now kinda solved by disabling un-needed entities from Configuration-Entities menu

Koenkk, 2020-05-13:
You can skip the discovery of a sensor with the device specific options. E.g. to prevent a temperature sensor from being discovered:

```js
devices:
  '0x00158d0001d82999':
    friendly_name: 'my_occupancy_sensor'
    retain: true
    homeassistant:
      temperature: null
```

to4ko, 2020-05-13:
> You can skip the discovery of a sensor with the device specific options. E.g. to prevent a temperature sensor from being discovered:
> 
> ```js
> devices:
>   '0x00158d0001d82999':
>     friendly_name: 'my_occupancy_sensor'
>     retain: true
>     homeassistant:
>       temperature: null
> ```

Thanks!!! need to do some changes in my config :)

to4ko, 2020-05-13:
> You can skip the discovery of a sensor with the device specific options. E.g. to prevent a temperature sensor from being discovered:
> 
> ```js
> devices:
>   '0x00158d0001d82999':
>     friendly_name: 'my_occupancy_sensor'
>     retain: true
>     homeassistant:
>       temperature: null
> ```

is it possible to make this kind of settings for all the devices simultaneously? i have like 132 devices at the moment and it would be good to have possibility to list unwanted type of sensors.

Koenkk, 2020-05-14:
In the latest dev branch this is now possibly by adding the following to your `configuration.yaml`.

```yaml
device_options:
  homeassistant:
    temperature: null
```

https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

closed 2020-05-20 by to4ko

to4ko, 2020-06-05:
@Koenkk am i doing something wrong?
here is devices.yaml
[image]

and i'm still having battery and linkquality in HA
[image]

Koenkk, 2020-06-06:
After this you probably need to clear the retained MQTT discovery message.

vitoralvimb, 2020-11-09:
How can I do that with the "cover"?
I have my Curtain completely closed, position = 100, and the Home Assistant have the Up Arrow disabled and Down Arrow avaliable. If I open the curtain putting position = 0, the Down arrow stay disabled.

I'd like to invert it.