Koenkk/zigbee2mqtt #3542: Errors with ConBee II and Centralite 3320-L

poblabs, 2020-05-12:
# Bug Report
## What happened

Starting the ConBee II stick and pairing Centralite Iris 3320-L contact sensors worked, however the logs show many errors trying to get state info, temperature info and other info. 

## What did you expect to happen

Contact state, battery and temperature info to be reported

## How to reproduce it (minimal and precise)

Just let it run, it errors on polling interval.

```
2020-05-12T14:24:02.016000251Z     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)

2020-05-12T14:24:02.016012872Z     at Generator.next (<anonymous>)

2020-05-12T14:24:02.016023015Z     at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))

2020-05-12T14:24:17.440308208Z zigbee2mqtt:info  2020-05-12 10:24:17: Configuring 'slider'

2020-05-12T14:25:18.028945812Z zigbee2mqtt:error 2020-05-12 10:25:18: Failed to configure 'slider', attempt 2 (Error: Bind 0x000d6f00057bdbec/1 msTemperatureMeasurement from '0x00212effff0596ff/1' failed (undefined)

2020-05-12T14:25:18.028973765Z     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)

2020-05-12T14:25:18.028979710Z     at Generator.throw (<anonymous>)

2020-05-12T14:25:18.028984439Z     at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

## Debug Info
Zigbee2mqtt version: 1.13.0 (commit #55e6283)
Adapter hardware: ConBee II
Adapter firmware version:

Koenkk, 2020-05-12:
Make sure that the device is awake when pairing the device (by pressing a button on it). If that doesn't solve it please post your logging here: https://github.com/Koenkk/zigbee-herdsman/issues/72

poblabs, 2020-05-13:
Going to close this as I am not using the sensor any longer. It's been nothing but trouble for me so time to replace it :smile:

closed 2020-05-13 by poblabs