Koenkk/zigbee2mqtt #3544: Innr RB 285 C not transitioning

Morphy99, 2020-05-12:
When sending transition with light.turn_on service in HA [this](https://www.zigbee2mqtt.io/devices/RB_285_C.html#innr-rb-285-c) bulb doesn't transition, it just switches straight away. I have several Hue and Tradfri bulbs and this works ok with those.
I'm sending this within the service dev tool:
```
entity_id: light.0x90fd9ffffe14562329_light
brightness: 1
transition: 5
```
I'm confused as to how HA sends the transition as it doesn't appear in the MQTT message:
```
{
    "state": "ON",
    "brightness": 255,
    "update_available": false
}
```
So maybe this is a HA bug, not sure.

## Debug Info
Zigbee2mqtt version: 1.13.0 (commit #55e6283)
Adapter hardware: CC2530
Adapter firmware version: 20190608

Koenkk, 2020-05-13:
As the transition is not included in the MQTT message, it indeed doesn't seem to be a zigbee2mqtt issue.

Morphy99, 2020-05-13:
That's what I thought but it doesn't explain how it works with both Hue and
Tradfri bulbs, how does HA send the transition to it? The example above was
to a Tradfri bulb. Sorry if I'm missing something obvious here...

Koenkk, 2020-05-14:
Did you define a `transition` in your zigbee2mqtt `configuration.yaml`?

Morphy99, 2020-05-14:
No, I tried this yesterday briefly and my entities disappeared from HA so think I did something wrong. Isn't that for the transition that's applied every time the light is activated though? I'm trying to achieve a specific transition time for an automation in HA.

Koenkk, 2020-05-14:
You need to add `transition` to the data of the service call. (Look at JSON Schema here https://www.home-assistant.io/integrations/light.mqtt/).

sjorge, 2020-05-14:
My Innr test bulb has ‘no’ transition when using say an IKEA remote with them either. So no coordinator involved. But my Trådfri and Hue bulbs do.

If I specify a transition in a mqtt payload all 3 do. So I think it’s the same for HA.
As it is not providing one the default from the bulb is used. For me Hue’s is slightly longer than Trådfris and my Innr has no transition at all.

Morphy99, 2020-05-16:
> You need to add `transition` to the data of the service call. (Look at JSON Schema here https://www.home-assistant.io/integrations/light.mqtt/).

I've seen this page but it's all a bit over my head. I've just enabled MQTT discovery within configuration.yaml and all my devices were added to HA. This page talks about manually defining MQTT entities which I haven't done. Do I need to add the Innr bulbs manually for transition to work?

I'm still confused how HA is sending transition to the hue and Tradfri bulbs and not the innr. 

> My Innr test bulb has ‘no’ transition when using say an IKEA remote with them either. So no coordinator involved. But my Trådfri and Hue bulbs do.
>
> If I specify a transition in a mqtt payload all 3 [...]

I think you're referring to the default transition which is used for every state change. I'm talking about sending a variable transition time in a HA automation. This would override any default transition set. Obviously there is something different to the way Innr bulbs handle this but I'm just trying to understand how HA is sending this command if it's not in the MQTT payload.

sjorge, 2020-05-16:
> I think you're referring to the default transition which is used for every state change. I'm talking about sending a variable transition time in a HA automation. This would override any default [...]

Yes, but are you sure HA is actually sending the transition for Hue bulbs as wel? It might not be sending a transition in the mqtt payload for any bulbs. (Which is my guess)

Morphy99, 2020-05-16:
It's not sending it in the MQTT payload, that's my point. If I trigger an automation to dim the light over 5 seconds as per my OP it will dim the light over 5 seconds but the MQTT topic has no mention of transition:

```
Message 0 received on zigbee2mqtt/Light_Landing at 11:35 AM:
{
    "state": "ON",
    "brightness": 1,
    "update_available": false
}
QoS: 0 - Retain: false
```

Morphy99, 2020-05-24:
Ok so I was looking at the zigbee2mqtt container logs expecting to see the transition there. I downloaded MQTT explorer and can see HA sending the transition now. I've updated both HA and zigbee2mqtt to the latest stable and now my Innr seems to be transitioning OK. Sorry for wasting your time.

closed 2020-05-24 by Morphy99