Koenkk/zigbee2mqtt #3547: ZNCLDJ12LM(Xiaomi Curtain motor B1) battery level support

nukusinji, 2020-05-13:
Hi.
Xiaomi curtain motor b1 has battery, so it didn't need external power.
Therefore we need to know battery level.
I find solution, but i have some problem in coding.

following this link(https://github.com/uqpqpu/st-driver/blob/master/devicetypes/shinjjang/xiaomi-curtain-b1.src/xiaomi-curtain-b1.groovy#L109)
The link has packet information.

Battery level read packet(comand) (It must send command for knowing battery level.)
 => payload = {0x0021: {value, type: 0x39}};
Battery level read packet(Read)
 => cluster:CLUSTER_POWER(0x0001), battery level : msg.data['61440'] (The value located same as cover position and percentage.)

I hope my information helped you.

Koenkk, 2020-05-14:
Can you try changing the entry in `devices.js` to:

```js
{
    zigbeeModel: ['lumi.curtain.hagl04'],
    model: 'ZNCLDJ12LM',
    vendor: 'Xiaomi',
    description: 'Aqara B1 curtain motor ',
    supports: 'open, close, stop, position',
    fromZigbee: [fz.ZNCLDJ11LM_ZNCLDJ12LM_curtain_analog_output, fz.cover_position_tilt, fz.ignore_basic_report, fz.battery],
    toZigbee: [tz.ZNCLDJ11LM_ZNCLDJ12LM_control, tz.ZNCLDJ12LM_options],
    onEvent: async (type, data, device) => {
        // The position (genAnalogOutput.presentValue) reported via an attribute contains an invaid value
        // however when reading it will provide the correct value.
        if (data.type === 'attributeReport' && data.cluster === 'genAnalogOutput') {
            await device.endpoints[0].read('genAnalogOutput', ['presentValue']);
        }
    },
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.endpoints[0];
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
},
```

After this I expect the battery to be reported within 24 hours. 

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to get to `devices.js`.

nukusinji, 2020-05-15:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/1237#issue-418348125

Working Good~
Thanks~ 

I closed~

closed 2020-05-15 by nukusinji