Koenkk/zigbee2mqtt #3549: Log please zigbee traffic, not mqtt messages only

a-x-, 2020-05-14:
Yes, you have an instruction how to sniff trafic.
But it requires second device and more effort.

zigbee2mqtt obviously works with zigbee and pass zigbee traffic through.
So, seems it’s easy to add sniffing out of the box.

Add, please 🤞 🙏

mr2c12, 2020-05-14:
Just a note: that would be still incomplete. Sometimes you need to sniff the traffic far away from the coordinator.
Happened to me last week: One sensor was sending the proper packet to a router, but the router for some reason didn't forward that packet. You could miss that case!
In this case you still need a second cc2531 to sniff the traffic.

Moreover, I doubt that z2m could log every info you can get from wireshark.

My 2 cent!

carldebilly, 2020-05-14:
As @mr2c12 said, may it's not possible...  but even incomplete messages would be a awesome feature :-).

sjorge, 2020-05-14:
If you enable all debugging (not just in configuration.yaml) it logs all package data. Not sure much more can be logged.

https://www.zigbee2mqtt.io/information/debug.html

Koenkk, 2020-05-14:
Zigbee2mqtt cannot act as a sniffer because not all data is received by it.
- Not all zigbee commands are send to the coordinator
- Not all commands received by the coordinator are passed to zigbee2mqtt

As @sjorge mentioned, debug logging will show some of the traffic (messages that are actually received by zigbee2mqtt).

closed 2020-07-20 by stale[bot]