Koenkk/zigbee2mqtt #3554: Important stability issues with CC26X2R1

Edzilla2000, 2020-05-15:
# Bug Report
## What happened
I've been using zigbee2mqtt for a while now.
About 6 months ago I got a CC26X2R1 to replace my CC2531.
When it works, it is a lot more stable than the CC2531 used to be (all the lights in a group come up at the same time now, transitions work great...)
However, several times a week zigbee2mqtt randomly stops working completely.
When it happens, I usually have to completely unplug the adapter, restart the docker container multiple times, wait a while, restart all the lights while Z2M is configured as "permit_join: true", and even then often I have to wait hours for everything to come back up.
I have tried everything I could think of. 

## What did you expect to happen
Z2M should be stable

## How to reproduce it (minimal and precise)
Wait a while, it happens usually every 2 days, can be less.

## Debug Info
Adapter hardware: CC26X2R1
I'm currently on the latest dev firmware (CC26X2R1_20200417.zip) and on the latest-dev docker container, but the same thing happens with stable firmware and stable docker-container.

Here is my docker-compose.yml, I'm using docker swarm: 
https://pastebin.com/kMmK2W8M

Here is my configuration.yaml, with secret bits removed:
https://pastebin.com/917BXUgM

Edzilla2000, 2020-05-15:
If I try to send a command a bulb when the network is misbehaving, this is usually what I get: 
https://pastebin.com/DZb4k55c

Koenkk, 2020-05-15:
Could you sniff the traffic when the no mac ack error occurs? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

awulf, 2020-05-18:
I've been having stability issues since upgrading from 1.12.0 to 1.13.0. Now reverting back to 1.12.0 to see if that fixes it again and to rule out something else. Had been very stable with no issue before, now crashes once a day. Forgot to record what the logs said in the std out, the log.txt files seems to show less.

The log files just say `error 2020-05-18 19:14:58: Adapter disconnected, stopping`

Edzilla2000, 2020-05-26:
In the meantime I got a CC1352P-2 to try and see if the issue came from my coordinator, but I have the exact same symptoms.

Here is the sniff trace: https://send.firefox.com/download/869d12bf8c2acd48/#HPnuPBBpIBWEV0IQie49QA
I hope I sniffed it correctly.

Koenkk, 2020-05-26:
Could you also shareyour network_key, otherwise I cannot decrypt the sniff. You can share it privately be sending it on telegram (@koenkk)

Edzilla2000, 2020-05-26:
I have sent you the network key via telegram, (@Edzilla)

Edzilla2000, 2020-05-29:
Following your advice, I have changed my wifi and Zigbee channel and repaired everything.
So far everything is working as it should.

Edzilla2000, 2020-06-04:
It would seem that it was indeed interferences, as everything has been stable for a week, where previously I had issues every other day.
Thank you so much for your help!

closed 2020-06-04 by Edzilla2000