Koenkk/zigbee2mqtt #3555: HEIMAN HS2WD-E won't pair

mxf14, 2020-05-16:
# Bug Report
## What happened
HEIMAN HS2WD-E does not pair -- Actually nothing happens visually so I did let run logging on herdsman and saw that as long as the siren was in pairing mode (green led flashing), there was some activity every 9 seconds involving the parser. Unfortunately I do not fully understand the log therefore I would like to get your help.

## What did you expect to happen
After pushing pair button I would have expected pairing as usual.

## How to reproduce it (minimal and precise)
Can be reproduced by simply activate pairing on the siren.
Of course permit_join is active !
The network is not too crowded (8 devices) and otherwise works well. Other devices could be paired successfully (Ikea and Xiaomi stuff).

## Debug Info
Zigbee2mqtt version: 1.13.0 - log attached was done with v. 1.11.0 but an update showed the same behavior
Adapter hardware: CC2531
Adapter firmware version: 20190608
Log output: [https://pastebin.com/F2Jak5Qv](url)

Koenkk, 2020-05-17:
I don't see any pairing related stuff in that log. Did you try the steps from: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

mxf14, 2020-05-18:
I followed the list but it did not help. Then I replaced the CC2531 dongle by a new one and it did pair. No clue why this did change something. With the old one communication with the siren seems to work as well.

closed 2020-07-25 by stale[bot]

sguimbal, 2024-07-30:
Hi all,
I was also impacted by the fact that i wasn't able to pair anymore this device after a long time working.
Just unplug the device until the battery is fully empty. (3 or 4 days)
Plug your device on power => Enable Pairing on Z2M => Pin the device to include.
It worked like a charm for me ! 
Stephane

Schermbecker, 2024-12-28:
> Hi all, I was also impacted by the fact that i wasn't able to pair anymore this device after a long time working.

Same issue here. I will give it a try and wil let the device unplugged until battery is empty for some more days. I tried with one or two days now but this did not help. 

Changing the z2m channel did not help, this was my first idea.