Koenkk/zigbee2mqtt #3557: How best to do this - running two instances of zigbee2mqtt up against the same MQTT broker

lancelon, 2020-05-16:
I was having issues with the hass.io addon zigbee2mqtt which for some reason wasn't allowing me to add new devices. It just silently...did nothing. I tried everything to get it working.

So I bought another CC2531 and have set it up as another coordinator and straight away without much fuss (still found the documentation for installing zigbee2mqtt wasn't totally straightforward, I had to install npm separately for example).

Because I'm still running zigbee2mqtt under Hass, I have set the new instance (hosted on a separate Pi) under the topic zigbee1mqtt instead of zigbee2mqtt like the other one.

Is this the recommended method for doing this? Is there a better way?

Thanks

IVI053, 2020-05-16:
The issue tracker is no forum!

Koenkk, 2020-05-17:
In case of running multiple zigbee2mqtt instances, you should indeed use different base topics. So the way you did it is correct. Assuming this can be closed now.

closed 2020-05-17 by Koenkk

yanirel, 2023-10-24:
want to  do it , if u have an a little explain ow it will be great , thanks