Koenkk/zigbee2mqtt #3558: MAC transaction expired' (240))

lancelon, 2020-05-16:
Hi there

Getting this error when trying to communicate with an IKEA blind.

I have once (once!) managed to get the blind to respond to instructions from MQTT but otherwise I send the message and then a minute or so later the above message is sent. ANy ideas why? Have tried powercycling the blind but didn't help. Thanks
```
 {
    "type": "zigbee_publish_error",
    "message": "Publish 'set' 'state' to '0xd0cf5efffe7ab117' failed: 'Error: Command 0xd0cf5efffe7ab117/1 closuresWindowCovering.downClose({}, {\"timeout\":10000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC transaction expired' (240))'",
    "meta": {
        "friendly_name": "0xd0cf5efffe7ab117"
    }
}
```

Koenkk, 2020-05-17:
Did you pair the device through the tradfri signal repeater (I don't have this device myself but users report increased stability: https://www.zigbee2mqtt.io/devices/E1757.html#pairing)

lancelon, 2020-05-18:
Hi, no I didn't, so I will do that. Thanks for the advice.  Is the range on the blind itself really poor?

skipper79, 2020-06-11:
Same issue after updating Supervisor and HA:

After updating Home Assistant from 3.14 to 4.10
And supervisor to 227 I can command my IKEA blinds. I get info from al my devices incl the battery status of my blinds, but een error occurs when I give a command to close or open the blinds
zigbee2mqtt:error 2020-06-11 15:59:00: Publish 'set' 'position' to '0x000d6ffffe9a881f' failed: 'Error: Command 0x000d6ffffe9a881f/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":100}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'MAC transaction expired' (240))'

Koenkk, 2020-06-11:
@skipper79 make sure to pair the blinds to the signal repeater (https://www.zigbee2mqtt.io/devices/E1757.html#pairing)

skipper79, 2020-06-11:
Until the update it was not nessesary... the pi with usb dongle is placed in the window under the blinds.

lancelon, 2020-06-12:
Hi @Koenkk - so I've now managed to pair the blind itself, the remote clicker and the repeater. The remote and the blind appear in zigbee2mqtt with proper names and descriptions (like IKEA blind) but the repeater doesn't seem to show up with a  nice name....

But anyway my question is this. When I had the repeater, remote and blind all next to one another the blind responded to requests. Now that I've put the blind back UP and put the repeater fairly near the blind (same room but not right next to it) I can't get the blind to move up/down anymore. Weirdly however the blind's position is being correctly reported and received by zigbee2mqtt... so it's not as if it's totally out of range.
My question is do i need to DO anything to get the repeater to HELP, or is it enough that it's simply part of the zigbee network?

Thanks

```
{
    "type": "devices",
    "message": [
        {
            "ieeeAddr": "0x00124b0001536f96",
            "type": "Coordinator",
            "networkAddress": 0,
            "friendly_name": "Coordinator",
            "softwareBuildID": "zStack12",
            "dateCode": "20190608",
            "lastSeen": 1591971948249
        },
        {
            "ieeeAddr": "0xd0cf5efffeea28ef",
            "networkAddress": 24046,
            "vendor": "-",
            "description": "-",
            "friendly_name": "0xd0cf5efffeea28ef",
            "lastSeen": 1591957604663
        },
        {
            "ieeeAddr": "0x000d6ffffe9bf6e0",
            "type": "EndDevice",
            "networkAddress": 10345,
            "model": "E1766",
            "vendor": "IKEA",
            "description": "TRADFRI open/close remote",
            "friendly_name": "0x000d6ffffe9bf6e0",
            "manufacturerID": 4476,
            "manufacturerName": "IKEA of Sweden",
            "powerSource": "Battery",
            "modelID": "TRADFRI open/close remote",
            "hardwareVersion": 1,
            "softwareBuildID": "2.2.008",
            "dateCode": "20190410",
            "lastSeen": 1591961823167
        },
        {
            "ieeeAddr": "0xd0cf5efffe7ab117",
            "type": "EndDevice",
            "networkAddress": 723,
            "model": "E1757",
            "vendor": "IKEA",
            "description": "FYRTUR roller blind",
            "friendly_name": "0xd0cf5efffe7ab117",
            "manufacturerID": 4476,
            "manufacturerName": "IKEA of Sweden",
            "powerSource": "Battery",
            "modelID": "FYRTUR block-out roller blind",
            "hardwareVersion": 1,
            "softwareBuildID": "2.2.007",
            "dateCode": "20190311",
            "lastSeen": 1591962417767
        }
    ]
}
```

lancelon, 2020-06-12:
Actually I think I've answered my own question. I've just generated a graph and [it's obvious something is wrong with the repeater](https://ibb.co/2Y8SybM)

skipper79, 2020-06-16:
Thanks @Koenkk I fixed it quit simple by reconnect the blind to HA without reapeater.

closed 2020-07-26 by stale[bot]