Koenkk/zigbee2mqtt #3560: Livolo new switch and socket support

jblankfeld, 2020-05-17:
Hi, I have received new livolo switch and socket (https://fr.aliexpress.com/item/4000385782177.html and https://fr.aliexpress.com/item/4000343575116.html), I'm trying to use them with z2m.
They pair but socket is seen as switch and remote toggle through mqtt does not work.
I have attached a full zigbee2mqtt debug log for the socket that includes pairing and a few toggle through device.
Has anyone been able to toggle them with raw ZCL ? I would like to continue this approach: https://github.com/Koenkk/zigbee-herdsman/issues/127 and https://github.com/Koenkk/zigbee2mqtt/issues/592#issuecomment-607381279
[livolo.log](https://github.com/Koenkk/zigbee2mqtt/files/4640305/livolo.log)

AlexLucky1983, 2020-05-23:
https://github.com/Koenkk/zigbee-herdsman/issues/127 @kirovilya was very close. 
The problem was in "Source Endpoint: 01" it must be 08.

turnOff
0000 00 06 01 00 04 01 08 de 7c d2 1a e9 02 01 00 01
0010 00 00 00 00 00 00 00 00

turnOn
0000 00 06 01 00 04 01 08 de 7c d2 1a e9 02 01 00 01
0010 01 00 00 00 00 00 00 00

I don't know how to get this sequence of bytes by API that zigbee2mqtt provides. I see just one ability here - to add some method with writing to buffer without any dependencies between endpoints, commands, clusters.

After I hardcoded this, I found another issue: We cannot separate switches and sockets. Because modelID is the same. I manually changed modelID directly in database.db and it helped to use different converters in devices.js. So I think it's also would be a greate extension to add posibility to specify modelID in configuration file aditionally to automatic determination.

AlexLucky1983, 2020-05-25:
[z2m_livolo.zip](https://github.com/Koenkk/zigbee2mqtt/files/4678410/z2m_livolo.zip)

In attached zip are all my changes:
database.db as example. It should be changed manually after joining. Changed just the first modelID to "Livolo_socket"
[image]

**zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js** 
extended "sendZclFrameToEndpoint" method by new "srcEndpoint" parameter with default value = 1 (for livolo socket we have to provide 8)

**zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\definition\frameControl.d.ts**
added new field "reservedBits: number;" to FrameControl interface. For livolo we have to use 0x7C frame control instead of 0x1C.

**zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\zclFrame.js**
extended "create" method by new "reservedBits" parameter.

**zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js**
copied new method "writeLivolo". And replace parameters that I described above in ZclFrame.create and sendZclFrameToEndpoint methods.

That's it. Now we have to update converters:
**zigbee2mqtt\node_modules\zigbee-herdsman-converters\devices.js**
added new section 'Livolo_socket' (do you remember our changes in database.db) copied from existing 'TI0001          '. Methods changes to 
```
        fromZigbee: [fz.livolo_socket_state_raw],
        toZigbee: [tz.livolo_socket_on_off],
```
"livolo_socket_state" I don't use.

**zigbee2mqtt\node_modules\zigbee-herdsman-converters\converters\fromZigbee.js**
Added "livolo_socket_state_raw" method. Copy of existing "livolo_switch_state_raw" with changed index from 15 to 14.

**zigbee2mqtt\node_modules\zigbee-herdsman-converters\converters\toZigbee.js**
Added "livolo_socket_on_off", copied from "existing livolo_switch_on_off", but command and payload changed according to https://github.com/Koenkk/zigbee-herdsman/issues/127#issuecomment-575781059

Actually the most part of changes is global and done according to zigbee standart. Only endpoint.js is livolo specific. So maybe @Koenkk can extend existing "write" method to do this more configurable.

kirovilya, 2020-05-31:
Hi. Here update to_zigbee converter for new switch and socket (working with herdsman 0.12.91):
```
await entity.command('genOnOff', 'toggle', {}, {});
const payloadOn = {0x0001: {value: Buffer.from([1, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
const payloadOff = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
await entity.write('genPowerCfg', (state === 'on') ? payloadOn : payloadOff,
    {manufacturerCode: 0x1ad2, disableDefaultResponse: true, reservedBits: 3,
        direction: 1, transactionSequenceNumber: 0xe9}
);
return {state: {state: value.toUpperCase()}, readAfterWriteTime: 250};
```
Now need the ability to _write_ without waiting for a response from the device. But the _write_ is awaiting a response, but it is not.

kirovilya, 2020-06-02:
and another one: transactionSequenceNumber can be 0 for toggle command.
now if 0, then a new one is generated

Koenkk, 2020-06-02:
Done, call with e.g.

```js
await entity.write('genPowerCfg', (state === 'on') ? payloadOn : payloadOff,
    {manufacturerCode: 0x1ad2, disableDefaultResponse: true, reservedBits: 3,
        direction: 1, transactionSequenceNumber: 0xe9, disableResponse: true}
);
```

jblankfeld, 2020-06-04:
Awesome work guys.
I tested it and I could toggle the socket :+1:

andrew-yu, 2020-06-12:
zigbee2mqtt 1.14.0

`zigbee2mqtt:debug 2020-06-12 17:06:28: Received MQTT message on 'zigbee2mqtt/0x00124b001801657f/right/set' with data 'ON'
zigbee2mqtt:error 2020-06-12 17:06:28: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'right' is given but no endpoints defined
    at Zigbee.resolveEntity (/zigbee2mqtt-1.14.0/lib/zigbee.js:200:25)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.14.0/lib/extension/publish.js:59:44)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.0/lib/controller.js:352:44)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.14.0/lib/controller.js:246:14)
    at MQTT.emit (events.js:223:5)
    at MQTT.onMessage (/zigbee2mqtt-1.14.0/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:223:5)
    at MqttClient._handlePublish (/zigbee2mqtt-1.14.0/node_modules/mqtt/lib/client.js:1269:12)
    at MqttClient._handlePacket (/zigbee2mqtt-1.14.0/node_modules/mqtt/lib/client.js:409:12)
    at work (/zigbee2mqtt-1.14.0/node_modules/mqtt/lib/client.js:320:12))`

Please, remove "right".

MiguelAngelLV, 2020-06-29:
How pair?

I press the toggle button about 5 seconds and the led blink blue/right, but Z2M logs not show anything

Ton1965, 2020-07-29:
Unstale it. Livolo products worth tracking

closed 2020-09-05 by github-actions[bot]

jujoparo, 2020-10-25:
Please,  Could you clarify whether it is supported or not?

andrew-yu, 2020-10-25:
I'm using it for a few months. Works fine, but all of my Livolo devices isolated in own Zigbee network.

jujoparo, 2020-10-25:
Is it necessary to have its own network? Why did you do that?

andrew-yu, 2020-10-25:
Livolo has own original sight how Zigbee should work... :)

jujoparo, 2020-10-25:
Andrew-yu thank you for your help and excuse me for being so insistent, I don't want to buy it if it does not work, that's why I am asking you.

Please, could you explain me if livolo devices require a only livolo zigbee network to work or it is just a recommendation?

andrew-yu, 2020-10-25:
Sorry that i didn't say it clear, my English is not perfect:

For Livolo devices i use separate cc2531 (any other supperted by zigbee2mqtt zigbee stick can be used).

It's optional to create separate network for Livolo devices, but it's good idea because:
1. Livolo devices work only at 26 channel
2. permit_join must be enabled (network is open, but can filter by device id with 'passlist' directive)
3. One cc2531 can handle about up to 8 livolo devices because of flood of packet from the devices.

Jangarz, 2020-11-04:
Hey guys I am looking to fill my house with livolo's and was wondering if this 'new' switch is better than the old one in any way? Is that small opening above the center a status LED? Thanks in advance.

martinlehofer, 2020-12-15:
I have several TI0001-switch devices, however, they are recognized as TI0001 devices instead which offer two switches. Reading/writing status doesn't work then. Somehow with resetting/repairing I have managed to get one device recognized as TI0001-switch, but I haven't managed the same with the others.

Let me know which data/logs/traces are helpful to diagnose/fix the issue.

wxiux, 2021-01-06:
Hello,

Thank you for the info, channel 26 and got it working.

I also got this "new" switch, however, i also got a "two gang" version of it, as per photo, the left one:
[image]

the problem with the single switch that got recognized as TI0001 can eventually be solved by keep repairing/resetting until it recognizes as TI0001-switch.

but this two gang version, I tried to repairing it like 100 times, still is recognized as TI0001, seems not supported at all yet.

andrew-yu, 2021-01-08:
I had same issue with Livolo socket - it was recognized as TI0001 instead of TI0001-socket. Was solved by repairing.

andrew-yu, 2021-01-08:
If you are using Home Assistant add-on Zigbee2mqtt use `ext_pan_id_string` instead of `ext_pan_id`:
`ext_pan_id_string: '33, 117, 141, 25, 0, 75, 18, 0'`

three-m, 2021-01-11:
Also have problems with pairing. I tried with different sticks on version 1.17 - after holding button blue led flashes but no result and logs are empty (26 channel)

andrew-yu, 2021-01-11:
Please, show your config and switch model.

three-m, 2021-01-11:
[socket](https://www.aliexpress.com/item/1005001354323235.html?spm=2114.12010612.8148356.1.4ed84c74FEjT1g)
[image]

andrew-yu, 2021-01-11:
What about permit_join?

three-m, 2021-01-11:
always true

andrew-yu, 2021-01-11:
Config looks good. How about logs at pairing time and switch model (or photo)?

three-m, 2021-01-11:
[image]
[image]
there is nothing logs

three-m, 2021-01-11:
did changed log level to debug and after multiple pairing attempts get
`Zigbee2MQTT:debug 2021-01-11 12:53:26: Received Zigbee message from '0x00124b001803dd08', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,8,221,3,24,0,75,18,0,7,1,6,1,0,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Zigbee2MQTT:debug 2021-01-11 12:53:35: Received Zigbee message from '0x00124b001803dd08', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,8,221,3,24,0,75,18,0,7,1,6,1,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Zigbee2MQTT:debug 2021-01-11 12:53:52: Received Zigbee message from '0x00124b001803dd08', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,8,221,3,24,0,75,18,0,7,1,6,1,0,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Zigbee2MQTT:debug 2021-01-11 12:53:55: Received Zigbee message from '0x00124b001803dd08', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,8,221,3,24,0,75,18,0,7,1,6,1,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Zigbee2MQTT:debug 2021-01-11 12:54:18: Received Zigbee message from '0x00124b001803dd08', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,8,221,3,24,0,75,18,0,7,1,6,1,0,11],"type":"Buffer"}' from endpoint 6 with groupID 0`

senufo, 2021-02-07:
My Livolo switch (<https://www.amazon.fr/gp/product/B08CB31XLH/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1>) is recognized as TI0001 while it is a TI0001-SWITCH. It was recognized and paired by z2M but then I could not send the ON / OFF commands
By commenting in **devices.js** the model TI0001 and replacing `zigbeeModel: TI0001-SWITCH` by `zigbeeModel: ['TI0001      ']` it works correctly. Is it possible to force the template in the z2m configuration to use the correct template?

alleonname, 2021-03-08:
I pair the Livolo socket (Zigbee2MQTT version 1.18.1-dev), instead of the socket I have a switch Livolo
[image]
[image]
[image]

natoriousbigg, 2021-03-24:
Anybody gotten the TI0001-switch-2gang to switch ON/OFF correct? I was able to get the 1 gang switch to switch ON/OFF, but not the 2 gang switch. It's giving me the following error:

`Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'right' is given but no endpoints defined at Zigbee.resolveEntity (/app/lib/zigbee.js:229:25) at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:52:44) at Controller.callExtensionMethod (/app/lib/controller.js:380:44) at runMicrotasks (<anonymous>) at processTicksAndRejections (internal/process/task_queues.js:93:5))`

razim7, 2022-01-25:
what about the 3 gang switch and 4 gang livolo zigbee? should they work too?