Koenkk/zigbee2mqtt #3561: bticino K4003C slow response

Maurotb, 2020-05-17:
# Bug Report
## What happened

## What did you expect to happen
when I press the button of my k4003c, it takes from 2 to 4 seconds for zigee2mqtt to notice
allo other zigbee device (aquara door and presence sensor), notice is immediate.
 
## How to reproduce it (minimal and precise)
Press button
after 2/ seconds in log i see
info  2020-05-17 16:07:32: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f5d', payload '{"state":"OFF","linkquality":110}'

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531
Adapter firmware version:
CC2531_MAX_DEVICES_20190223

Maurotb, 2020-05-19:
Hi have sniffed myzigbee network with another cc2531 and wireshark.
When i push button on k4003c , it send immediatly a zigbee message but zigbee2mqtt show this message after 2 4 seconds

Koenkk, 2020-05-19:
This is probably related to a weak network or that the CC2531 is not powerful enough. How many devices do you have in your network?

Some things for now:
- You are using an older firmware, make sure to use the latest source routing firmware for best stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
- Make sure the adapter is connected through a USB extension cable
- Make sure that the network quality is good: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

Maurotb, 2020-05-20:
> This is probably related to a weak network or that the CC2531 is not powerful enough. How many devices do you have in your network?
>
> Some things for now:
>
> * You are using an older firmware, make [...]
I have try with  CC2531_DEFAULT_20190608,same result
> * Make sure the adapter is connected through a USB extension cable
yes
> * Make sure that the network quality is good: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

I have added to my cc2531 an external antenna and test is made closest to device (1mt).
I have other 12 aquara device (movement sensor,door sensor and cube-9 ,all work bettere and response is immediate....
In whireshark i see zigbee packet immidiatly but in zigbee2mqtt log is delayed...
In an italian forum i have see another user with same problem, 
but he has gone to deconz to fix it, I prefer to report to see if it can be resolved together

Maurotb, 2020-05-20:
@Koenkk i have make a debug,
when i push a buton i receive immediatly this message

debug 2020-05-20 22:21:56: Received Zigbee message from '0x00047400007d7f5d', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":0}' from endpoint 1 with groupID 0
debug 2020-05-20 22:21:56: No converter available for 'K4003C' with cluster 'genBinaryInput' and type 'attributeReport' and data '{"presentValue":0}'
debug 2020-05-20 22:21:56: Received Zigbee message from '0x00047400007d7f5d', type 'attributeReport', cluster 'genIdentify', data '{"identifyTime":60}' from endpoint 1 with groupID 0
info  2020-05-20 22:21:56: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f5d', payload '{"state":"OFF","linkquality":31,"action":"identify"}'
info  2020-05-20 22:21:56: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f5d', payload '{"state":"OFF","linkquality":31,"action":""}'
info  2020-05-20 22:21:56: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f5d/action', payload 'identify'

and after 2/4 sec this

debug 2020-05-20 22:22:00: Received Zigbee message from '0x00047400007d7f5d', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-05-20 22:22:00: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f5d', payload '{"state":"OFF","linkquality":31}'

or sometime i push button and i receive immediatly this

debug 2020-05-20 22:22:03: Received Zigbee message from '0x00047400007d7f5d', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
debug 2020-05-20 22:22:03: No converter available for 'K4003C' with cluster 'genBinaryInput' and type 'attributeReport' and data '{"presentValue":1}'

and after 2/4 sec this

debug 2020-05-20 22:22:08: Received Zigbee message from '0x00047400007d7f5d', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-05-20 22:22:08: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f5d', payload '{"state":"OFF","linkquality":23}'

Koenkk, 2020-05-21:
Added this in the latest dev branch, will provide (action: `on` and `off` now).

Maurotb, 2020-05-21:
> (action: `on` and `of`
@Koenkk
Good!
In bticino app i  have two settings for k4003c
1)set led on when low luminosity
2)set led on when relay is on
I can't decode with whireshark the zigbee message for this settings when k4003c is connected to bticino hub, i have
sniffed network key but trust center link key is not A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39
Can help me to get trust center key to complete the k4003c integration with zigbee2mqtt?

Koenkk, 2020-05-21:
You should be able to get a "transport key" message which is send while the device is being paired. This contains the network key.

Maurotb, 2020-05-21:
@Koenkk 
ok i find this with whireshark

Detectable switch in the dark
Command: Write Attribute (0x02)
Attribute 0x0001 
Data Type bool 

Show when the light is on
Command: Write Attribute (0x02)
Attribute 0x0002  
Data Type bool 

Identify switch
Command: Trigger effect (0x40)
Effect: 0x03
Variant:0

Is possible tto implement this?
Thanks

Koenkk, 2020-05-22:
Can you make a screenshot of these commands (with all possible fields expanded). I also need to know the cluster and value being send.

Maurotb, 2020-05-22:
Yes!

[image]
[image]
[image]

Koenkk, 2020-05-22:
I think this is already supported

- https://www.zigbee2mqtt.io/devices/067771.html#led-configuration
- https://www.zigbee2mqtt.io/devices/067771.html#identify

Maurotb, 2020-05-22:
good, i try this, thanks

Maurotb, 2020-05-23:
Good i try.
I have installed dev docker image to test your patch  but is same

debug 2020-05-23 09:47:33: Received Zigbee message from '0x00047400007d7f6d', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
debug 2020-05-23 09:47:33: No converter available for 'K4003C' with cluster 'genBinaryInput' and type 'attributeReport' and data '{"presentValue":1}'
after 2/s secs
debug 2020-05-23 09:47:36: Received Zigbee message from '0x00047400007d7f6d', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-05-23 09:47:36: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f6d', payload '{"linkquality":28,"state":"OFF","action":"on"}'
info  2020-05-23 09:47:36: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f6d', payload '{"linkquality":28,"state":"OFF","action":""}'
info  2020-05-23 09:47:36: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f6d/action', payload 'on'

Koenkk, 2020-05-23:
Made a small mistake in the converter. Fix will be available in the latest dev branch in a few hours.

Maurotb, 2020-05-24:
@Koenkki have downloaded 1.13.1-dev (commit #49df575) but same problem

debug 2020-05-24 17:40:10: Received Zigbee message from '0x00047400007d7f6d', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
debug 2020-05-24 17:40:10: No converter available for 'K4003C' with cluster 'genBinaryInput' and type 'attributeReport' and data '{"presentValue":1}'

after some seconds

debug 2020-05-24 17:40:15: Received Zigbee message from '0x00047400007d7f6d', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-05-24 17:40:15: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f6d', payload '{"linkquality":18,"state":"OFF","action":"on"}'
info  2020-05-24 17:40:15: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f6d', payload '{"linkquality":18,"state":"OFF","action":""}'
info  2020-05-24 17:40:15: MQTT publish: topic 'zigbee2mqtt/0x00047400007d7f6d/action', payload 'on'

Koenkk, 2020-05-25:
Are you sure your `devices.js` is not overridden by a manual one? (while following https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)

Maurotb, 2020-05-25:
@Koenkk yes,
i have try now to remove docker image and reinstall, same problem 
 No converter available for 'K4003C' with cluster 'genBinaryInput' and type 'attributeReport' and data '{"presentValue":1}'

Maurotb, 2020-05-25:
@Koenkk i have try the latest dev now and now work!
Now i install in real environment and test it to command my philips hue with this switch.

Only one questions, in HA i see published only action,link quality and switch

For this
>I think this is already supported (where permanent_led = low luminosity)
>https://www.zigbee2mqtt.io/devices/067771.html#led-configuration
>https://www.zigbee2mqtt.io/devices/067771.html#identify

I need to send mqtt command manually from HA?
Thanks

Koenkk, 2020-05-26:
Yes you need to send a MQTT command manually

Maurotb, 2020-05-26:
@Koenkk is not possible to add two entities
permanent_led 
led_when_on 
so we have same function of bticino hub? ;)

Ps i have tested in real environment and now work correctly, the delay is < 0.5 sec

Koenkk, 2020-05-27:
@Maurotb currently we don't discover device related options into home assistant, I think it will give a lot of extra entities which are almost never used. You can always create an manual entry yourself in case you often use this.

I will close this now as the issue has been solved. Thanks!

closed 2020-05-27 by Koenkk