Koenkk/zigbee2mqtt #3562: Philips 7146060PH - Button does not change state

duesseldorferjung, 2020-05-17:
When I am using the button on my Hue Go to turn it on or off, the state doesn't get updated in my Homeassistant. When I am using the Button on my Osram Smart+ plug for example State gets updated. I would expect the behaviour of the Hue Button to be the same.

Koenkk, 2020-05-17:
Do you use https://www.zigbee2mqtt.io/information/report.html ?

duesseldorferjung, 2020-05-23:
Again I am not able to keep up with your response speed. :) Any specific please where I need to add it to the configuration.yaml ? I am getting an invalid configuration Notification.

Koenkk, 2020-05-23:
It shown here: https://www.zigbee2mqtt.io/information/report.html.

Add the following to your `configuration.yaml`.

```yaml
advanced:
  report: true
```

duesseldorferjung, 2020-05-23:
> 
# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Text to speech
#tts:
#  - platform: google_translate

# state changes to zigbee2mqtt even when the change is not directly done by zigbee2mqt
advanced:
  report: true

Koenkk, 2020-05-23:
Can you share the `database.db` entry of this device?

duesseldorferjung, 2020-05-23:
Well maybe I was not able to explain myself. When I add the two lines to the configuration.yaml I get an error. Invalid configuration notification in HA stating that it was not able to setup the platform advanced:

Koenkk, 2020-05-23:
You have to add this in the zigbee2mqtt config, not the home assistant.

duesseldorferjung, 2020-05-23:
I also tried that it is telling me Invalid Yaml. Should not be that hard to enter two lines of code correctly :) What am I missing?
`advanced:`
  `   report: true`
I am entering also the spaces, somehow that is not showing now.

duesseldorferjung, 2020-05-23:
found it

duesseldorferjung, 2020-05-23:
There was already and advanced section in the config, so just needed to add the report: true part. For some reason still not working, I will check tomorrow again. Thanks for the quick support!

duesseldorferjung, 2020-05-27:
`{"id":4,"type":"Router","ieeeAddr":"0x00178801047920f9","nwkAddr":19313,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LLC020","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LLC020","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170317","swBuildId":"5.90.1.20202"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001936dc59","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170317","swBuildId":"5.90.1.20202","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590379597288}`

I hope this is what you looking for, or would the file itself be more helpful?

Koenkk, 2020-05-28:
It seems that reporting fails to setup for this device. Likely it contains a too recent firmware which does not support reporting anymore (https://www.zigbee2mqtt.io/information/report.html#support)

duesseldorferjung, 2020-06-07:
Well I guess not too much I can do about it. The main purpose I am trying to solve is get correct status after power loss (which in India happens quite often). Another option would be to set the power on behaviour to recovery but I guess for that my firmware is too old. It seems also the OTA function is available there is no update available for this specific devices, am I right?

Koenkk, 2020-06-07:
I don't think there is a ota update available (at least mine doesn't show). If you are able to detect the power loss, you can also send to `zigbee2mqtt/[FRIENDLY_NAME]/set` with payload `{"state": ""}` to retrieve the current state.

duesseldorferjung, 2020-06-07:
Good to know. Thanks for the help. I am actually experiencing the same issue with my Innr Bulbs, who are supposed to have a report function. Should I open a new issue for that?

Koenkk, 2020-06-08:
Can you share your databse.db entry of this device and do you see any logging of the reporting setup failed?

duesseldorferjung, 2020-06-11:
{"id":13,"type":"Router","ieeeAddr":"0x00158d00030001a0","nwkAddr":23036,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RB 285 C","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RB 285 C","manufacturerName":"innr","powerSource":1,"zclVersion":2,"appVersion":16,"stackVersion":21,"hwVersion":1,"swBuildId":"2.0"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":16,"stackVersion":21,"hwVersion":1,"swBuildId":"2.0","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1591516353750}

duesseldorferjung, 2020-06-11:
Where do I check the logs for the reporting setup?

Koenkk, 2020-06-11:
No reporting is setup indeed. You see it in the log some moments after starting zigbee2mqtt. You should see either `Failed to setup reporing for..` OR `Successfully setup reporting for`. But before checking this make sure that the indentation is correct in your `configuration.yaml`, there should be 2 spaces before `report`.

```yaml
advanced:
  report: true
```

duesseldorferjung, 2020-06-14:
Ok I check that and since I did a untwanted recovery on the weekend it was not activated. Now again it is. I also see a success in the log 
`zigbee2mqtt:info  2020-06-14 16:42:59: Successfully setup reporting for '0x00158d00030001a0' - 1 - genOnOff`
Then I double the check the following scenario. I turned the light off via Hass which is also the Status shown in Hass. Then I simulate a power cut. Physical switch off and on. The light is on but the status in Hass remains off. In my understanding of reporting that should change correct?

Koenkk, 2020-06-14:
@duesseldorferjung this is not what reporting does (https://www.zigbee2mqtt.io/information/report.html)

> When report is enabled, devices will report their state changes to zigbee2mqtt even when the change is not directly done by zigbee2mqtt. This is especially handy when the state of a bulb is changed by e.g. a remote or via groups.

To retrieve a state update when a device went offline you have to use availability: https://www.zigbee2mqtt.io/information/availability.html

duesseldorferjung, 2020-06-21:
That certainly does the "magic" I was looking for. What is the specific reason why this is not set to "on" by default?

Koenkk, 2020-06-24:
Because this will lead to much more messages on the network. This can make a network running on a less powerful adapter unstable (CC2531/CC2530). (https://www.zigbee2mqtt.io/information/availability.html#note-for-cc2530cc2531-users)

I assume this can be closed now.

closed 2020-06-24 by Koenkk