Koenkk/zigbee2mqtt #3563: Regression: `transition` parameter influences brightness

Otah, 2020-05-17:
# Bug Report
## What happened
Regression of a bug #2220

That bug was already fixed half a year ago, but with one of the latest versions it was reintroduced. Briefly:

> When a light is on and has e.g. `brightness: 50`, sending a command `{"state": "ON", "transition": 1}` sets the brightness to maximum (255). Whereas if you send just `{"state": "ON"}`, nothing happens (= the light stays on with brightness at level 50).

## How to reproduce it (minimal and precise)
Exactly the same way as in case of #2220

## Debug Info
Zigbee2mqtt version: 1.13.0 (commit #ac0d17c)
(I have upgraded recently, but not sure what was the previous version I have had deployed, where it was working)

Koenkk, 2020-05-18:
Fixed in latest dev branch. Added a test case this time to make sure it doesn't break another time :smile: Thanks for reporting!

closed 2020-05-18 by Koenkk

Otah, 2020-05-18:
Thanks a lot @Koenkk! That was pretty quick response, very appreciated! I have deployed the dev version and I confirm the fix is working well ;-)