Koenkk/zigbee2mqtt #3565: unresponsive s. 6 weeks - appearing on several devices

wojciechczyz, 2020-05-18:
# Bug Report
## What happened

I see 2 situations when this happens:
1. After restarting zigbee2mqtt, I see frequently all or some of the devices having
unresponsive s. 6 weeks or unresponsive s. 5 weeks
In case 1 after some time device are again connected and last seen time goes down to few seconds up to few minutes.
2. Sometimes out of the blue, after weeks of using zigbee2mqtt this pops up. 
when this happens, I cannot fix it by restarting, I sometimes have to repair some of those devices. strangely, after I fix some of them the rest works fine, even if they were not routers.

I see them on list with devices via zigbee2mqttassistant 0.3.157 

## What did you expect to happen

I would expect last seen time to be minutes or "never" or "not yet seen" but not 5-6 weeks ago.

## How to reproduce it (minimal and precise)

Sometimes I see this after restarting the entire HomeAssistant or only zigbee2mqtt.

## Debug Info
Zigbee2mqtt version:
1.13.0
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
CC2531
Adapter firmware version:
latest, with source routing

Koenkk, 2020-05-18:
This looks more like a zigbee2mqtt assistant issue then a zigbee2mqtt issue? Probably it fits better here: https://github.com/yllibed/Zigbee2MqttAssistant/issues

wojciechczyz, 2020-05-19:
Thanks,
I have filed it on zigbee2mqttassistant as well. Lets see how it goes:
https://github.com/yllibed/Zigbee2MqttAssistant/issues/280

Strange that part of my zigbee network went offline just like that all of the devices staying with same visibility 6 weeks ago, I was operating them just days ago.

nikdoof, 2020-05-25:
Without reporting switched on in Z2M, Z2MA shows all devices time as the date they joined the network IIRC (whatever the last_seen value is).

closed 2020-07-26 by stale[bot]