Koenkk/zigbee2mqtt #3566: Support for Aqara Smart Wall Switch (No Neutral, Double Rocker)

sickboy2137, 2020-05-18:
# Bug Report
## What happened
I'm trying to add support for the switch.  I've tried using 

`    {
        zigbeeModel: ['lumi.switch.b2laus01'], 
        model: 'WSDCGQ01LM',
        vendor: 'Aqara',
        description: 'Smart Wall Switch (No Neutral, Double Rocker)',
        supports: '',
        fromZigbee: [],
        toZigbee: [],
    },`
But the device ends up being seen as a WSDCGQ01LM
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-05-19:
Can you share your `database.db` and a URL (link) the the device?

sickboy2137, 2020-05-20:
Here is a link to the item. 
https://www.aqara.com/us/smart_switch_no_neutral.html

and here is the line from database.db regarding this item.

`{"id":24,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c75ed5f","nwkAddr":39660,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b2laus01","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.b2laus01","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":22,"stackVersion":2,"hwVersion":1,"dateCode":"07-23-2019"}}},"binds":[]},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":22,"stackVersion":2,"hwVersion":1,"dateCode":"07-23-2019","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1589921063877}`

Ive also updated the model number in my system to correctly show 
`        model: 'WS-USC02',
`
I also don't know if it will help, but someone has gotten this switch to work in smarthings.
https://community.smartthings.com/t/aqara-double-rocker-wall-switch-discussion-and-dth-updated-may-2-2020/192205

Koenkk, 2020-05-20:
With the following definition:

```js
{
    zigbeeModel: ['lumi.switch.b2laus01'],
    model: 'WS-USC02',
    vendor: 'Xiaomi',
    description: 'Aqara smart wall switch (no neutral, double rocker)',
    supports: 'on/off',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'top': 1, 'bottom': 2};
    },
},
```

You should be able to control it with `zigbee2qmtt/0x04cf8cdf3c75ed5f/top/set`/`zigbee2qmtt/0x04cf8cdf3c75ed5f/bottom/set` payload `{"state": "ON"}`

sickboy2137, 2020-05-20:
You are awesome sir. It's working! (albeit with correcting the typo). I've manually added in two switches into my HomeAssistant, and they appear to be working well. 

Unless there's more to do, I'd say this ticket can be closed.

sickboy2137, 2020-05-20:
I typed too soon. Controlling from mqtt both ways works. But button presses on physical device are not updated on mqtt.

Koenkk, 2020-05-21:
And with:

```js
{
    zigbeeModel: ['lumi.switch.b2laus01'],
    model: 'WS-USC02',
    vendor: 'Xiaomi',
    description: 'Aqara smart wall switch (no neutral, double rocker)',
    supports: 'on/off',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    meta: {multiEndpoint: true, configureKey: 1},
    endpoint: (device) => {
        return {'top': 1, 'bottom': 2};
    },
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(1);
        await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint1);
        const endpoint2 = device.getEndpoint(2);
        await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint2);
    },
},
```

sickboy2137, 2020-05-21:
Awesome. That last bit did it. It's all working now.

Koenkk, 2020-05-21:
Added, will be supported in the latest dev branch in a few hours. https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

closed 2020-05-21 by Koenkk

windjoechen, 2021-07-30:
Is it possible to add the option 「 Power outage memory」to     WS-USC01, WS-USC02, WS-USC03, WS-USC04 ?

visualage, 2023-01-02:
> Is it possible to add the option 「 Power outage memory」to WS-USC01, WS-USC02, WS-USC03, WS-USC04 ?

The following PR includes this for WS-USC02 and WS-USC04. I don't have WS-USC01 nor WS-USC03 to test, so I cannot include them in this PR. I suspect it would be the same for WS-USC01 and WS-USC03.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5242

eobando00, 2023-04-10:
Hello team, I have the USC03 and I really need the operation_mode (de-attach mode) .. I am happy to test but I need some help.