Koenkk/zigbee2mqtt #3567: Problem with CC1352P-2

schurgan, 2020-05-18:
Hello, I can't get the service to start. I have a CC1352P-2 as an adapter. I flashed it with UNIFLASH according to the instructions. The following error occurs when I start the service. Please support. thank you

alex@alex-desktop:/opt/zigbee2mqtt$ sudo npm start

> zigbee2mqtt@1.13.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-05-18 22:09:42: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-05-18.22-09-42' filename: log.txt
zigbee2mqtt:info  2020-05-18 22:09:43: Starting zigbee2mqtt version 1.13.0 (commit #55e6283)
zigbee2mqtt:info  2020-05-18 22:09:43: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-18 22:10:51: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-18 22:10:51: Failed to start zigbee
zigbee2mqtt:error 2020-05-18 22:10:51: Exiting...
zigbee2mqtt:error 2020-05-18 22:10:51: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:158:23
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/alex/.npm/_logs/2020-05-18T20_10_51_316Z-debug.log

## Debug Info
Zigbee2mqtt version:1.13.0
Adapter hardware: CC1352P-2
Adapter firmware version:latest

mr2c12, 2020-05-19:
As reported in the log:

> Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel

You can change them in the configuration file: https://www.zigbee2mqtt.io/information/configuration.html

schurgan, 2020-05-19:
What should I change?  I don't know what I did wrong

mr2c12, 2020-05-20:
You should try to change both panid and channel (you'll need to repair any device already paired).
I can't say what you did wrong but the error seems pretty clear.
Those settings are in the configuration.yaml file, in the zigbee2mqtt folder.

closed 2020-06-04 by schurgan