Koenkk/zigbee2mqtt #3568: restart after power off, why the Xiaomi MiJia light intensity sensor (GZCGQ01LM) data can not be received, must press a light sensor above the button once before returning to normal

hyzgit, 2020-05-19:
restart after power off, why the Xiaomi MiJia light intensity sensor (GZCGQ01LM) data can not be received, must press a light sensor above the button once before returning to normal

hyzgit, 2020-05-19:
When you press the button, you print the log

zigbee2mqtt:debug 2020-05-15 17:56:06: Device '0x04cf8cdf3c772313' announced itself
,zigbee2mqtt:info  2020-05-15 17:56:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x04cf8cdf3c772313"}}'

sjorge, 2020-05-19:
It's an issuew it most (all?) Xiaomi devices. They will not change parent once the existing one disapears for a bit.

Hitting the button will make it look for a new one. It's super annoying but this is something that z2m has no control over.

hyzgit, 2020-05-20:
> It's an issue with most (all?) Xiaomi devices. They will not change parent once the existing one disapears for a bit.
> 
> Hitting the button will make it look for a new one. It's super annoying but this is something that z2m has no control over.

Is there no way to solve this problem? When connecting to Xiaomi Gateway, there should be no such problem, then how he realized it

hyzgit, 2020-06-03:
Any friends have a good solution?

hyzgit, 2020-07-23:
Still have this problem, any solution?

Koenkk, 2020-07-23:
To debug this we would need to sniff the traffic of this issue: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2020-08-31 by github-actions[bot]