Koenkk/zigbee2mqtt #3569: [feature request] Binary sensors invert option

to4ko, 2020-05-19:
# Feature request
## What happened
Would be handful to be able to have option to invert logic of binary sensors logic (like door\window ones) thru devices.yaml

Will help to avoid creation of unnecessary entities in software like HA.

## What did you expect to happen
Be able to invert binary logic
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13
Adapter hardware: CC2538

Koenkk, 2020-05-19:
Similar to #3541 this can be done with:

```yaml
device_options:
  homeassistant:
    occupancy:
      payload_on: false
      payload_off: true
```

to4ko, 2020-05-20:
> Similar to #3541 this can be done with:
> 
> ```yaml
> device_options:
>   homeassistant:
>     occupancy:
>       payload_on: false
>       payload_off: true
> ```

Thanks! Was not able to find it thru manual (

closed 2020-05-20 by to4ko