Koenkk/zigbee2mqtt #3571: Xiaomi GZCGQ01LM is extreemly chatty

sjorge, 2020-05-19:
Splitting of from this comment: https://github.com/Koenkk/zigbee2mqtt/issues/3177#issuecomment-630264362

As my zzh came I thought I'd look into the GZCGQ01LM, after removing it form my main network the stability went up a lot.

With the change (and bumping the configureKey):
```node
            await configureReporting.illuminance(
                endpoint, {minimumReportInterval: 10, reportableChange: 100},
            );
```

it indeed only report every 10 seconds instead of ... nearly every second.

```
zigbee2mqtt:info  2020-05-19 16:40:29: Successfully configured '0x04cf8cdf3c776322'
zigbee2mqtt:info  2020-05-19 16:40:32: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17559,"illuminance_lux":56,"linkquality":156}'
zigbee2mqtt:info  2020-05-19 16:40:35: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17076,"illuminance_lux":50,"linkquality":141}'
zigbee2mqtt:info  2020-05-19 16:40:38: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":150}'
zigbee2mqtt:info  2020-05-19 16:40:41: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17243,"illuminance_lux":52,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:40:44: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17243,"illuminance_lux":52,"linkquality":150}'
zigbee2mqtt:info  2020-05-19 16:40:47: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:40:50: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:40:55: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:40:55: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:00: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:41:05: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17243,"illuminance_lux":52,"linkquality":141}'
zigbee2mqtt:info  2020-05-19 16:41:05: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17243,"illuminance_lux":52,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:10: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17243,"illuminance_lux":52,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:15: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":150}'
zigbee2mqtt:info  2020-05-19 16:41:15: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:41:20: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17076,"illuminance_lux":50,"linkquality":141}'
zigbee2mqtt:info  2020-05-19 16:41:20: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17076,"illuminance_lux":50,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:25: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17076,"illuminance_lux":50,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:30: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:41:30: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:41:35: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:35: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:40: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:41:45: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:45: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":141}'
zigbee2mqtt:info  2020-05-19 16:41:50: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:50: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:41:55: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17161,"illuminance_lux":51,"linkquality":150}'
zigbee2mqtt:info  2020-05-19 16:42:00: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:42:00: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":141}'
zigbee2mqtt:info  2020-05-19 16:42:05: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:42:05: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:42:10: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:42:15: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17076,"illuminance_lux":50,"linkquality":138}'
zigbee2mqtt:info  2020-05-19 16:42:15: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17076,"illuminance_lux":50,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:42:20: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:42:20: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:42:25: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":144}'
zigbee2mqtt:info  2020-05-19 16:42:30: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":150}'
zigbee2mqtt:info  2020-05-19 16:42:30: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:42:35: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:42:35: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:42:40: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:42:44: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:42:44: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:42:49: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":150}'
zigbee2mqtt:info  2020-05-19 16:42:49: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:42:54: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17482,"illuminance_lux":55,"linkquality":147}'
[... cut ...]
zigbee2mqtt:info  2020-05-19 16:43:49: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:43:49: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":147}'
zigbee2mqtt:info  2020-05-19 16:43:54: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776322', payload '{"illuminance":17404,"illuminance_lux":54,"linkquality":153}'
```

The sensor is just sitting on my desk far away from a window so light level variance is very small... It's still extremely chatty.

sjorge, 2020-05-19:
I played around with some other values, like report once every 60 seconds reportChange 0... but it does not seem to have any effect.

@Koenkk I currently have 3 of these (but will not be using them anymore because of the above mention problem and them like all Xiaomi not moving routers), I can send you one if you like as you live just across the border I so it should not cost me much.

sjorge, 2020-05-19:
Also some funky stuff:
- remove from network (was successful, device stopped spamming)
- insert battery
- device announces and joins network and resumes spamming
- double check, permit_join was disabled!

I can repeat the above!

Koenkk, 2020-05-19:
Can you play a bit with the reportable change value? Even every 10 seconds doesn't make any sense (only when there was a big change). Maybe `reportableChange: 1000`?

About the remove, how did you remove the device? (through `remove` or `force_remove`?)

sjorge, 2020-05-19:
> Can you play a bit with the reportable change value? Even every 10 seconds doesn't make any sense (only when there was a big change). Maybe `reportableChange: 1000`?

I think I already tried, but will try again tomorrow or Thursday (more likely).

> About the remove, how did you remove the device? (through remove or force_remove?)

normal remove, but I have to hit the button on the device else it fails (battery powered ZED). But if I do it leaves properly and stops sending messages.

I then remove the battery for storage, next time I insert the battery it rejoined even without having permit_join or or pushing the button on the device to join.

sjorge, 2020-05-20:
I tried with:
```{minimumReportInterval: 60, reportableChange: 1000},```
and
```{minimumReportInterval: 60, reportableChange: 0},```

(of course bumping the configureKey)

No effect, I would have expected the later to only report every 60 seconds but it keeps spamming.

sjorge, 2020-05-20:
```{minimumReportInterval: 60, reportableChange: 10000},```

After this one the device seems to have stopped working? It configure OK but then nothing is received... after removing/inserting the battery there is also no more device announce.

sjorge, 2020-05-20:
Removing the battery for about an hour seems to make it join again... still seems to ignore all configuration options.

Koenkk, 2020-05-21:
In my case it also ignores all configure settings (keeps spamming every 5 seconds). This device has only 1 endpoint so this is the correct one.

To further investigate this someone should sniff the traffic with the original gateway (to see what the behaviour is there). Unfortunately my Xiaomi gateway does not support this device.

sjorge, 2020-05-21:
Yeah, I'm going to avoid these for now... if I add all 3 I have the network just goes to shit :/

z4rn0x, 2020-11-15:
Any progress made with this sensor?