Koenkk/zigbee2mqtt #3573: Migration from CC2531 to ConBee II

LukeChannings, 2020-05-19:
Hello,

I have a CC2531 that I want to move away from (I have 60+ devices), and I bought a ConBee II. I know the ConBee support is still ongoing, and I'm using the latest-dev tagged docker image.

I copied the exact configuration that I was using previously (network_key, channel, database, state.json, everything), and I started up zigbee2mqtt against the ConBee instead of the CC2531.

What I got was an absolute ton of repeated messages, like:

```
[zigbee2mqtt-5555976c54-vkh8k] zigbee2mqtt:info  2020-05-19 19:13:32: MQTT publish: topic 'zigbee2mqtt/office_control', payload '{"counter":1,"linkquality":210}'
```

Repeated a few thousand (MQTT Explorer says 57,000 on a single topic!) times.

I also get:

```
[zigbee2mqtt-5555976c54-vkh8k] zigbee2mqtt:info  2020-05-19 19:12:36: MQTT publish: topic 'homeassistant/sensor/0x00158d00036b8dfc/pressure/config', payload '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","state_topic":"zigbee2mqtt/office_climate","json_attributes_topic":"zigbee2mqtt/office_climate","name":"office_climate_pressure","unique_id":"0x00158d00036b8dfc_pressure_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00036b8dfc"],"name":"office_climate","sw_version":"Zigbee2mqtt 1.13.0-dev","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

Here's some sample output: [log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4652689/log2.txt)

So I guess the questions are:
- Is this something that should work?
- Is it a bug in the ConBee II support?

Do I just need to re-pair everything?

sjorge, 2020-05-19:
I don't think you can migrate between a CC2531 and a ConBee II without repairing all devices.

andriej, 2020-05-20:
How about staying on Conbee and migrating to zigbee2mqtt (with same stick)?

Koenkk, 2020-05-20:
@andriej migrating from Deconz to zigbee2mqtt also requires repairing

andriej, 2020-05-20:
Oh, ok. I thought I'll be all good at least with that direction.
Two questions then:
1. When do you estimate it will be 'ready enough' for first stable-enthustiasts to move on? :-) I have 60+ devices right now. Like non-dev release and info in docs.
2. Device's support - after moving to z2m all devices z2m supports will be available on conbee stick or there's still some 'backhaul' of "needs to be supported by deconz" first?

Koenkk, 2020-05-21:
1. Hard to say, it's seems to work good now, but not as stable as the TI CC adapters I expect (exposure time is just much less). A good thing is that in case of issues Dresden responds fast (https://github.com/Koenkk/zigbee-herdsman/issues/72)  
2. Device support is the same.

closed 2020-07-20 by LukeChannings