Koenkk/zigbee2mqtt #3574: New Device CALEX Smart Wall Switch - does not fully pair

VincentSC, 2020-05-20:
# Bug Report
## What happened
Device Calex 421782 wall switch gets to announce, is in the networkmap (connected to a device), but the device does not get into the "connected" state. This means that pressing a button does not send the command, but triggers a new try to connect. 

[image]

So it looks like a problem with the device, where it expects some acknowledgement? Resetting does not help.

## What did you expect to happen
To connect and buttons to work, like in https://github.com/Koenkk/zigbee2mqtt/issues/2102  
The device should be in the devices list with a trick to overcome its stubbornness

## How to reproduce it (minimal and precise)
[image]
Press K1+K4 for 3 seconds. Device gets announced. Then nothing happens. After a while pressing any button triggers a new announce.

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-05-21:
Could you share the herdman debug log when pairing the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

VincentSC, 2020-05-24:
Not sure if done correctly, because I did not see the messages in the normal debug-windows anymore.

https://gist.github.com/VincentSC/7c72e37a9a3011ff441adf876056e84b

Koenkk, 2020-05-25:
I don't see any pairing happening in this log. First remove the device through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove -> repair and capture the full log.

VincentSC, 2020-05-31:
Removal failed - tried sending `0x00158d00024f395a` to `zigbee2mqtt/bridge/config/remove`, doing a factory reset and removing the battery.

In this log there is an `Device '0x00158d00024f395a' announced itself`: https://gist.github.com/VincentSC/b3b1000cc16af64419690b34063a53a4

Does this help?

Koenkk, 2020-06-01:
In case removal fail you should do a force_remove (check https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove)

VincentSC, 2020-06-01:
Yes, did that. 

Does the new log help?

Koenkk, 2020-06-01:
I've checked again and it seems that the device has been paired succesfully (`  zigbee-herdsman:controller:log Not interviewing '0x00158d00024f395a', completed 'true', in progress 'false' +1ms
`) otherwise it woudln't be completed.

In order to receive button events you probably need to setup binds (e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1490)

The `database.db` entry should reveal the endpoints and the clusters the the device supports.

VincentSC, 2020-06-01:
I'm a bit confused. 

Will the announce-messages go away when the bindings are done correctly? 

This is something different than https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html ? I'm not seeing words like "bind" and "database.db" there. Can you point me to some documentation, as I seem to lack context.

Koenkk, 2020-06-02:
This is something that can hardly be documented as it differs per device. Some devices require binding to be setup, and this could be one of them (this is just a guess based on my earlier experience). I'm not sure if the announces will be gone after that (only way to find out is to try).

If you could share your `data/database.db` entry of this device I can do an initial guess on how to setup.

VincentSC, 2020-06-02:
This is also context. Here seemingly some trial-on-error is required. I'll look at other devices, how these were done, experiment (and thus learn) and then come back to you (with progress and the database.db). Thanks for all the support till now!

VincentSC, 2020-11-28:
Took https://github.com/Koenkk/zigbee2mqtt/issues/2038 as a base and that works pretty well. I can use all buttons now. The color-ring is slow (only Hue, Saturation always 254 and not other value), and the up+down levels are not giving any value just an up or down. The W-button is weird - expected just white, but it rotates through 5 numbers. So it works, but we must never speak of its design.

The original problem still exists: it announces itself continuously. No idea how to solve this.

VincentSC, 2020-11-28:
@Koenkk Can you please reopen this issue?

closed 2020-12-07 by github-actions[bot]