Koenkk/zigbee2mqtt #3575: Failure to configure Centralite 4257050-RZHAC

nanobra1n, 2020-05-20:
I see in my log the following error for the supported Centralite 4257050-RZHAC: 
```
Failed to configure 'Outlet', attempt 1 (Error: Bind 0x000d6f0004482310/1 genOnOff from '0x00124b001ca5cb1c/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```
The database entry for this device is: 
```
{"id":5,"type":"Router","ieeeAddr":"0x000d6f0004482310","nwkAddr":25889,"manufId":49887,"manufName":"CentraLite","powerSource":"Mains (single phase)","modelId":"4257050-RZHAC","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2820,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"4257050-RZHAC","manufacturerName":"CentraLite","powerSource":1,"zclVersion":1,"stackVersion":0,"hwVersion":3,"dateCode":"����������������"}},"genOnOff":{"attributes":{"onOff":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":0,"activePower":0,"rmsCurrent":0,"acCurrentMultiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5cb1c","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5cb1c","endpointID":1}]}},"stackVersion":0,"hwVersion":3,"dateCode":"����������������","zclVersion":1,"interviewCompleted":true,"meta":{"configured":3,"reporting":1},"lastSeen":1579303902523}
```
## What happened
I see three attempts in my log, after which z2m gives up. The device is still controllable. Not using the power measurement capability, so can't say if it works or not
## What did you expect to happen.
No error
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC1352P-2
Adapter firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'

Koenkk, 2020-05-21:
Can you try repairing? It seems to somehow the device does not respond to requests, which is strange for an AC powered device. Is it controllable when moving it closer to the coordinator?

nanobra1n, 2020-05-21:
Oh, it the device still works:
```
zigbee2mqtt:info  2020-05-21 09:50:02: MQTT publish: topic 'zigbee2mqtt/Outlet', payload '{"state":"ON","linkquality":63,"last_seen":"2020-05-21T09:50:02-04:00","voltage":0,"power":0,"current":0}'
```
(Nothing connected to it, I am just using it as a router).

I was paired many versions ago (1.8?) but this error didn't show up until maybe 1.12 or 1.11.

I can try and repair the device later if you think that helps.

nanobra1n, 2020-05-24:
Removed the device and paired again, same problem:
```
zigbee2mqtt:info  2020-05-24 16:49:30: MQTT publish: topic 'zigbee2mqtt/Outlet', payload '{"power":-1,"linkquality":147,"last_seen":"2020-05-24T16:49:30-04:00"}'
zigbee2mqtt:info  2020-05-24 16:49:30: Configuring 'Outlet'
zigbee2mqtt:error 2020-05-24 16:49:31: Failed to configure 'Outlet', attempt 3 (Error: Read 0x000d6f0004482310/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:182:23)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```
Again, the outlet reacts to ON/OFF commands and I plugged something in and it appears to report power:
```
zigbee2mqtt:info  2020-05-24 16:55:18: MQTT publish: topic 'zigbee2mqtt/Outlet', payload '{"power":15,"linkquality":177,"state":"ON","last_seen":"2020-05-24T16:55:18-04:00"}'
```
No idea if the power measurement is correct, but except for zigbee2mqtt throwing that error message on the config attempts, everything appears to be working okay.

Koenkk, 2020-05-25:
Try changing the entry in `devices.js` to 

```js
{
    zigbeeModel: ['4257050-RZHAC'],
    model: '4257050-RZHAC',
    vendor: 'Centralite',
    description: '3-Series smart outlet',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await configureReporting.onOff(endpoint);
        // await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.rmsVoltage(endpoint, {'reportableChange': 2}); // Voltage reports in V
        await configureReporting.rmsCurrent(endpoint, {'reportableChange': 10}); // Current reports in mA
        await configureReporting.activePower(endpoint, {'reportableChange': 2}); // Power reports in 0.1W
    },
},
```

If it now configures successful, make sure that the reported `power` is in watts, `current` is in ampere and `voltage` is in volts. My concern is that after this change the multiplier is incorrect, so actual power could be e.g. 150 while 15 is reported.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to get to devices.js

nanobra1n, 2020-05-25:
Did that. Now Configure is succcessful:
```
info  2020-05-25 16:54:57: Successfully configured 'Outlet'
```
Report is 
```
zigbee2mqtt:info  2020-05-25 17:07:28: MQTT publish: topic 'zigbee2mqtt/Outlet', payload '{"power":1,"linkquality":108,"state":"ON","voltage":114,"current":3,"last_seen":"2020-05-25T17:07:28-04:00"}'
```

Koenkk, 2020-05-26:
What divisor/multiplier has to be used? Can you attach a known load (e.g. 50 watt bulb) and see what it reports?

nanobra1n, 2020-05-31:
Turns out I have no incandescent bulbs anymore, so I had to try with an LED bulb. The bulb claims 120V, 93 mA, 11W and I get this from the outlet:
```
zigbee2mqtt:info  2020-05-31 09:34:36: MQTT publish: topic 'zigbee2mqtt/Outlet', payload '{"power":112,"linkquality":150,"state":"ON","voltage":117,"current":97,"last_seen":"2020-05-31T09:34:36-04:00"}'
```
So a divisor of 10.

Koenkk, 2020-06-01:
Fixed it, fix is available in the latest dev branch in a few hours.