Koenkk/zigbee2mqtt #3576: Innr SP-120 sends no power consumption data anymore

vdbrink, 2020-05-20:
# Bug Report
## What happened
My Innr SP120 socket doesn't send power consumption (metered) data anymore.
In the previous version, 1.13.0, this was also broken but I didn't saw that direct (because I don't use it yet). I only saw the error at boot and so I logged issue #3488 for that.
Now I see there is no power consumption data at all anymore. In an earlier version this worked correct and I had a constant stream of power consumption visible in the logs. 
Now also the payload Is reduced to: 
```
{"state":"OFF"}
```

## What did you expect to happen
A constant stream of logs like this
```
{"state":"OFF","linkquality":57,"power":20,"voltage":231,"current":0}
```

## How to reproduce it (minimal and precise)
Pair the SP120 power socket to zigbee2mqtt and put the socket in the wall.

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-05-21:
Can you try to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

vdbrink, 2020-05-22:
I reconfigured it, as described, this is what I see in the logs:
```
Configuring 'innrsmartplug2'
Successfully configured 'innrsmartplug2'
```
But still the payload is only the `state` and only when the state is changed.

Koenkk, 2020-05-22:
Could you provide the debug logging when attaching a load to the plug? (e.g. kettle)

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

vdbrink, 2020-05-24:
I solved it.
After I removed the device via `zigbee2mqtt/bridge/config/remove` it directly joined again and now I get the current and power data again.

closed 2020-05-24 by vdbrink