Koenkk/zigbee2mqtt #3577: The quest to find the best antenna with SMA connector

copystring, 2020-05-20:
Excuse me if this has been asked anywhere else. I did not find one.

Now with zzh being out (I got mine today) I want to swap the antenna for a "stronger" one. As strong as I can find.
The signal of zzh is good. Sure. But I want the best possible signal.

Having no clue what makes an antenna "good" and emit a "strong" signal I think it helps everyone to collect knowledge about good antennas here (or should this be in the forum instead? let me know).

So my questions are:

**What makes an antenna emit a strong signal?**

**How does gain (high gain) mean?**

**Does orientation of the antenna matter?**

**How does the antenna signal spread in space? Is it a sphere or sphere like? Is it cone shaped?**

sjorge, 2020-05-20:
I think most 2.4GHz wifi antennas should work, but I haven't tried any. I got zzh yesterday and the bundled one is already a massive improvement over my CC2531.

copystring, 2020-05-20:
I found this at home: https://www.ebay.de/itm/SMA-2-4GHz-9DBI-Wireless-Wifi-WLAN-5-X-Range-Booster-Antenna-Extender-Base-w/353032695186
It seems to have 3dbi gain and seems a little better than the one that came with the zzh. I'm looking for more.
The Eightwood antenna seems to have 9dbi gain. If you would find that one to test and compare to the standard zzh antenna might give some insight.

peca89, 2020-05-21:
> So my questions are:
> **What makes an antenna emit a strong signal?**
> **How does gain (high gain) mean?**
> **Does orientation of the antenna matter?**
> **How does the antenna signal spread in space? Is it a sphere or sphere like? Is it cone shaped?**

This is very general info and applies to all antennas. Ideal antenna (isotropic radiator) emmits the signal in all directions equally and the signal amplification in all directions looks like a sphere. The problem is, such antenna does not exist in practice. The close things are so called omnidirectional antennas. They usually emit equally in all directions in the plane perpendicular to them. The most famous shape is dipole antenna - those sticks you have on your wifi router. The higher the angle in respect to that plane, weaker the signal. Theoretically, nothing is emmitted in direction that dipole antenna "points", but in practice, you will get some signal reflections everywhere. Antenna is a passive element, it cannot amplify signal. However, it can "redistribute" its radiation pattern in space meaning that it can emmit in one direction stronger than in the other. The antenna gain is virtual amplification effect that you see in one direction only - the maximum radiation direction. So, yes, antenna direction does matter. The antenna gain is the measurement by how much higher amount does particular antenna emmit in one direction in comparison to imaginary isotropic radiator. One more thing: all antennas are symmetric devices, meaning everything I said before about emitting, also applies for receiving a signal, the same directions, gains and radiation patterns.

giejay, 2020-06-26:
Any updates? I'm also trying to improve the signal, it's pretty crappy right now. I ordered this one: https://m.nl.aliexpress.com/item/32913680322.html?trace=wwwdetail2mobilesitedetail&spm=a2g0s.12269583.0.0.7b0346bcbTPJ7S. Will post the results when it's arrived

copystring, 2020-07-07:
Yes. However I went a slightly different route. I ordered a 2.4 GHz 4 watt amplifier.
This amplifier needs higher input than zzh/CC2652R provides.
So I will wait until zzh with power amplifier is available.

giejay, 2020-07-07:
The one I ordered needs a higher input?

copystring, 2020-07-08:
No. You ordered antenna. Not an amplifier. Some amplifiers need higher input.

closed 2020-08-16 by github-actions[bot]