Koenkk/zigbee2mqtt #3582: Integration with Danalock V3 is not properly setup

elbueno222, 2020-05-21:
# Bug Report
## What happened

The integration with Danalock is not properly setup, the lock status is being retrieved from the "state" value sent by the lock, but it should be sent from the "lockState" value. The "state" value is sending "lock" or "unlock" regardless of the lock or unlock operation being successful. According to the testing I have been doing the device is sending "lockState" value of 2 for unlocked, 1 for locked and 0 for error. In order to see these values I had to change the log level to debug (using topic zigbee2mqtt/bridge/config/log_level with payload=debug) as the lockState value only appears under zigbee2mqtt:debug messages.

This setup is causing zigbee2mqtt to report the wrong status to the hub (Home Assistant in my case) when the lock cannot be locked for some reason. In this case the hub shows status "locked" when it should show status "unlocked" or "error".

## What did you expect to happen

Lock status for the Danalock should be retrieved from the "lockState" value instead of the "state" value.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC2531
Adapter firmware version: Z-Stack_Home_1.2 (default)

Koenkk, 2020-05-21:
Please try changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/b9d2237e69d32c23e3f2aa39210393bc803bc52c/converters/toZigbee.js#L1318 to `return {readAfterWriteTime: 200};`. Does it work as expected then?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to get to `toZigbee.js`.

elbueno222, 2020-05-21:
I tried to follow the instructions for Hass.io addon (https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#hassio-addon) but the link in the first step (https://developers.home-assistant.io/docs/en/hassio_debugging.html) is broken.

I already have access to Home Assistant host through ssh (I am not sure if it is the same thing), but when I execute the command "docker ps" (third step of the instructions) I get the error "-bash: docker: command not found". I am running Home Assistant in a Raspberry Pi.

Koenkk, 2020-05-22:
I've fixed the link in the documentation (https://developers.home-assistant.io/docs/operating-system/debugging/#ssh-access-to-the-host). 

It seems to be important that you connect through port 2222. Not 22 as it won't give all permissions, maybe that's the reason why you cannot access the docker client.

elbueno222, 2020-05-22:
I have followed the steps and updated the line in my toZigbee.js (it was line 1311 instead of 1318, not sure if that´s important) but after restarting the container the error is still happening.

I have done a new test sending a mqtt message with the topic `zigbee2mqtt/[danalock_id]/get` and in the response I get by listening to the same topic (zigbee2mqtt/[danalock_id]) I don´t see the field lockState. This is the response I am getting:

```
{
    "state": "LOCK",
    "user": 65535,
    "source": 1,
    "linkquality": 60,
    "battery": 88
}
```

The only way I can see the field lockState (which contains the lock status) is by activating the debug mode in zigbee2mqtt and then I can see this message containing the lockState value:

`zigbee2mqtt:debug 2020-05-22 13:13:05: Received Zigbee message from '[danalock_id]', type 'readResponse', cluster 'closuresDoorLock', data '{"lockState":0}' from endpoint 1 with groupID 0`

elbueno222, 2020-05-22:
I think this is really close to exceed my coding skills, but I think I found the issue and how to solve it. I found the value lockState in https://github.com/Koenkk/zigbee-herdsman-converters/blob/b9d2237e69d32c23e3f2aa39210393bc803bc52c/converters/fromZigbee.js#L345 and if I am not wrong the sentence is returning "LOCK" for every value except 2, and that is incorrect as my lock is returning 0 when the lock cannot lock the door so it´s in unlocked status (at least for the Danalock). The code should return "LOCKED" for 1, "UNLOCKED" for 2, and "ERROR" for 0. And I say "LOCKED" instead of "LOCK" because I think it is more clear for lock status to return LOCK and UNLOCKED rather than LOCK and UNLOCK (this way is easy to see the difference between the commands sent to the lock and the commands received from the lock with the status).

As a small test I updated that line to `return {state: msg.data.lockState == 2 ? 'UNLOCKED' : 'LOCKED'};` and even the status is still not correct (because the value 0 is not considered) at least I can see the messages containing "LOCKED" and "UNLOCKED" in the zigbee2mqtt log. This test was just to ensure I was looking at the right place.

So just to summarize, can that line of code be updated to consider the 3 different values (0 for error, 1 for LOCKED, and 2 for UNLOCKED), and this value be used to report the status of the lock?

Koenkk, 2020-05-22:
We can only provide the value `LOCKED` or `UNLOCKED` (it's binary). In case of an error the state should stay `UNLOCK`.

elbueno222, 2020-05-22:
If it is binary then I agree, and I guess the code should be `return {state: msg.data.lockState == 1 ? 'LOCKED' : 'UNLOCK'};`. Let me test it and I will let you know if it works

elbueno222, 2020-05-22:
I have tested and it works but only partially. The attribute "state" of the lock shows the correct status in uppercase characters ("LOCKED" only when value is 1 and "UNLOCKED" otherwise) but the State of the lock shows the wrong one and lowercase characters  ("locked" when the attribute "state" shows "UNLOCKED"). I am not sure why they are not in sync.

elbueno222, 2020-05-23:
I think the final solution I mentioned in my last comment can be achieved by changing 3 lines: the line you mentioned before (toZigbee.js#L1318), the one I mentioned (fromZigbee.js#L345) but using states LOCK and UNLOCK instead of LOCKED and UNLOCKED (after my testing I realised the lock entity is waiting for LOCK and UNLOCK values and cannot be changed, otherwise the status in not updated) and also https://github.com/Koenkk/zigbee-herdsman-converters/blob/b9d2237e69d32c23e3f2aa39210393bc803bc52c/converters/fromZigbee.js#L334 to `request: unlockCodes.includes(msg.data['opereventcode']) ? 'UNLOCK' : 'LOCK',` so the value returned is called "request" instead of "status" and it doesn't affect the lock status.

I have applied this 3 changes to my files and it looks like the status of the lock is now reflecting the correct status.

elbueno222, 2020-05-23:
I just thought it would be a good idea to return the lockState as an attribute as well, so we could use that attribute to know the real status of the lock (2=Locked, 1=UNLOCKED, 0=ERROR) so I added a new line after https://github.com/Koenkk/zigbee-herdsman-converters/blob/b9d2237e69d32c23e3f2aa39210393bc803bc52c/converters/fromZigbee.js#L345 to return that value.

Just to summarise, these are all the updates I have done:
- https://github.com/Koenkk/zigbee-herdsman-converters/blob/b9d2237e69d32c23e3f2aa39210393bc803bc52c/converters/fromZigbee.js: updated line 334 and a new line added after 345, this is the final version:

```
    lock_operation_event: {
        cluster: 'closuresDoorLock',
        type: 'commandOperationEventNotification',
        convert: (model, msg, publish, options, meta) => {
            const unlockCodes = [2, 9, 14];
            return {
                command: unlockCodes.includes(msg.data['opereventcode']) ? 'UNLOCK' : 'LOCK',
                user: msg.data['userid'],
                source: msg.data['opereventsrc'],
            };
        },
    },
    lock: {
        cluster: 'closuresDoorLock',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('lockState')) {
                return {
                    state: msg.data.lockState == 1 ? 'LOCK' : 'UNLOCK',
                    lockState: msg.data['lockState'],
                };
            }
        },
    },
```

- https://github.com/Koenkk/zigbee-herdsman-converters/blob/b9d2237e69d32c23e3f2aa39210393bc803bc52c/converters/toZigbee.js#L1318 updated to `return {readAfterWriteTime: 200};`

Koenkk, 2020-05-23:
Could you make a PR with these changes? (this makes it easier to review)

elbueno222, 2020-05-24:
Never done it before, but it is always a good time to start, what steps should I do? is there any guide I can follow to make a PR?

Koenkk, 2020-05-24:
The easiest way to do this is by clicking the pencil icon on https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js (right of history). 

More detailed guide can be found here: https://opensource.com/article/19/7/create-pull-request-github

elbueno222, 2020-05-24:
I have created two pull requests (#1259 and #1260) with the updates to the two files (not sure if they can be combined in just one)

Koenkk, 2020-05-25:
Merged both so assuming this can be solved (changes will be available in latest dev branch in a few hours).

closed 2020-05-25 by Koenkk

elbueno222, 2020-05-25:
Thank you very much, will this update be included in the next version of the zigbee2mqtt add-on for Home Assistant?

Koenkk, 2020-05-26:
Yes (its already in the edge version of the hassio addon)