Koenkk/zigbee2mqtt #3583: Heiman HS1SA-M ConfigureReporting failing w/ UNSUPPORTED_ATTRIBUTE

middelink, 2020-05-21:
# Bug Report
## What happened
Recently had to repair my two HEIMAN HS1SA-M smoke detectors. However, after they were joined, iasZone and all went well, zigbee2mqtt tries to configure them. This fails.
I then restarted z2m in the hopes it was in a confused mode, but no.

First the two smokedetectors fail, likely due to them being asleep:
```
error 2020-05-21 15:45:09: Failed to configure 'HEIMAN Kantoor Smokedetector', attempt 1 (Error: Bind 0x000d6f0014bb997e/1 genPowerCfg from '0x00124b001949ee82/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
error 2020-05-21 15:45:19: Failed to configure 'HEIMAN Woonkamer Smokedetector', attempt 1 (Error: Bind 0x000d6f0014b9bda4/1 genPowerCfg from '0x00124b001949ee82/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```
However, later on, when a smoke detector sends some battery/link info, it tries configureing again, this time the device is NOT yet alseep and report:
```
zigbee2mqtt:debug 2020-05-21 16:03:57: Received Zigbee message from 'HEIMAN Woonkamer Smokedetector', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":198}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-05-21 16:03:57: MQTT publish: topic 'zigbee2mqtt/HEIMAN Woonkamer Smokedetector', payload '{"enrolled":true,"linkquality":31,"smoke":false,"battery_low":false,"battery":99}'
zigbee2mqtt:info  2020-05-21 16:03:57: Configuring 'HEIMAN Woonkamer Smokedetector'
zigbee2mqtt:error 2020-05-21 16:04:05: Failed to configure 'HEIMAN Woonkamer Smokedetector', attempt 2 (Error: ConfigureReporting 0x000d6f0014b9bda4/1 genPowerCfg([{"attribute":"batteryAlarmState","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

some database information (no idea what those empty ids are doing there, but ala)
```
$ fgrep 0x000d6f0014b9bda4 database.db 
{"id":7,"ieeeAddr":"0x000d6f0014b9bda4","nwkAddr":37894,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1590054951795}
{"id":19,"ieeeAddr":"0x000d6f0014b9bda4","nwkAddr":37894,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1590057768431}
{"id":22,"ieeeAddr":"0x000d6f0014b9bda4","nwkAddr":37894,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":null}
{"id":34,"type":"EndDevice","ieeeAddr":"0x000d6f0014b9bda4","nwkAddr":37894,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"SmokeSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SmokeSensor-EM","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":1,"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.4.20","swBuildId":"1.1.1"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001949ee82","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":193}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001949ee82","endpointID":1}]}},"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.4.20","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590059028671}
```
(both of them are of the same fw, app, date.)

## What did you expect to happen
Not failing. In fact, when I got them about half a year ago, pairing them went without an issue as far as I recall, and in fact I would definitely recall seeing these errors in the log.

## How to reproduce it (minimal and precise)
(I had the sneaking suspicion the smoke detectors were unpaired as I did not receive battery/link updates in days. not sure why though, my network is a bit wacky like that)
- Reset a HEIMAN HS1SA-M Smoke detector
- Force remove from z2m
- Pair it again
- Wait...

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC2531
Adapter firmware version: dunno, I start zigbee2mqtt with warn/error and it leaves out crucial information like that.

Koenkk, 2020-05-21:
Should be fixed by changing the definition in `devices.js` to

```js
{
        zigbeeModel: [
            'SMOK_V16',
            'b5db59bfd81e4f1f95dc57fdbba17931',
            '98293058552c49f38ad0748541ee96ba',
            'SMOK_YDLV10',
            'SmokeSensor-EM',
            'FB56-SMF02HM1.4',
        ],
        model: 'HS1SA-M',
        vendor: 'HEIMAN',
        description: 'Smoke detector',
        supports: 'smoke',
        fromZigbee: [
            fz.heiman_smoke,
            fz.battery,
            fz.heiman_smoke_enrolled,

        ],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to get to `devices.js`.

middelink, 2020-05-21:
I can confirmed it worked. After hitting the self-test button, I se in the logs:
```
zigbee2mqtt:debug 2020-05-21 18:24:24: Device 'HEIMAN Kantoor Smokedetector' announced itself
zigbee2mqtt:info  2020-05-21 18:24:24: Configuring 'HEIMAN Kantoor Smokedetector'
zigbee2mqtt:info  2020-05-21 18:24:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"HEIMAN Kantoor Smokedetector"}}'
zigbee2mqtt:info  2020-05-21 18:24:29: Successfully configured 'HEIMAN Kantoor Smokedetector'
```
and the other as well:
```
zigbee2mqtt:debug 2020-05-21 18:25:14: Received Zigbee message from 'HEIMAN Woonkamer Smokedetector', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":33,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-05-21 18:25:14: MQTT publish: topic 'zigbee2mqtt/HEIMAN Woonkamer Smokedetector', payload '{"enrolled":true,"linkquality":28,"smoke":true,"battery_low":false,"battery":100}'
zigbee2mqtt:info  2020-05-21 18:25:14: Configuring 'HEIMAN Woonkamer Smokedetector'
zigbee2mqtt:debug 2020-05-21 18:25:15: Received Zigbee message from 'HEIMAN Woonkamer Smokedetector', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":32,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-05-21 18:25:15: MQTT publish: topic 'zigbee2mqtt/HEIMAN Woonkamer Smokedetector', payload '{"enrolled":true,"linkquality":28,"smoke":false,"battery_low":false,"battery":100}'
zigbee2mqtt:info  2020-05-21 18:25:21: Successfully configured 'HEIMAN Woonkamer Smokedetector'
```

Although, weirdly, they don't seem to respond the same with the self-test. One seemed not to have seem the z2m restart. Oh well.

Koen, willl you update devices.js?

Koenkk, 2020-05-21:
Done, will be available in the dev branch tomorrow.

closed 2020-05-21 by Koenkk