Koenkk/zigbee2mqtt #3585: "interview_failed" when pairing Aqara Curtain B1

mpuff, 2020-05-21:
**Bug Report**
What happened
Pairing of a Aqara Curtain B1 fails with "interview_failed".

**What I tried**
-clear entries of new device in devices.yaml & database.db

**What did you expect to happen**
to pair with "interview succesful", as the device is on the supported devices list (ZNCLDJ12LM)

**Debug Info**
zigbee2mqtt version: 1.13.1
CC1352P_2 firmware version: 20191106

Log:
info  2020-05-21 20:29:10: Device '0x04cf8cdf3c72bab1' joined
info  2020-05-21 20:29:10: Starting interview of '0x04cf8cdf3c72bab1'
error 2020-05-21 20:31:29: Failed to interview '0x04cf8cdf3c72bab1', device has not successfully been paired
info  2020-05-21 20:31:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x04cf8cdf3c72bab1"}}'

artist67, 2020-05-23:
Looks like the interview runs into a timeout.
Maybe, the situation ist similar to the aqara cubes, for which nother single pairing button press after 5 secs helps to keep the communication open.

Koenkk, 2020-05-23:
Please share the herdsman debug logs, but DON'T directly post them here (but on pastebin.com and link it here).

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

mpuff, 2020-05-27:
i have stopped zigbee2mqtt, than i have tried the command DEBUG=zigbee-herdsman* npm start, after this i get these error:

npm ERR! npm star [<pkg>...]
npm ERR! npm unstar [<pkg>...]
npm ERR!
npm ERR! alias: unstar

mpuff, 2020-05-27:
and i have one info about these curtain b1, it was connected to zigbee2mqtt since 1 year, i have tried something than i have force removed from zigbee2mqtt, now i cant connect it again.
i have also a secound one that is currently connected to zigbee2mqtt

mpuff, 2020-05-28:
https://pastebin.com/JS91pyzt

Koenkk, 2020-05-28:
You should execute this command in the directory where zigbee2mqtt is installed (`/opt/zigbee2mqtt`)

mpuff, 2020-05-28:
Thank you, now i have the logs:

Part 1 bevore allow to join:
https://pastebin.com/X7kqXSrA

Part 2 after allow to join until the error message inerview_failed:
https://pastebin.com/h5s6SyGg

Koenkk, 2020-05-29:
Can you try replacing it close to the coordinator? Also make sure to check the steps from: https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all

mpuff, 2020-05-30:
Yes, I have also removed the battery and plug on the power supply

closed 2020-07-26 by stale[bot]