Koenkk/zigbee2mqtt #3586: My zigbee devices no longer communicate

kwetnico, 2020-05-21:
# Bug Report
Hello,

I have been using this solution for 6 months without problem with my zigbee sensors (lots of xiaomi, temperature, contact, presence).
Since this morning, there is a break in the lifts of my devices on my zigbee network.
I have my coordinator (raspberry pi4) and I had 2 gateways (a Heiman siren and 1 xiaomi zigbee electrical outlet) and 21 peripherals.
Everything worked very well, the network was well distributed over the different elements.
I updated the latest zigbeemqtt version yesterday, I had no problem except an error message my npm version, so I updated my npm version and after more error message , everything worked fine.
This morning I got my 2 Siren and Xiaomi plug devices with this error:
failed to execute lqi
suddenly all the devices connected to it found themselves disconnected from the network, I tried to restart, impossible to re-associate it doesn't matter.
I decided to reflash with the CC2531_DEFAULT_20190608 version my CC2531 key.
My devices are now connected to nothing.
When I try to reappear them I have no message in the zigbeemqtt log.
I do not know what to do ..
(same problem with Dev version)
Thanks for your help.

Extract from my configuration file:
"homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt: // localhost'
advanced:
  log_level: debug
serial:
  disable_led: true
  port: / dev / ttyACM0
devices:
  '0x00158d000411ea0c':
    friendly_name: Temperature_Lyne
    retain: false
  '0x00158d00041193d8':
    friendly_name: Temperature_Bureau
    retain: false
  '0x00158d000411eb43':
    friendly_name: Temperature_Chambre
    retain: false
  '0x00158d000411eaa3':
    friendly_name: Temperature_Velux
    retain: false
.... "

Zigbeemqtt start message:
May 21 21:07:22 raspberrypi systemd [1]: Started zigbee2mqtt.
May 21 21:07:25 raspberrypi npm [28081]:> zigbee2mqtt@1.13.1 start / opt / zigbee2mqtt
May 21 21:07:25 raspberrypi npm [28081]:> node index.js
May 21 21:07:28 raspberrypi npm [28081]: zigbee2mqtt: info 2020-05-21 21:07:28: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-05-21.21-07 -28 'filename: log.txt
May 21 21:07:28 raspberrypi npm [28081]: zigbee2mqtt: debug 2020-05-21 21:07:28: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-05-21.20-17-15 ''
May 21 21:07:28 raspberrypi npm [28081]: zigbee2mqtt: debug 2020-05-21 21:07:28: Loaded state from file /opt/zigbee2mqtt/data/state.json
May 21 21:07:28 raspberrypi npm [28081]: zigbee2mqtt: info 2020-05-21 21:07:28: Starting zigbee2mqtt version 1.13.1 (commit # 3046680)
May 21 21:07:28 raspberrypi npm [28081]: zigbee2mqtt: info 2020-05-21 21:07:28: Starting zigbee-herdsman ...
May 21 21:07:28 raspberrypi npm [28081]: zigbee2mqtt: debug 2020-05-21 21:07:28: Using zigbee-herdsman with settings: '{"network": {"panID": 6754, "extendedPanID" : [221,221,221,221,221,221,221,221], "channelList": [11], "networkKey": "HIDDEN"}, "databasePath": "/ opt / zigbee2mqtt / data / database.db", "databaseBackupPath": "/ opt / zigbee2mqtt / data /database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ ttyACM0 "}," adapter ": {" competitor ": null}} '
May 21 21:07:30 raspberrypi npm [28081]: zigbee2mqtt: info 2020-05-21 21:07:30: zigbee-herdsman started
May 21 21:07:30 raspberrypi npm [28081]: zigbee2mqtt: info 2020-05-21 21:07:30: Coordinator firmware version: '{"type": "zStack12", "meta": {"transportrev": 2, "product": 0, "majorrel": 2, "minorrel": 6, "maintrel": 3, "revision": 20190608}} '
May 21 21:07:30 raspberrypi npm [28081]: zigbee2mqtt: debug 2020-05-21 21:07:30: Zigbee network parameters: {"panID": 6754, "extendedPanID": "0xdddddddddddddddd", "channel": 11}
May 21 21:07:30 raspberrypi npm [28081]: zigbee2mqtt: info 2020-05-21 21:07:30: Currently 20 devices are joined:

Zigbee2mqtt version: 1.13.1 (commit #55e6283)
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-05-22:
Is it still possible to control devices?

kwetnico, 2020-05-22:
> Est-il toujours possible de contrôler des appareils?

Hello, neither device answered.
I lowered a backup I had of my raspberry with zigbemqtt in version 1.21 and there I was able to re-pair all my devices.
But as soon as I migrate to 1.13.1 (I had no problem in 1.13) my routers have the LQI message suddenly I lose the associated peripherals.

kwetnico, 2020-05-22:
i'm not alone : 
[https://github.com/Koenkk/zigbee2mqtt/issues/2348](url)

Koenkk, 2020-05-22:
Can you provide a herdsman debug log on zigbee2mqtt 1.13.0 and 1.13.1?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

kwetnico, 2020-05-22:
> Can you provide a herdsman debug log on zigbee2mqtt 1.13.0 and 1.13.1?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

unfortunately not because since I migrated to 1.3.1 to retest it no longer worked, so I restore my raspberry backup and it is 1.2.2.
zigbee2mqtt@1.12.2 start / opt / zigbee2mqtt

Koenkk, 2020-05-22:
Good that it works now. But in order to further debug this I need these logs.

kwetnico, 2020-05-22:
No don't works when i have migrate to 1.3.1 again not works..

can you help me ? :(
I don't understand what has happened since the migration to 1.3.1, nothing is stable anymore.
I just restored in 1.2.2 and nothing happens when I want to pair a device ... I reflash my CC2531 key to be blank and reappear cleanly but nothing happens ..
Debug log:

pi@raspberrypi:/opt/zigbee2mqtt/data $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.12.2 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-05-22 17:33:35: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-05-22.17-33-35' filename: log.txt
zigbee2mqtt:debug 2020-05-22 17:33:35: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-04-25.10-52-47'
zigbee2mqtt:debug 2020-05-22 17:33:35: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-05-22 17:33:35: Starting zigbee2mqtt version 1.12.2 (commit #7d27a54)
zigbee2mqtt:info  2020-05-22 17:33:35: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-05-22 17:33:35: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +11ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +1ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,1,26] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,1,26] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,1] - 26 +3ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":377} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +12ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +36ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +30ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,144,21,52,1,2,0,0,0,0] - 218 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608} +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}) +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +26ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +26ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +34ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +37ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +25ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +24ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +22ms
[... cut ...]
zigbee2mqtt:info  2020-05-22 17:33:37: MQTT publish: topic 'zigbee2mqtt/Presence_Salon', payload '{"battery":100,"voltage":3055,"illuminance":539,"illuminance_lux":539,"linkquality":47,"occupancy":true}'
zigbee2mqtt:info  2020-05-22 17:33:37: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.2","commit":"7d27a54","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'

Koenkk, 2020-05-22:
I don't understand you. In 1.12.2 everything works fine?

kwetnico, 2020-05-22:
> I don't understand you. In 1.12.2 everything works fine?

sorry for my English.
This morning I restored in 1.2.2 the peripherals were able to reassociate but to give you elements to debug I updated and I immediately had error messages with LQI ...
So I put all my backup (raspberry) with version 1.2.2 + reflash the CC2531 key to leave prore and since I can not add anything, when I press to re-pair nothing happens.
I have my backup of the files in the data folder.

kwetnico, 2020-05-22:
I managed to reintegrate my peripherals on my CC2531 coordinator but impossible to add my siren and my xiami socket always LQI message .... I came back to a backup of 1 month back .. I did not activate anything LQI or whatever I don't understand ..

kwetnico, 2020-05-23:
hello, this morning I decided to start from zero, I reflashed my cc2531 key and deleted the 3 files:
configuration.yaml, database and json.

I re-paired all my devices, changed the 2.4 GHz wifi channel (from 1 to 8), moved my raspberry pi 4 upstairs with a USB extension cable for the 3 meter cc2531.

I was able to re-pair everything but 1 in 3 shots I have my router (xiaomi socket) which tells me a LQI message failed. I have to unplug it and plug it back in and it works again. If I stop zigbeemqtt idem, it is necessary to disconnect again ..

Assessment I think that from a certain number of component this key is limited it is not possible otherwise.

jwillmer, 2020-05-26:
I have a similar issue. since the upgrade one of my sensors loses contact every other day. a restart of the gateway fixes it so I think it has nothing to do with the sensor - it is also the sensor with the best battery live. 
I will also try to revert back to make it stable again

closed 2020-07-26 by stale[bot]