Koenkk/zigbee2mqtt #3587: Probelm adding device to zigbee2mqtt in hassos - accessing the file

bartlebyxxx, 2020-05-22:
Hello,
I hope I'm posting at the right location, haven't had much github interaction yet.

I have a little porblem adding a device to zigbee2mqtt. I'm running homeassistant on a pi running hassos. There is a " [Tutorial](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html) " on how I am suppost to add stuff but i can't access node_modules/zigbee-herdsman-converters/devices.js via ssh. But i do belive that i need to follow the docker tutorial anyways, problem there is that the page in step 1 is not there anymore so i don't know what's going on there. entering docker into the comand line does not work.

The device I wan't to add is a Switch 4x EU-LIGHTIFY already in an "[official](https://github.com/Koenkk/zigbee-herdsman-converters/commit/812a549ae253c37065843561cf540f6f3af531ae)" file, so i would just need to copy these Files . The item is not yet in the [dev Branch](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js) so i guess i need to do this manually.

I already added plugs, lights and the switch to the dongle but the switch is as expected listed as not supported.

If anybody could point me in the right direction to get those files into the addon that would be realy great. I've been fighting with my lightswitches for the past 2 weeks and feel like the solution is right there I'm just missing the IT skills.

Big thanks allready to everybody trying to help

Koenkk, 2020-05-22:
Link has been fixed (https://developers.home-assistant.io/docs/operating-system/debugging/#ssh-access-to-the-host)

bartlebyxxx, 2020-05-24:
Thanks so much for the support, i did get the switch working but now I'm stuck on the Part:

> Be aware that changes are not persistent, any changes that recreate the Docker container will result in the changes being lost, so make sure your modifications are provided back to the project for out-of-the-box support. Also using the Restart button from the web interface will remove your changes.

Is there a way to keep the edits permanently or will this have to be updated in the addon? Cause I will have to restart hassio eventually.

Koenkk, 2020-05-24:
Yes, by creating a PR or posting the code here so that it can officially be supported

closed 2020-07-26 by stale[bot]