Koenkk/zigbee2mqtt #3588: Zigbee2mqtt not using system proxy setting?

aforch, 2020-05-22:
Version of Zigbee2Mqtt: 1.13.0

I have my RasPi running z2m on an IoT network that is isolated from both the internet and my LAN. I have a squid proxy configured to allow access to whitelisted sites for updating hosts in the IoT network. I can therefore update raspbian, I can run the z2m update script, etc.... but when checking for OTA updates for my devices, I never see any requests in the squid logs and the OTA request times out. 
The proxy is set via system environment variables:
declare -x http_proxy="http://192.168.x.xxx:3128"
declare -x https_proxy="http://192.168.x.xxx:3128"

I have also whitelisted the 2 IP addresses that I have found in the z2m logs:
40.113.128.107
151.101.36.133

This was obviously never an issue until the introduction of the feature for OTA updates. Can support for system proxy be added to allow OTA updates in this case?

Koenkk, 2020-05-22:
See https://jjasonclark.com/how-to-setup-node-behind-web-proxy/

aforch, 2020-05-23:
I have added the proxy and https-proxy per the above how-to, and now have this in my ~/.npmrc file:
proxy=http://192.168.x.xxx:3128/
https-proxy=http://192.168.x.xxx:3128/

I restarted the z2m service and tested, and did not see any traffic reaching squid. I then rebooted the RasPi and tested again, and still do not see any traffic reaching squid. 
I then installed tcpdump and dumped all traffic to the squid host and do not see anything leaving the RasPi.

I'm not even remotely an expert on NPM, but all of the other articles that I've found for this reference this as the solution for "NPM install" behind a proxy, but I *think* maybe the applications themselves must be configured to make use of the proxy with something like proxy agent:
https://github.com/TooTallNate/node-proxy-agent

Koenkk, 2020-05-23:
This indeed seems to be required.

Can you try again with the latest dev branch (in +-3  hours from now)? You should be able to set the proxy settings by settings `HTTPS_PROXY` environment variable.

aforch, 2020-05-24:
I was able to test this morning. I have updated to 1.13.1.
As of now I still do not see any traffic created when initiating an OTA check.
Confirmed proxy environment variables are still set to:
declare -x http_proxy="http://192.168.x.xxx:3128"
declare -x https_proxy="http://192.168.x.xxx:3128"
declare -x no_proxy=" localhost, 127.0.0.1"

Koenkk, 2020-05-24:
`https_proxy` is not correct, should be `HTTPS_PROXY` (capitals)

aforch, 2020-05-24:
Added 
declare -x HTTPS_PROXY="http://192.168.x.xxx:3128"
to environment variables and rebooted. Testing again still does not show any traffic generated per tcpdump.
As a test of existing environment variables I have done "wget http://54.192.87.121/file.txt" which is a non-existant file, but is one of the previously used IP addresses for updating and therefore has been whitelisted. This traffic reaches squid as expected:
1590318842.999     34 192.168.3.9 TCP_MISS/403 1377 GET http://54.192.87.121/file.txt - HIER_DIRECT/54.192.87.121 text/html

Koenkk, 2020-05-24:
Are you sure an OTA request was done by the device?

Can you try to manually trigger an OTA update check for a device?  (https://www.zigbee2mqtt.io/information/ota_updates.html#check-if-an-update-is-available)

aforch, 2020-05-24:
I perform the update checks manually via MQTT Explorer.
This is the log output
```
info  2020-05-24 17:48:27: Checking if update available for 'Stairs'
info  2020-05-24 17:48:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Checking if update available for 'Stairs'","meta":{"status":"checking_if_available","device":"Stairs"}}'
error 2020-05-24 17:49:01: Failed to check if update available for 'Stairs' (connect ETIMEDOUT 52.222.167.160:443)
info  2020-05-24 17:49:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Failed to check if update available for 'Stairs' (connect ETIMEDOUT 52.222.167.160:443)","meta":{"status":"check_failed","device":"Stairs"}}'
```

I am no longer on the dev branch as Zigbee-herdsman began failing to start (though the resolution was to reflash my cc-2530 board for some reason) but I can do additional testing if desired.

Koenkk, 2020-05-25:
As I cannot test this locally we need to do some debugging on your system.

Can you change https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/ota/common.js#L316 

To:

```js
function getAxios() {
    let config = {};
    const proxy = process.env.HTTPS_PROXY;
    if (proxy) {
        console.log('\n\n USING PROXY\n\n' + proxy + '\n\n');
        config = {
            proxy: false,
            httpsAgent: new HttpsProxyAgent(proxy),
        };
    } else {
       console.log('\n\nNOT USING PROXY\n\n');
    }

    const axios = require('axios').create(config);
    return axios;
}
```

You should see this logging during startup of zigbee2mqtt. The file can be found under `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/ota/common.js`

aforch, 2020-05-25:
Sure thing, that resulted in:

```
npm start

> zigbee2mqtt@1.13.1-dev start /opt/zigbee2mqtt
> node index.js

 USING PROXY
http://192.168.1.152:3128

 USING PROXY
http://192.168.1.152:3128

 USING PROXY
http://192.168.1.152:3128

 USING PROXY
http://192.168.1.152:3128

 USING PROXY
http://192.168.1.152:3128

```

Running an OTA check with debug on resulted in
```
debug 2020-05-25 22:06:31: Received MQTT message on 'zigbee2mqtt/bridge/ota_update/check' with data '0x90fd9ffffed7232e'
info  2020-05-25 22:06:31: Checking if update available for 'Stairs'
info  2020-05-25 22:06:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Checking if update available for 'Stairs'","meta":{"status":"checking_if_available","device":"Stairs"}}'
debug 2020-05-25 22:06:31: Check if update available for '0x90fd9ffffed7232e' (TRADFRI bulb E14 WS opal 400lm)
debug 2020-05-25 22:06:31: Using endpoint '1'
debug 2020-05-25 22:06:33: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
debug 2020-05-25 22:06:33: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}'
debug 2020-05-25 22:06:36: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
debug 2020-05-25 22:06:39: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
debug 2020-05-25 22:06:42: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
debug 2020-05-25 22:06:45: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
error 2020-05-25 22:07:06: Failed to check if update available for 'Stairs' (connect ETIMEDOUT 143.204.89.94:443)
info  2020-05-25 22:07:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Failed to check if update available for 'Stairs' (connect ETIMEDOUT 143.204.89.94:443)","meta":{"status":"check_failed","device":"Stairs"}}'

```

The other interesting thing I found was this log entry
```
error 2020-05-25 21:57:22: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: connect ETIMEDOUT 151.101.192.133:443
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1134:16))
```

Koenkk, 2020-05-26:
Is this the same error when not defining a proxy?

aforch, 2020-05-26:
The error is the same, yes
```
npm start

> zigbee2mqtt@1.13.1-dev start /opt/zigbee2mqtt
> node index.js

NOT USING PROXY

NOT USING PROXY

NOT USING PROXY

NOT USING PROXY

NOT USING PROXY
```
```
debug 2020-05-26 18:38:43: Received MQTT message on 'zigbee2mqtt/bridge/ota_update/check' with data '0x90fd9ffffed7232e'
info  2020-05-26 18:38:43: Checking if update available for 'Stairs'
info  2020-05-26 18:38:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Checking if update available for 'Stairs'","meta":{"status":"checki
ng_if_available","device":"Stairs"}}'
info  2020-05-26 18:38:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Checking if update available for 'Stairs'","meta":{"status":"checking_if_available","device":"Stairs"}}'
debug 2020-05-26 18:38:43: Check if update available for '0x90fd9ffffed7232e' (TRADFRI bulb E14 WS opal 400lm)
debug 2020-05-26 18:38:43: Using endpoint '1'
debug 2020-05-26 18:38:45: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
debug 2020-05-26 18:38:45: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}'
debug 2020-05-26 18:38:48: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
debug 2020-05-26 18:38:51: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
debug 2020-05-26 18:38:54: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
debug 2020-05-26 18:38:57: Received Zigbee message from 'Stairs', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}' from endpoint 1 with groupID 0
error 2020-05-26 18:39:18: Failed to check if update available for 'Stairs' (connect ETIMEDOUT 99.86.116.47:443)
info  2020-05-26 18:39:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Failed to check if update available for 'Stairs' (connect ETIMEDOUT 99.86.116.47:443)","meta":{"status":"check_failed","device":"Stairs"}}'

```
These seem to be recurring periodically without being prompted by an update request
```
error 2020-05-26 18:48:32: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: connect ETIMEDOUT 40.113.128.107:443
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1134:16))
error 2020-05-26 18:54:27: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: connect ETIMEDOUT 151.101.128.133:443
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1134:16))
error 2020-05-26 19:25:47: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: connect ETIMEDOUT 151.101.64.133:443
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1134:16))
```

Koenkk, 2020-05-26:
First execute `npm install tunnel` in `/opt/zigbee2mqtt` after that change the function to

```js
function getAxios() {
    let config = {};
    const proxy = process.env.HTTPS_PROXY;
    if (proxy) {
        console.log('\n\n USING PROXY\n\n' + proxy + '\n\n');
        const tunnel = require('tunnel');
        const agent = tunnel.httpsOverHttp({
            proxy: {
               host: 'http://192.168.1.152',
               port: 3128,
               proxyAuth: '',
             },
            rejectUnauthorized: false,
        });

        config = {
            proxy: false,
            httpsAgent: agent,
        };
    } else {
       console.log('\n\nNOT USING PROXY\n\n');
    }

    const axios = require('axios').create(config);
    return axios;
}
```

aforch, 2020-05-26:
```
zigbee2mqtt:debug 2020-05-26 21:29:44: Received MQTT message on 'zigbee2mqtt/bridge/ota_update/check' with data '0x0017880103224b35'
zigbee2mqtt:info  2020-05-26 21:29:44: Checking if update available for 'Corner'
zigbee2mqtt:info  2020-05-26 21:29:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Checking if update available for 'Corner'","meta":{"status":"checking_if_available","device":"Corner"}}'
zigbee2mqtt:debug 2020-05-26 21:29:44: Check if update available for '0x0017880103224b35' (LCT015)
zigbee2mqtt:debug 2020-05-26 21:29:44: Using endpoint '11'
zigbee2mqtt:debug 2020-05-26 21:29:45: Received Zigbee message from 'Corner', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16781568}' from endpoint 11 with groupID 0
zigbee2mqtt:debug 2020-05-26 21:29:45: Got OTA request '{"fieldControl":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16781568}'
zigbee2mqtt:error 2020-05-26 21:29:45: Failed to check if update available for 'Corner' (tunneling socket could not be established, cause=getaddrinfo ENOTFOUND http://192.168.1.152)
zigbee2mqtt:info  2020-05-26 21:29:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Failed to check if update available for 'Corner' (tunneling socket could not be established, cause=getaddrinfo ENOTFOUND http://192.168.1.152)","meta":{"status":"check_failed","device":"Corner"}}'
zigbee2mqtt:debug 2020-05-26 21:29:55: Received Zigbee message from 'Corner', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16781568}' from endpoint 11 with groupID 0
zigbee2mqtt:debug 2020-05-26 21:29:55: Check if update available for '0x0017880103224b35' (LCT015)
zigbee2mqtt:error 2020-05-26 21:29:55: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: tunneling socket could not be established, cause=getaddrinfo ENOTFOUND http://192.168.1.152
    at ClientRequest.onError (/opt/zigbee2mqtt/node_modules/tunnel/lib/tunnel.js:180:17)
    at Object.onceWrapper (events.js:313:26)
    at ClientRequest.emit (events.js:223:5)
    at Socket.socketErrorListener (_http_client.js:406:9)
    at Socket.emit (events.js:223:5)
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:81:21))
```

Koenkk, 2020-05-27:
Are you sure the provided proxy information is correct? It seems to fail to connect to it.

Is your proxy password protected?

```
tunneling socket could not be established, cause=getaddrinfo ENOTFOUND http://192.168.1.152
```

aforch, 2020-05-27:
Below shows a test that I just did using wget. The https_proxy environment variable was initially not set, therefore it tried to directly connect. 
I then set the environment variable and re-ran the command.

```
pi@raspberrypi:~ $ wget https://raw.githubusercontent.com/Koenkk/zigbee2mqtt/master/README.md
--2020-05-27 16:19:45--  https://raw.githubusercontent.com/Koenkk/zigbee2mqtt/master/README.md
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... ^C
```
```
pi@raspberrypi:~ $ export https_proxy="http://192.168.1.152:3128"
```
```
pi@raspberrypi:~ $ wget https://raw.githubusercontent.com/Koenkk/zigbee2mqtt/master/README.md
--2020-05-27 16:20:14--  https://raw.githubusercontent.com/Koenkk/zigbee2mqtt/master/README.md
Connecting to 192.168.1.152:3128... connected.
Proxy request sent, awaiting response... 200 OK
Length: 4383 (4.3K) [text/plain]
Saving to: ‘README.md.2’

README.md.2                                   100%[==============================================================================================>]   4.28K  --.-KB/s    in 0.002s  

2020-05-27 16:20:15 (2.43 MB/s) - ‘README.md.2’ saved [4383/4383]
```

aforch, 2020-05-27:
One other note, when testing above last night I was dumping the interface with:
```
sudo tcpdump host 192.168.1.152 and not tcp port 1883
```
I saw no traffic at all other than a host alive ping test from HA/NodeRed running on the same host as the proxy.

Koenkk, 2020-05-27:
It's a bit hard for me to continue with this as I don't know what the exact problem is. 

How can I setup a small test environment myself?

aforch, 2020-05-27:
I'm running squid in docker, here is my docker-compose for it
```
  squid:
    container_name: squid
    image: sameersbn/squid:3.5.27-2
    restart: unless-stopped
    ports:
      - 3128:3128
    volumes:
      - /mnt/local_storage/squid:/etc/squid
      - /mnt/local_storage/squid/cache:/usr/local/var/cache
    environment:
      - TZ=Europe/Amsterdam
      - PUID=xxx
      - PGID=xxx
```

After that you just need to allow hosts other than localhost to connect
https://www.pks.mpg.de/~mueller/docs/suse10.2/html/opensuse-manual_en/manual/sec.squid.configfile.html

Specifically these 2 things need to be set
```
acl localnet src 192.168.x.x/24 # RFC1918 possible internal network
http_access allow localnet
```
You can view the access log by exec'ing a bash shell in the container and tailing /var/log/squid/access.log

closed 2020-07-26 by stale[bot]