Koenkk/zigbee2mqtt #3590: Lamps can not be switched via MQTT

probutus, 2020-05-22:
I'm on latest 1.13.1 branch and have a ikea tradfri 900lm  and a philips hue e27 color lamp connected. Both work flawlessly with the original hue 2.1. bridge.

I paired both lamps with zigbee2mqtt and now I try to switch the lamps via MQTT (homeassistant is OFF; I'm using plain MQTT)

Independently what I send to zigbee2mqtt the lamp is switched off and its brightness is set to 0. This behaviour is the same with the ikea and the hue Lamp. I found a similar post where it was written that it should work exactly this way hence I'm filing a bug report
I sent these commands (initially, I switched the lamp manually on)
```

tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":"ON"\,"brightness":"255"}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":\ "ON"}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":\ "ON"}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":\ "ON"}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":\ "ON"}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":"ON"}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":"TOGGLE"}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":"OFF"}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":"OFF"\,"brightness":100}
tux@Darkstar:~$ mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {"state":"ON"\,"brightness":100}
```

Debug Log is here:
```
> zigbee2mqtt@1.13.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-05-22 17:34:54: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-05-22.17-34-54' filename: log.txt
zigbee2mqtt:debug 2020-05-22 17:34:54: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-05-21.21-02-42'
zigbee2mqtt:debug 2020-05-22 17:34:54: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-05-22 17:34:54: Starting zigbee2mqtt version 1.13.1 (commit #3046680)
zigbee2mqtt:info  2020-05-22 17:34:54: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-05-22 17:34:54: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"},"adapter":{"concurrent":null}}'
zigbee2mqtt:info  2020-05-22 17:34:56: zigbee-herdsman started
zigbee2mqtt:info  2020-05-22 17:34:56: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:debug 2020-05-22 17:34:56: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2020-05-22 17:34:56: Currently 6 devices are joined:
zigbee2mqtt:info  2020-05-22 17:34:56: tradfri_remote (0x90fd9ffffe71ad17): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
zigbee2mqtt:info  2020-05-22 17:34:56: 0x000b57fffeec690b (0x000b57fffeec690b): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
zigbee2mqtt:info  2020-05-22 17:34:56: cc2531_repeater (0x00124b000e0eacb6): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
zigbee2mqtt:info  2020-05-22 17:34:56: paulmann_remote (0x00158d0001c0b6be): 500.67 - Paulmann RGB remote control (EndDevice)
zigbee2mqtt:info  2020-05-22 17:34:56: hue_farblampe_ez (0x0017880108b16e4b): 9290022166 - Philips Hue white and color ambiance E26/E27 (Router)
zigbee2mqtt:info  2020-05-22 17:34:56: tradfri_900lm_ez (0x90fd9ffffe4cfea3): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Router)
zigbee2mqtt:info  2020-05-22 17:34:56: Zigbee: disabling joining new devices.
zigbee2mqtt:info  2020-05-22 17:34:56: Connecting to MQTT server at mqtt://andromeda.fritz.box
zigbee2mqtt:info  2020-05-22 17:34:56: Connected to MQTT server
zigbee2mqtt:info  2020-05-22 17:34:56: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-05-22 17:34:56: MQTT publish: topic 'zigbee2mqtt/tradfri_remote', payload '{"update_available":true,"linkquality":0,"battery":47}'
zigbee2mqtt:info  2020-05-22 17:34:56: MQTT publish: topic 'zigbee2mqtt/0x000b57fffeec690b', payload '{"update_available":false}'
zigbee2mqtt:info  2020-05-22 17:34:56: MQTT publish: topic 'zigbee2mqtt/cc2531_repeater', payload '{"state":true,"led_state":false,"linkquality":89,"description":"37516/0x000B57FFFEEC690B","type":"NEIGHBOR","rssi":46,"led":false}'
zigbee2mqtt:info  2020-05-22 17:34:56: MQTT publish: topic 'zigbee2mqtt/paulmann_remote', payload '{"linkquality":47}'
zigbee2mqtt:info  2020-05-22 17:34:56: MQTT publish: topic 'zigbee2mqtt/hue_farblampe_ez', payload '{"state":"OFF","brightness":0,"linkquality":0,"update_available":false}'
zigbee2mqtt:info  2020-05-22 17:34:56: MQTT publish: topic 'zigbee2mqtt/tradfri_900lm_ez', payload '{"update_available":true,"state":"OFF","brightness":0}'
zigbee2mqtt:info  2020-05-22 17:34:56: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.13.1","commit":"3046680","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}'

zigbee2mqtt:debug 2020-05-22 17:35:15: Received MQTT message on 'zigbee2mqtt/tradfri_900lm_ez/set' with data '{state:ON}'
zigbee2mqtt:debug 2020-05-22 17:35:15: Publishing 'set' 'state' to 'tradfri_900lm_ez'
zigbee2mqtt:info  2020-05-22 17:35:15: MQTT publish: topic 'zigbee2mqtt/tradfri_900lm_ez', payload '{"update_available":true,"state":"OFF","brightness":0}'

zigbee2mqtt:debug 2020-05-22 17:35:31: Received MQTT message on 'zigbee2mqtt/tradfri_900lm_ez/set' with data '{state:TOGGLE}'
zigbee2mqtt:debug 2020-05-22 17:35:31: Publishing 'set' 'state' to 'tradfri_900lm_ez'
zigbee2mqtt:info  2020-05-22 17:35:31: MQTT publish: topic 'zigbee2mqtt/tradfri_900lm_ez', payload '{"update_available":true,"state":"OFF","brightness":0}'

zigbee2mqtt:debug 2020-05-22 17:35:39: Received MQTT message on 'zigbee2mqtt/tradfri_900lm_ez/set' with data '{state:OFF}'
zigbee2mqtt:debug 2020-05-22 17:35:39: Publishing 'set' 'state' to 'tradfri_900lm_ez'
zigbee2mqtt:info  2020-05-22 17:35:39: MQTT publish: topic 'zigbee2mqtt/tradfri_900lm_ez', payload '{"update_available":true,"state":"OFF","brightness":0}'
zigbee2mqtt:debug 2020-05-22 17:35:56: Received MQTT message on 'zigbee2mqtt/tradfri_900lm_ez/set' with data '{state:OFF,brightness:100}'
zigbee2mqtt:debug 2020-05-22 17:35:56: Publishing 'set' 'state' to 'tradfri_900lm_ez'
zigbee2mqtt:info  2020-05-22 17:35:56: MQTT publish: topic 'zigbee2mqtt/tradfri_900lm_ez', payload '{"update_available":true,"state":"OFF","brightness":0}'
zigbee2mqtt:debug 2020-05-22 17:36:09: Received MQTT message on 'zigbee2mqtt/tradfri_900lm_ez/set' with data '{state:ON,brightness:100}'
zigbee2mqtt:debug 2020-05-22 17:36:09: Publishing 'set' 'state' to 'tradfri_900lm_ez'
zigbee2mqtt:info  2020-05-22 17:36:09: MQTT publish: topic 'zigbee2mqtt/tradfri_900lm_ez', payload '{"update_available":true,"state":"OFF","brightness":0}'
```

Koenkk, 2020-05-22:
The payloads you send are not valid JSON, e.g. `{state:ON}` should be `{"state": "ON"}`. Try with `mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m '{"state": "ON"}'`

probutus, 2020-05-22:
you' re right. I got tricked by the BASH

This works:
mosquitto_pub -h andromeda.fritz.box -p 1883 -t zigbee2mqtt/tradfri_900lm_ez/set -m {\"state\":\"ON\"\,\"brightness\":255}

closed 2020-05-22 by probutus