Koenkk/zigbee2mqtt #3594: Supporting the HOMA1001

kalsan, 2020-05-22:
After letting a few months pass, I decided to give it another go for the HOMA1001. It is the same device as referenced in #2555 and #2655 and thanks to @Koenkk's fix to the herdsman repo, it can now be supported by following the "How to support new devices" guide.

This time, I got as far as being able to turn the device on and off and change the intensity. But wait, there's more! HOMA1001 is supposed to support RGB and even RGBW LED stripes, thus this discussion goes further than #3109. And this is where I'm stuck once again.

Here's the code I added to devices.js:

```
{
        zigbeeModel: ['HOMA1001'],
        model: 'HLC610-Z', // I'm not sure about this!
        vendor: 'Shenzhen Homa',
        description: 'Wireless dimmable controller',
        extend: generic.light_onoff_brightness_colorxy_colortemp,
    },
```

The only way to change the color is by setting color_temp. Around 50, the color is pink. By increasing the value, it goes to white and ends up at green with values around 800.

When trying to send "color" messages (no matter if xy or rgb), the device does nothing. It reports weird messages as shown in the following example:

What I send:

```
{"color": {"x": 0.999, "y": 0.999}}
```

What the device does: Nothing

What the log says:

```
zigbee2mqtt_1  | zigbee2mqtt:info  2020-05-22 21:38:21: MQTT publish: topic 'homeassistant/light/0x00124b001eb08d44/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/stairs_lamp/set","state_topic":"zigbee2mqtt/stairs_lamp","json_attributes_topic":"zigbee2mqtt/stairs_lamp","name":"stairs_lamp_light","unique_id":"0x00124b001eb08d44_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001eb08d44"],"name":"stairs_lamp","sw_version":"Zigbee2mqtt 1.13.0-dev","model":"Wireless dimmable controller (HLC610-Z)","manufacturer":"Shenzhen Homa"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt_1  | zigbee2mqtt:info  2020-05-22 21:38:21: MQTT publish: topic 'homeassistant/sensor/0x00124b001eb08d44/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/stairs_lamp","json_attributes_topic":"zigbee2mqtt/stairs_lamp","name":"stairs_lamp_linkquality","unique_id":"0x00124b001eb08d44_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001eb08d44"],"name":"stairs_lamp","sw_version":"Zigbee2mqtt 1.13.0-dev","model":"Wireless dimmable controller (HLC610-Z)","manufacturer":"Shenzhen Homa"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-22 21:38:27: Received MQTT message on 'zigbee2mqtt/stairs_lamp/set' with data '{"color": {"x": 0.999, "y": 0.999}}'
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-22 21:38:27: Skipping state because of Home Assistant
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-22 21:38:27: Publishing 'set' 'color' to 'stairs_lamp'
zigbee2mqtt_1  | zigbee2mqtt:info  2020-05-22 21:38:27: MQTT publish: topic 'zigbee2mqtt/stairs_lamp', payload '{"state":"ON","brightness":255,"color_temp":370,"color":{"x":0.999,"y":0.999}}'
```

Another example:

I send:

`{"color": { "rgb": "200,0,0"}}`

The logs say:

```
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-22 21:40:39: Received MQTT message on 'zigbee2mqtt/stairs_lamp/set' with data '{"color": { "rgb": "200,0,0"}}'
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-22 21:40:39: Skipping state because of Home Assistant
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-22 21:40:39: Publishing 'set' 'color' to 'stairs_lamp'
zigbee2mqtt_1  | zigbee2mqtt:info  2020-05-22 21:40:39: MQTT publish: topic 'zigbee2mqtt/stairs_lamp', payload '{"state":"ON","brightness":255,"color_temp":37,"color":{"x":0.7006,"y":0.2993}}'
```

Actual color: still green.

So I'm officially stuck here. Any ideas on how to develop this any further? Help is much appreciated!

Koenkk, 2020-05-22:
Can you share the `database.db` entry of this device?

kalsan, 2020-05-23:
Thanks for coming back so quickly. Here's the line:

`{"id":39,"type":"Router","ieeeAddr":"0x00124b001eb08d44","nwkAddr":32888,"manufId":0,"manufName":"ShenZhen_Homa","powerSource":"Mains (single phase)","modelId":"HOMA1001","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"HOMA1001","manufacturerName":"ShenZhen_Homa","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2"}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590176606021}`

Koenkk, 2020-05-23:
Very strange that it doesn't work, not sure how to fix it. Do you have a bridge where it works?

kalsan, 2020-05-24:
Unfortunately I don't have anything other than zigbee2mqtt. The device exists in Europe, I could try sending one to the Netherlands if it helps

kalsan, 2020-07-19:
I didn't get any smarter about this. Unless someone has a good idea, I'll have to discard and replace the device.

MattWestb, 2020-07-19:
Hope you finding one working solution for this product.
Have looking on it more times and finding interesting (I having some HOMA LED Dimmers and LED drives running im my systems).
The software problem i see its its acting in different ways on the zigbee side if you changing the hardware settings but it's also making the product very flexible and more interesting.

kalsan, 2020-07-19:
@MattWestb which model and modes are you using exactly? Maybe my device is buggy or my connections are bad

MattWestb, 2020-07-19:
I don't have any HOMA 1001 was only looking at it and wos not finding any support for it so i was not buying one.
The dimmers its model HOMA 1031 and the LED drive its HOMA1008 and i also have one 3 channel on /off but its not in the network for the moment.
(all names its from the firmware names not wot its written outside)
The 1008 and 1031 its working with Ikeas GW and deCONZ but i have not trying them with Z2M but its coming very soon.

kalsan, 2020-07-19:
Thanks for your feedback! I think I might just replace the 1001 by another model.

MattWestb, 2020-07-19:
That's sad because its not easy finding good RGB(W) drives and this one its possible using as singer channels or multiple WS LEDs and RGB/RGBW.
Have looking very long for something better then my ESPHome that driving my RGB stripes.
One way i was thinkin trying off its using one module (ICC-A-1) from one ikea bulb and loading the firmware for one ikea RGBW bulb. Its easy done if having the firmware (can being extracted for the OTA file) and the user data block from one bulb (its the configuration file not a part of the main firmware). The first i have the second i need killing my one and only RGBW bulb that i don't have done.

TorainX, 2020-11-30:
> Thanks for your feedback! I think I might just replace the 1001 by another model.

Hi, I bought those devices as well and now came to the point that they were not integrated in Z2M already...
Have you achieved any progress or did you buy other devices and if which one?
I'm currently stuck (as I'm really new to that all) at that point: copied your config into the devices.js.
I'm running on Z2M on my home assistant Raspberry..
I'm not even able to the led stripe ON or OFF.
Some help would be really appreciated! :-)

kalsan, 2020-12-02:
I still haven't given up completely but I haven't made any progress either. So my last post from July is still as up to date as it got in the meantime ;-)
What I have done: I followed the guide for supporting a new device and copied from similar devices until something worked. Got stuck there and no progress ever since.

TorainX, 2020-12-06:
I opened a new ticket asking again for help...
https://github.com/Koenkk/zigbee2mqtt/issues/5211
Sorry for coming back with delay, was quite busy the last days...

Now I'm really stuck with checking all those dependencies to those special treatments, regarding the different vendors in /converters/toZigbee.js
This is where I have to state: Unfortunately I'm not a developer :-(
I hope they could help us with that device...

Koenkk, 2020-12-11:
I see @kalsan tried the standard Zigbee way to control this device. Seems it uses some special protocol. 

@TorainX In case you want to continue with this you have to sniff the traffic with the original gateway (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

TorainX, 2020-12-12:
Hi @Koenkk,
first of all thanks for your support here. Just for clarification for me:
- I will remove my CC2531 from Raspberry Hassio 
(Correct Sniffer FW should be already installed on CC2531 as it is already working with some Osram lights, Tasmota devices, etc., right? I used that one: CC2531_DEFAULT_20190608.zip)

- Plug CC2531 into my windows machine
- download ZBOSS (configuring all the encryption stuff...)
- install Wireshark

- then need to connect a hue bridge or similar to HOMA1001 and hope it's working 
- then sniffing all the commands while playing around with the different light settings over the HUE bridge

Questions:
Do I need to flash a different FW to CC2531?
I have a second CC2531 with the same FW installed. 
Does it make more sense to add Hue to the existing network and use the second CC2531 for the sniffing actions? Because otherwise I'm not sure how to figure out the encryption keys from the closed system:
 Hue Bridge-->HOMA1001-->CC2513 (in the Windows machine... how to configure the encyrptions etc.?)

So better?:

Hassio-->CC2531-->Hue-->HOMA1001-->2nd CC2531 (using the encryption config of Z2M of Hassio)

Koenkk, 2020-12-13:
You need a second CC2531 and set it up accoridng to https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

TorainX, 2020-12-13:
+ Hue bridge or Tradfri or so, right?
(because I have to buy one if needed)

Koenkk, 2020-12-13:
@TorainX I recommend buying the original gateway (where the vendor advertises it work with). I don't expect this device to work with Hue since we should already use the same configuration.

TorainX, 2020-12-13:
This pictures below how it is advertised.
Unfortunatley in German...
There is the option with some DIPs to differentiate the function, dependad to the LED stripe which will be connected.
Could this cause the problems?

[image]
[image]
[image]
[image]

Koenkk, 2020-12-13:
@TorainX I suggest try playing with those switches, the config in the OP (https://github.com/Koenkk/zigbee2mqtt/issues/3594#issue-623443503) should work.

TorainX, 2020-12-13:
(Un)fortunately I have those RGBW Stripes connected. So if I go for [D1=1+D2=0], then I would probably lose the function of configuring the separte White LED?
But even more important: Do you think something is going to be broken if I configure a "wrong" config that doesn't match the LED connected?

Koenkk, 2020-12-13:
@TorainX I have no clue (don't know this device)

TorainX, 2020-12-13:
OK. coming back to the sniffing part:
So will use an HUE2 and while doing all the different settings, parallel sniffing could bring light into dark?
So sharing those sniffing info with you would help?

In the meanwhile playing around with the differnt DIP settings (currently DIP=1:0; so RGB instead of RGBW)

Koenkk, 2020-12-13:
@TorainX if it works with Hue this can definitely help

TorainX, 2020-12-14:
Just ordered the Hue2..
In the meanwhile:
DIP switch is still on 1:0
now at least on/off is working and dimming. But the bright white LED is missing as expected.

config in devices.js is: (also tried with generic or ledvance-but no difference in the extend config)

> // Shenzhen Homa
    {
        zigbeeModel: ['HOMA1001'],
        model: 'HLC610-Z', // I'm not sure about this!
        vendor: 'Shenzhen Homa',
        description: 'Wireless dimmable controller',
        extend: hue.light_onoff_brightness_colortemp_colorxy,
    },

I'm also also using an OSRAM light (from the logfile)
> Zigbee2MQTT:info  2020-12-14 01:46:13: 0x7cb03eaa0a0058dc (0x7cb03eaa0a0058dc): 4052899926110 - OSRAM Flex RGBW (Router)
Zigbee2MQTT:info  2020-12-14 01:46:13: 0x00124b001eb0821f (0x00124b001eb0821f): HLC610-Z - Shenzhen Homa Wireless dimmable controller (Router)

any idea why the colour configuration is not showing up like it is for the OSRAM (in Home Assistant?

[image]
[image]

LeoS3, 2020-12-27:
I also have some of these controllers..  I bought the Ajaxonline.co.uk version:

[image]

They work well with Phoscon\deconz, and shows up as ShenZhen_Homa device with model HOMA1001 (version 1.0.2)
I think the OEM model is HLC502-Z-3.0 or HLC502-Z

I have mine jumpered as CC+CW device that worked well with deconz... but they showed up as Unsupported under zigbee2mqtt.

kalsan, 2020-12-29:
@LeoS3 the writing on the product is different but the shape and plasic parts are exactly the same, this looks like a HOMA1001 all right.

Do you folks think the Phoscon could give additional info? Perhaps they store config files in plain text...

LeoS3, 2020-12-30:
I will check...
Here's what I found from the sqlite of my deconz backup:

{"attr/id":"66","attr/lastannounced":null,"attr/lastseen":"2020-12-30T12:50Z","attr/manufacturername":"ShenZhen_Homa","attr/modelid":"HOMA1001","attr/name":"Living Room Light","attr/swversion":"1.0.2","attr/type":"Color temperature light","attr/uniqueid":"00:12:4b:00:xx:xx:xx:xx-xx","config/colorcapabilities":16,"config/ctmax":65279,"config/ctmin":null,"state/alert":null,"state/bri":192,"state/colormode":"ct","state/ct":500,"state/on":true,"state/reachable":true}

TorainX, 2021-01-05:
> @TorainX if it works with Hue this can definitely help. When sharing the sniff make also sure to share the network key.

Sorry took a while:
Did two sniffs: 
- One with joining the device to Hue Bridge (I think in that sniff I gathered the Network key from the screenshot)
- The other one with chaning colors and dimming the device via Hue App
*******************************************************
- Is it fine to post the full Network key here in public? (Not quite sure if this is branded in the Hue Bridge?/Is it a security risk to put it in public?)
 - Where should I upload/post the .pcapng files from Wireshark? (Directly here in this thread?)
- Do you need both or just the "play around" sniff?

[image]

closed 2021-02-13 by github-actions[bot]

gitjho, 2021-11-21:
I also have this device (ShenZhen HOMA1001). I tinkered a bit around with settings in a device file and made it work with this one:

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['HOMA1001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'HOMA1001', // Vendor model number, look on the device for a model number
    vendor: 'ShenZhen_Homa', // Vendor of the device (only used for documentation and startup logging)
    description: 'Smart LED driver', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: extend.light_onoff_brightness_color(),
    exposes: [e.light_brightness_colorxy().withEndpoint('rgb'), e.light_brightness().withEndpoint('white')], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    endpoint: (device) => {
       return {white: 10, rgb: 11};
       },
};
module.exports = definition;

The device has two endpoints; White is controlled by endpoint 10 and rgb is controlled by endpoint11. With the above config i am able to control the two endpoints independently in z2m frontend.

When the device joins i get this error:
Failed to read state of '0x1234123412341234' after reconnect (Read 0x1234123412341234/10 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
I have not figured out how to avoid this.
Maybe someone can use this - or even make it better.

Regards Jens

sv87411, 2021-12-12:
I know this is a closed thread, but I think a few people have these devices and it'd be great to get them working on Z2M.

I have 5 of these and have to control them via a Phillips Hue hub - all my other devices are via Z2M. I have to own and run the Hue hub just for these controllers. I have a major investment in Zigbee devices but I hate the lack of standardisation in the Zigbee world! (rant over)

I use the controller with an RGBW LED string and when paired with the Hue hub it offers the White and RGB channels separately which is exactly what I want and what @gitjho describes above.

I once went through the process of adding device support on Z2M, but that was for a very simple device; I suspect these will be more difficult because you can control how the controller behaves through the switches on it. I assume it will announce itself differently depending on how the switches are set.

It would be great to get these supported on Zigbee2Mqtt with the  white and RGB channels controllable separately.

sv87411, 2021-12-12:
I have just, I think (I'm not good with GIT, but I think I managed it) submitted an update for zigbee-herdsman-converters to add support for the HOMA1001. It's based on the code that @gitjho provided above.

I tested the code in an external converter file on my dev branch and it has worked for two of my HOMA1001 devices that are using RGBW strips. I didn't the errors that @gitjho listed above and my devices joined and configured OK.

The white and RGB strips present as independent endpoints and I can control on/off, brightness for both and the colour for the RGB strip as below:

[image]

I can't work out from the support docs how to add an image for the device, maybe someone else or @Koenkk can do this for me. The documentation for adding images isn't very clear. I didn't really know where to start.

Hopefully this helps someone. It means I can now hopefully get rid of my Hue Hub!!!

MrLeeches, 2022-05-20:
This is great and works apart from the colour changing on the rgb channel. 

I can control both on off states and brightness but the colour does not work.

I used the code above as standard. Any ideas?

LeoS3, 2022-06-12:
Oh man, does that work with WW+CW?

ukmgranger, 2022-11-30:
I also have one of these devices and I've just moved from Deconz over to Z2m.  The device worked perf ctly in Deconz, so I was surprised that it didn't in Z2m!!

I see someone above got it working, but not sure how / where to add their code.  Any help would be appreciated.

kapellmeister84, 2023-01-03:
@sv87411 Hey man, im trying to get this controller to work. Since there is no native support for it, can you please explain, how I can set it up myself? Would be awesome! thanks a lot