Koenkk/zigbee2mqtt #3596: Duplicate device IDs

schester44, 2020-05-22:
Had a CC2351 setup as a coordinator. Bought another CC2351, put an antenna on it, and made it the coordinator. I took the original coordinator and reflashed it to be a router... Now, the network map shows the router has having the same device ID as the new coordinator... 

Is there a way to force the router to use a new device ID?

Koenkk, 2020-05-23:
What device ID do you mean?

schester44, 2020-05-24:
Sorry, looking at the zigbee map, I see the coordinator has some sort of ID (0x00124b0018e2eb07).. My thought was that somehow the router and coordinator are sharing that same ID due to the way I went about configuring the network.

I'm probably off base with those assumptions but what made me think that was because the coordinator has an arrow pointing to itself when the router is plugged in and that arrow isn't there when the router is not plugged in.

Any idea what is going on?

Koenkk, 2020-05-24:
Is the ieeAddress of both devices the same? You can check this with the flash programmer utility -> Read IEEE. Screenshot: https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html#windows step 7

schester44, 2020-05-26:
Do I need the debugger to run that? I flashed these using a Pi

Koenkk, 2020-05-26:
I think so, it doesn't seem that cc_flash has an option to read the ieeeAddr (https://github.com/jmichault/flash_cc2531)

schester44, 2020-05-26:
Okay. Any idea what may be happening here so I can look into this a little more?

Koenkk, 2020-05-26:
I think the devices are flashed with the same ieeeAddr (which is causing confusion on the zigbee2mqtt side as the identification is done based on this, as ieeeAddr should always be unique).

I'm also assuming you are running the latest zigbee2mqtt (1.13.1), previously there was a bug causing the ieeeAdddr of the coordinator not to update when it changed.

schester44, 2020-05-26:
I pulled down the latest docker image and re-paired the router to the network.. I am seeing the router in the generated map now but there is still an arrow pointing from the coordinator back to itself.

Any idea what that could be? I think its safe to close this, I should have enough info to debug this later if needed.

Koenkk, 2020-05-26:
Could you share the graphviz networkmap?

schester44, 2020-05-27:
https://i.imgur.com/098Psxc.png

On an unrelated note, i'm not sure where the disconnected Xiaomi sensors are coming from either. I have 4 in the house but the map shows 6.

Koenkk, 2020-05-27:
At least the router has a different ieeeAddr which is good.

Can you provide the herdsman debug log when executing the network scan?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

schester44, 2020-05-27:
```
info  2020-05-27 13:19:51: Logging to console and directory: '/app/data/log/2020-05-27.13-19-51' filename: log.txt
debug 2020-05-27 13:19:51: Removing old log directory '/app/data/log/2020-05-22.13-19-24'
debug 2020-05-27 13:19:51: Loaded state from file /app/data/state.json
info  2020-05-27 13:19:51: Starting zigbee2mqtt version 1.13.1 (commit #3046680)
info  2020-05-27 13:19:51: Starting zigbee-herdsman...
debug 2020-05-27 13:19:51: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
info  2020-05-27 13:19:53: zigbee-herdsman started
info  2020-05-27 13:19:53: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
debug 2020-05-27 13:19:53: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-05-27 13:19:53: Currently 21 devices are joined:
info  2020-05-27 13:19:53: 0xb0ce1814031e3891 (0xb0ce1814031e3891): E11-G13 - Sengled Element Classic (A19) (EndDevice)
info  2020-05-27 13:19:53: 0x90fd9ffffe84c567 (0x90fd9ffffe84c567): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Router)
info  2020-05-27 13:19:53: 0xb0ce1814031def4f (0xb0ce1814031def4f): E11-G13 - Sengled Element Classic (A19) (EndDevice)
info  2020-05-27 13:19:53: 0xb0ce1814031e2681 (0xb0ce1814031e2681): E11-G13 - Sengled Element Classic (A19) (EndDevice)
info  2020-05-27 13:19:53: 0xb0ce1814031e2af8 (0xb0ce1814031e2af8): E11-G13 - Sengled Element Classic (A19) (EndDevice)
info  2020-05-27 13:19:53: 0x00158d000273558e (0x00158d000273558e): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2020-05-27 13:19:53: 0xf0d1b80000111707 (0xf0d1b80000111707): 72922-A - Sylvania SMART+ Smart Plug (Router)
info  2020-05-27 13:19:53: 0xf0d1b80000114c6f (0xf0d1b80000114c6f): 72922-A - Sylvania SMART+ Smart Plug (Router)
info  2020-05-27 13:19:53: 0xf0d1b80000021489 (0xf0d1b80000021489): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
info  2020-05-27 13:19:53: 0x00158d0002752b44 (0x00158d0002752b44): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2020-05-27 13:19:53: 0x00158d00025186d6 (0x00158d00025186d6): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2020-05-27 13:19:53: 0x00158d00025314d4 (0x00158d00025314d4): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2020-05-27 13:19:53: 0xb0ce1814035ebbb6 (0xb0ce1814035ebbb6): E11-G13 - Sengled Element Classic (A19) (EndDevice)
info  2020-05-27 13:19:53: 0xb0ce1814035f127b (0xb0ce1814035f127b): E11-G13 - Sengled Element Classic (A19) (EndDevice)
info  2020-05-27 13:19:53: 0x00158d000253a618 (0x00158d000253a618): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2020-05-27 13:19:53: 0x00158d000253a5fb (0x00158d000253a5fb): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2020-05-27 13:19:53: 0xb0ce1814035f386b (0xb0ce1814035f386b): E11-G13 - Sengled Element Classic (A19) (EndDevice)
info  2020-05-27 13:19:53: 0xb0ce1814035f19b5 (0xb0ce1814035f19b5): E11-G13 - Sengled Element Classic (A19) (EndDevice)
info  2020-05-27 13:19:53: 0x00158d0002a23f1c (0x00158d0002a23f1c): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
info  2020-05-27 13:19:53: 0x0017880103b7e500 (0x0017880103b7e500): Not supported (Router)
info  2020-05-27 13:19:53: 0x00124b000be8ae2f (0x00124b000be8ae2f): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
warn  2020-05-27 13:19:53: `permit_join` set to  `true` in configuration.yaml.
warn  2020-05-27 13:19:53: Allowing new devices to join.
warn  2020-05-27 13:19:53: Set `permit_join` to `false` once you joined all devices.
info  2020-05-27 13:19:53: Zigbee: allowing new devices to join.
info  2020-05-27 13:19:53: Connecting to MQTT server at mqtt://localhost
info  2020-05-27 13:19:53: Connected to MQTT server
info  2020-05-27 13:19:53: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-05-27 13:19:53: MQTT publish: topic 'zigbee2mqtt/0xb0ce1814031e3891', payload '{"update_available":false,"state":"OFF","brightness":255}'
info  2020-05-27 13:19:53: MQTT publish: topic 'zigbee2mqtt/0x90fd9ffffe84c567', payload '{"state":"OFF","update_available":true,"brightness":255,"color_temp":500}'
info  2020-05-27 13:19:53: MQTT publish: topic 'zigbee2mqtt/0xb0ce1814031def4f', payload '{"update_available":false,"state":"OFF","brightness":255}'
info  2020-05-27 13:19:53: MQTT publish: topic 'zigbee2mqtt/0xb0ce1814031e2681', payload '{"state":"OFF","update_available":false,"brightness":255}'
info  2020-05-27 13:19:53: MQTT publish: topic 'zigbee2mqtt/0xb0ce1814031e2af8', payload '{"update_available":false,"state":"OFF","brightness":255}'
info  2020-05-27 13:19:53: MQTT publish: topic 'zigbee2mqtt/0x00158d000273558e', payload '{"battery":100,"voltage":3155,"illuminance":59,"illuminance_lux":59,"linkquality":76,"occupancy":false}'
[... cut ...]
error 2020-05-27 13:25:59: Failed to call 'HomeAssistant' 'onZigbeeEvent' (TypeError: Cannot read property 'ID' of null
    at HomeAssistant.getDevicePayload (/app/lib/extension/homeassistant.js:2043:66)
    at /app/lib/extension/homeassistant.js:1885:35
    at Array.forEach (<anonymous>)
    at HomeAssistant.discover (/app/lib/extension/homeassistant.js:1850:37)
    at HomeAssistant.onZigbeeEvent (/app/lib/extension/homeassistant.js:2037:18)
    at Controller.callExtensionMethod (/app/lib/controller.js:350:44))

```

Koenkk, 2020-05-28:
I don't see any network scan occurring in this log, also please post the logging on pastebin.com and link it here for better readability.

schester44, 2020-05-29:
Whoops, try this https://pastebin.com/WRm2HHQv

Koenkk, 2020-05-30:
I think the issue occurs because the network address is the same. As you can see there are two links with a lqi of exactly 142, to the coordinator itself and to the router. 

For the coordinator try changing the `nwkAddr` property in `database.db` to `0`. (`"nwkAddr":0`)

schester44, 2020-06-01:
The router is still periodically blinking 3 lights indicating its having trouble communicating with the coordinator so I will give that a shot -- thanks for looking into this!

schester44, 2020-06-01:
Any recommended steps for putting that change into motion?

I've changed nwkAddr to 0 for the Coordiator, saved the file, and restarted the container but the `nwkAddr` value resets to 15006, the same as the router.

Koenkk, 2020-06-02:
You first have to stop the container, then edit and start again (otherwise it will be overwritten).

schester44, 2020-06-02:
There we go 👍 That took care of the coordinator pointing at itself. Appreciate it

Koenkk, 2020-06-02:
Great! I assume this can be closed now.

closed 2020-06-02 by Koenkk