Koenkk/zigbee2mqtt #3598: Switching off GL-C-008-2ID in a zigbee group

SebiGo, 2020-05-23:
# Bug Report
## What happened
With Fix #3436 Koenkk introduced to control the GL-C-008-2ID RGB and CCT seperately, thanks!
My Gledopto device is also a member of zigbee groups. If I send "state":"OFF" to a group, the GL-C-008-2ID should switch off regardless of the RGB/CCT status. However, this works only for the RGB LEDs, the CCT LEDs are unaffected. 
This issue seems to be related to the state topic on a GL-C-008-2ID. Effectively, it would be nice to have a mains switch for the GL-C-008-2ID allthough it is two lamps in one.

## What did you expect to happen
If I send OFF to the state topic, it should act like a main switch for both, RGB LEDs and CCT LEDs. 
(If I send ON to the state topic, it should act like a main switch for both, RGB and CCT, similar to sending ON to both state topics zigbee2mqtt/glc0082id/rgb/set and zigbee2mqtt/glc0082id/cct/set. If one is already on, it should switch on the other.)

## How to reproduce it
Pair a GL-C-008-2ID, switch on both CCT and RGB, then switch it off with
zigbee2mqtt/glc0082id/set
```
{
  "state": "OFF"
}
```
CCT LEDs will stay on, RGB-LEDs are off.

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200417.zip

Koenkk, 2020-05-23:
In case you want to control both `rgb` and `cct` through group you should add both endpoints to the group. 

This can be done with: https://www.zigbee2mqtt.io/information/groups.html#adding-a-specific-endpoint. In your case the payload should be `glc0082id/rgb` and `glc0082id/cct`.

SebiGo, 2020-05-23:
Thank you, works nicely this way. I created a PR for the documentation.

closed 2020-05-23 by SebiGo