Koenkk/zigbee2mqtt #3599: Osram Smart+ Classic E27 Dimmable Interview Failed

hutauf, 2020-05-23:
# Bug Report
## What happened
Trying to pair a Osram Smart+ Classic E27 Dimmable, which should be supported, but pairing fails.
Here is the link to the device: https://www.zigbee2mqtt.io/devices/AC10786-DIM.html
I am not 100% sure it is exactly this device, the name is exactly the same, but couldn't find the model number "AC10786" on the package.
The CC2531 should work, a Honeywell smoke detector is already connected and working.

## What did you expect to happen
I expected it to pair.

## How to reproduce it (minimal and precise)
Turn it off, wait a bit, turn it on again, it will try to pair again but will fail again.

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

## Interview Fails Procedure

I tried to pair it again for about 10 times. I have 2 bulbs, they both behave the same.
The device does not join as 0x000..., see log.
Here is the output of the zigbee-herdsman debug logging:
https://hastebin.com/dutahozinu.bash

Koenkk, 2020-05-23:
It seems that the device does not respond to any request. In this case I would recommend to ty repairing through touchlink: https://www.zigbee2mqtt.io/information/touchlink.html

hutauf, 2020-05-23:
Here is the log:
https://hastebin.com/edagoyalak.http
It seems like it doesn't answer. I held the CC2531 right next to the bulb base (~2-4cm apart).

hutauf, 2020-05-23:
I tried to factory reset the device using the 5x 5sec on 5sec off trick, but it didn't blink in the end signaling the factory reset was successful. Can this be the issue? It is joining the hub, so I guess it's in search mode or something?

hutauf, 2020-05-24:
After several retries, I finally made the 5x 5sec on 5sec off trick work, so the bulb blinked in the end signaling a successful factory reset. But the issue is still the same.

Koenkk, 2020-05-24:
Did you also chck all steps from: https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all ?

hutauf, 2020-05-24:
Yes, except reflashing the CC2531. I broke a pin and it will be tough to solder a connection again.
Do you think this might be the issue?

hutauf, 2020-05-24:
Should I try another channel? I don't live in the city, there is probably nobody around also using zigbee.

Koenkk, 2020-05-25:
Maybe you can try, as that will also reset the stick itself (which should be the same as reflashing).

hutauf, 2020-05-25:
I changed the channel to 25.
It is now paired.
There is a wifi router about 0.5m next to the CC2531, the wifi is on channel 6, which should be away from zigbee channel 11 (default).
I also changed the wifi in the basement (1 level lower) from channel 11 to channel 1, so that it will not interfere with zigbee 25.
So wifi channel 1 was free (except for neighbors, but they are ~10m away with walls in between..

So still not sure why it is working now, but it is. Thanks for your support!

closed 2020-05-25 by hutauf