Koenkk/zigbee2mqtt #3603: Aqara temperature/humidity sensor not updating

shalwia, 2020-05-24:
I'm using zigbee2mqtt in Domoticz to get the temperature and humidity values from the Aqara sensors. After pairing the sensors with the zigbee2mqtt controller, they're visible in the Custom/Zigbee2mqtt Domoticz tab. It also seems they're updating the status within some time intervals as "Last Seen" date/time is changing over the time.

The problem is that the devices created in Domoticz are not updating the temperature and humidity values - they're simply the same like during the time when sensors were added to the network. What may cause that problem?

tibortiz, 2020-06-01:
@shalwia Did you see Zigbee packets on the zigbee2mqtt logs ? Maybe you have to rejoin the devices ? You can force the Aqara device to send a message by pressing the small button at the top (the blue led will blink). The transmission criteria is not very clear but if the temperature, humidity and pressure is stable the device will send less message. A strong variation on these values should generate a message. On regular basis i'm getting a message every 20/30 minutes

couloum, 2020-06-27:
I had same issue with 3 temperature sensors. I don't know if it is related to distance. However, a simple repairing fixed it each time.

closed 2020-08-08 by stale[bot]