Koenkk/zigbee2mqtt #3604: Xiaomi QBKG12LM not pairing

JaspervRijbroek, 2020-05-24:
Hello all,

I am trying to pair a [Xiaomi QBKG12LM](https://www.zigbee2mqtt.io/devices/QBKG12LM.html) however I seem to get EAGAIN errors.

My log can be found here: https://gist.github.com/JaspervRijbroek/199b29c16fd8116afe94120f40ad3b1d

I press the buttons (both) for about 10 seconds, first they flash red, and then they start blinking blue untill they stop at some point. However the device doesn't seem to connect to the network.

Below a map of my network. I don't think that the link could be an issue as I paired a [Xiaomi MCCGQ11LM](https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html) at a further distance from the coordinator before.

[image]

Could anybody help me paring this device?

Kind Regards and Thanks in Advance,

Jasper

Koenkk, 2020-05-25:
These `EGAIN` errors are not a problem (normal logging of the underlying used serialport libary).

You probably want to check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

JaspervRijbroek, 2020-05-26:
@Koenkk Turns out that the particular device that I have has close to no range at all ;) Thanks for the support and amazing project!

closed 2020-05-26 by JaspervRijbroek