Koenkk/zigbee2mqtt #3607: adapter hangs at  zigbee2mqtt startup

arturdd, 2020-05-25:
# Bug Report
## What happened
CC2531 + 19 devices. This setup was working fine for about 6 months on rpi3, then 2 months on rpi4
Since yesterday, it no longer works. The LED on the adapter goes off right after zigbee2mqtt launch
No software updates / configuration changes were made before the failure.
permit_join was false all the time. No new devices were added to the network

Regarding FAQ
modem manager is not installed, port test is OK
reset button does not help

I've updated the Zigbee2mqtt  to the newest available version but issue still occurs

Any suggestions?

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

herdsman debug logging:
https://pastebin.com/V1bhmMYU

configuration.yaml
https://pastebin.com/McQgUsg1

Koenkk, 2020-05-25:
I would suggest reflashing the source routing firmware (for improved stability): https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

arturdd, 2020-05-26:
After many hours of reflashing the usb stick back and forth and changing the zb2mqtt config I've finally brought my cc2531 with 256k flash back to life.

Performed steps:
clean zigbee2mqtt installation (default config, no database etc)
reflashing with source_routing firmware - no change (endless red led blinking, no reaction to S2 button)
reflashing with default coordinator  firmware 20190619 - no change
crying a lot, hitting the desk - no change
but reflashing Z-Stack_3.0.x firmware CC2531_20190425.zip somehow did the trick :)

Now I am back with Z-Stack_Home_1.2 (source_routing) and it works like a charm.

@Koenkk does reflashing erase and restore all memory/registers to it's defaults?

Koenkk, 2020-05-27:
@arturdd it depends on the action, I expect it to do when "Erase, program and verify" is set (https://www.zigbee2mqtt.io/images/smartrf.png)

closed 2020-07-26 by stale[bot]