Koenkk/zigbee2mqtt #3609: Feature/Help request : PTVO Custom device - Sending something to UART port.

netweaver1970, 2020-05-25:
Hi,

Including @ptvoinfo as well, as he's he ultimate developer for the custom firmware.

I'm playing with a CC2530 Dev board, flashed with the latest PTVO firmware. I can see it's properly recognized by z2m and I can see the 5 default switches popping up in HomeAssistant. I specifically configured the firmware to have UART output at P0.2, as suggested.
But how do I send something (some ASCII text initially) to the device, from HomeAssistant (or any other MQTT client) over Z2M? I expect using the MQTT flow but I'm a bit lost on exact syntax.

In issue #3320 I can see the following topic suggestion:
     zigbee2mqtt/[FRIENDLY_NAME]/set/ep1 

But when trying to do a publish from HomeAssistant, to try, I get below in my z2m log:
```
zigbee2mqtt:debug 2020-05-25 13:40:24: Received MQTT message on 'zigbee2mqtt/rolluik/set/ep1' with data '11'
zigbee2mqtt:error 2020-05-25 13:40:24: No converter available for 'ep1' (11)
```

This is the entry in my database.db:
`{"id":2,"type":"Router","ieeeAddr":"0x00124b000b0ea049","nwkAddr":56059,"manufId":4447,"manufName":"ptvo.info","powerSource":"Mains (single phase)","modelId":"ptvo.switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2821],"outClusterList":[0,2821,20,20],"clusters":{"genBasic":{"attributes":{"modelId":"ptvo.switch","manufacturerName":"ptvo.info","powerSource":1,"zclVersion":1,"hwVersion":1,"dateCode":"2019-06-21"}}},"binds":[]}},"hwVersion":1,"dateCode":"2019-06-21","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590413840461}`

After double checking my config in the Firmware config tool, I think I corrected it: Defined Output 5, P0.2 as UART and Input 1, P0.3 as UART.

Now I'm getting this in the database.db:
`{"id":2,"type":"Router","ieeeAddr":"0x00124b000b0ea049","nwkAddr":56059,"manufId":4447,"manufName":"ptvo.info","powerSource":"Mains (single phase)","modelId":"ptvo.switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2821],"outClusterList":[0,2821,20,20],"clusters":{"genBasic":{"attributes":{"modelId":"ptvo.switch","manufacturerName":"ptvo.info","powerSource":1,"zclVersion":1,"hwVersion":1,"dateCode":"2019-06-21"}}},"binds":[]}},"hwVersion":1,"dateCode":"2019-06-21","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590413840461}`

Seems to be the same as before.

Also the publishing of a sample '11' to the specified topic leads to the same error as before.

What (is/am I doing) wrong? Syntax issue?
Also, the (UART related) screenshot shown in the announcement [here](https://ptvo.info/zigbee-configurable-firmware-features/) shows a configurable baud rate and end character. It looks like the version you show of in the announcement is different from the downloadable one.

I can only download this file from the website : cc2530_io_1.2.2a.44539_firmware.zip

Sorry to address it to both of you @Koenkk @ptvoinfo but it seems (beginners impression) it's somewhere at the lower layers, where you 2 normally meet up :)

If it's a simple syntax error, please clarify it here so it can be found again by other people, later on.
Also, if I wanted to send arbitrary ASCII codes (instead of plain text) over, what would be the syntax?

Thanks for helping a starting Zigbee fan out.

ptvoinfo, 2020-05-26:
@netweaver1970 Hi.

Please, disable the configured input. You need to configure P02 only. [More info](https://ptvo.info/zigbee-configurable-firmware-features/).

In zigbee2mqtt, you may send text data to UART data through MQTT topic `zigbee2mqtt/[FRIENDLY_NAME]/set/action`
  
or send hex data in the JSON format {"action": [4,3,2,1,0]} to `zigbee2mqtt/[FRIENDLY_NAME]/set`

netweaver1970, 2020-05-26:
@ptvoinfo Hi,

I did use the correct topic as you suggested and it all works fine. Also the receiving end works nicely, using the onboard CH340G UART2USB convertor on the CC2530 Dev board (send '11', receive 'test' back):
```
zigbee2mqtt:debug 2020-05-26 09:44:38: Received MQTT message on 'zigbee2mqtt/rolluik/set/action' with data '11'
zigbee2mqtt:debug 2020-05-26 09:44:38: Publishing 'set' 'action' to 'rolluik'
zigbee2mqtt:debug 2020-05-26 09:46:26: Received Zigbee message from 'rolluik', type 'attributeReport', cluster 'genMultistateValue', data '{"stateText":"test"}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-05-26 09:46:26: MQTT publish: topic 'zigbee2mqtt/rolluik', payload '{"state_bottom_left":"OFF","action":"test","linkquality":84}'
zigbee2mqtt:info  2020-05-26 09:46:26: MQTT publish: topic 'zigbee2mqtt/rolluik', payload '{"state_bottom_left":"OFF","linkquality":84,"action":""}'
zigbee2mqtt:info  2020-05-26 09:46:26: MQTT publish: topic 'homeassistant/device_automation/0x00124b000b0ea049/action_test/config', payload '{"automation_type":"trigger","type":"action","subtype":"test","payload":"test","topic":"zigbee2mqtt/rolluik/action","device":{"identifiers":["zigbee2mqtt_0x00124b000b0ea049"],"name":"rolluik","sw_version":"Zigbee2mqtt 1.13.1-dev","model":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)","manufacturer":"Custom devices (DiY)"}}'
zigbee2mqtt:info  2020-05-26 09:46:26: MQTT publish: topic 'zigbee2mqtt/rolluik/action', payload 'test'
```

Some further questions:

- I can see in the readme, in the downloaded archive, that the baudrate is fixed to 9600 while in your [link](https://ptvo.info/zigbee-configurable-firmware-features/) you show a configurable baudrate + configurable EndOfTransmission char.
  So it seems  in the released version it's not there (yet). Any plans to release that one too?

- Probably more for @Koenkk, on the Z2M side: the 5 switches + 1 button are always present in the device, even if not anything configured for it in the FW config. I guess it might be not easy to variably expose the -by default- variable FW config via Zigbee to Z2M to avoid creating endpoints for non-existing devices. Is there a way to prevent, already in the Z2M layer, the passing through of those ghost switches + button to MQTT/HA?

- more a joint issue, the receiving of a text message via uart over MQTT into Z2M also includes the state of 2 (of of 5?) switches ... very weird and unexpected. Is this a herdsman issue or a ptvo FW issue?

Thanks,

ptvoinfo, 2020-05-26:
@netweaver1970 
1. I've updated the firmware on my website
2. If you are trying different configurations for one device, you should clear the Z2M database and the state file.
3.  Could you attach a link to a screenshot with that issue?

netweaver1970, 2020-05-26:
1. the filename was the same for old and new, so the browser was caching the download and I ended up with same file over-and-over again. I used another browser and now I could download the same-name but updated (bigger) file

[image]

2. I did delete the database.yaml file before doing the new test with the update firmware. Still same behaviour though. This is the current database.yaml content for the ptvo device:

[image]

`{"id":2,"type":"Router","ieeeAddr":"0x00124b000b0ea049","nwkAddr":56059,"manufId":4447,"manufName":"ptvo.info","powerSource":"Mains (single phase)","modelId":"ptvo.switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2821],"outClusterList":[0,2821,20],"clusters":{"genBasic":{"attributes":{"modelId":"ptvo.switch","manufacturerName":"ptvo.info","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"2019-06-21","swBuildId":"2019-06-21"}}},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"2019-06-21","swBuildId":"2019-06-21","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590497673806}`

3. same as before (still that switch also being reported):
`
zigbee2mqtt:debug 2020-05-26 13:01:09: Received MQTT message on 'zigbee2mqtt/rolluik/set/action' with data '11'
zigbee2mqtt:debug 2020-05-26 13:01:09: Publishing 'set' 'action' to 'rolluik'
zigbee2mqtt:debug 2020-05-26 13:01:17: Received Zigbee message from 'rolluik', type 'attributeReport', cluster 'genMultistateValue', data '{"stateText":{"type":"Buffer","data":[116,101,115,116]}}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-05-26 13:01:17: MQTT publish: topic 'zigbee2mqtt/rolluik', payload '{"state_bottom_left":"OFF","linkquality":94,"action":"test"}'
zigbee2mqtt:info  2020-05-26 13:01:17: MQTT publish: topic 'zigbee2mqtt/rolluik', payload '{"state_bottom_left":"OFF","linkquality":94,"action":""}'
zigbee2mqtt:info  2020-05-26 13:01:17: MQTT publish: topic 'homeassistant/device_automation/0x00124b000b0ea049/action_test/config', payload '{"automation_type":"trigger","type":"action","subtype":"test","payload":"test","topic":"zigbee2mqtt/rolluik/action","device":{"identifiers":["zigbee2mqtt_0x00124b000b0ea049"],"name":"rolluik","sw_version":"Zigbee2mqtt 1.13.1-dev","model":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)","manufacturer":"Custom devices (DiY)"}}'
zigbee2mqtt:info  2020-05-26 13:01:17: MQTT publish: topic 'zigbee2mqtt/rolluik/action', payload 'test'`
[image]

Are there more files implicated in this? state.json maybe?

Thanks

ptvoinfo, 2020-05-27:
@netweaver1970 Is it possible you use an older version of Z2M converters?

netweaver1970, 2020-05-27:
I removed again the database.db, and now also the state.json and I reflashed the Dev board (quicker than triggering a re-pair).
Same issue, again getting the 5 switches announced to HA.
I'm on the Z2M Dev channel 1.13.1-Dev so that includes the latest Herdsman converters.

Joining log below. If there's anymore info you need, let me know.

`
zigbee2mqtt:info  2020-05-27 07:42:13: Device '0x00124b000b0ea049' joined
zigbee2mqtt:info  2020-05-27 07:42:13: Starting interview of '0x00124b000b0ea049'
zigbee2mqtt:info  2020-05-27 07:42:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b000b0ea049"}}'
zigbee2mqtt:info  2020-05-27 07:42:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b000b0ea049"}}'
zigbee2mqtt:debug 2020-05-27 07:42:13: Device '0x00124b000b0ea049' announced itself
zigbee2mqtt:info  2020-05-27 07:42:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b000b0ea049"}}'
zigbee2mqtt:debug 2020-05-27 07:42:13: Received Zigbee message from '0x00124b000b0ea049', type 'readResponse', cluster 'genBasic', data '{"modelId":"ptvo.switch"}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-05-27 07:42:13: MQTT publish: topic 'homeassistant/switch/0x00124b000b0ea049/switch_bottom_left/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state_bottom_left }}","command_topic":"zigbee2mqtt/0x00124b000b0ea049/bottom_left/set","state_topic":"zigbee2mqtt/0x00124b000b0ea049","json_attributes_topic":"zigbee2mqtt/0x00124b000b0ea049","name":"0x00124b000b0ea049_switch_bottom_left","unique_id":"0x00124b000b0ea049_switch_bottom_left_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b000b0ea049"],"name":"0x00124b000b0ea049","sw_version":"Zigbee2mqtt 1.13.1-dev","model":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)","manufacturer":"Custom devices (DiY)"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-05-27 07:42:13: MQTT publish: topic 'homeassistant/switch/0x00124b000b0ea049/switch_bottom_right/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state_bottom_right }}","command_topic":"zigbee2mqtt/0x00124b000b0ea049/bottom_right/set","state_topic":"zigbee2mqtt/0x00124b000b0ea049","json_attributes_topic":"zigbee2mqtt/0x00124b000b0ea049","name":"0x00124b000b0ea049_switch_bottom_right","unique_id":"0x00124b000b0ea049_switch_bottom_right_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b000b0ea049"],"name":"0x00124b000b0ea049","sw_version":"Zigbee2mqtt 1.13.1-dev","model":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)","manufacturer":"Custom devices (DiY)"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-05-27 07:42:13: MQTT publish: topic 'homeassistant/switch/0x00124b000b0ea049/switch_top_left/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state_top_left }}","command_topic":"zigbee2mqtt/0x00124b000b0ea049/top_left/set","state_topic":"zigbee2mqtt/0x00124b000b0ea049","json_attributes_topic":"zigbee2mqtt/0x00124b000b0ea049","name":"0x00124b000b0ea049_switch_top_left","unique_id":"0x00124b000b0ea049_switch_top_left_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b000b0ea049"],"name":"0x00124b000b0ea049","sw_version":"Zigbee2mqtt 1.13.1-dev","model":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)","manufacturer":"Custom devices (DiY)"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-05-27 07:42:13: MQTT publish: topic 'homeassistant/switch/0x00124b000b0ea049/switch_top_right/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state_top_right }}","command_topic":"zigbee2mqtt/0x00124b000b0ea049/top_right/set","state_topic":"zigbee2mqtt/0x00124b000b0ea049","json_attributes_topic":"zigbee2mqtt/0x00124b000b0ea049","name":"0x00124b000b0ea049_switch_top_right","unique_id":"0x00124b000b0ea049_switch_top_right_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b000b0ea049"],"name":"0x00124b000b0ea049","sw_version":"Zigbee2mqtt 1.13.1-dev","model":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)","manufacturer":"Custom devices (DiY)"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-05-27 07:42:13: MQTT publish: topic 'homeassistant/switch/0x00124b000b0ea049/switch_center/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state_center }}","command_topic":"zigbee2mqtt/0x00124b000b0ea049/center/set","state_topic":"zigbee2mqtt/0x00124b000b0ea049","json_attributes_topic":"zigbee2mqtt/0x00124b000b0ea049","name":"0x00124b000b0ea049_switch_center","unique_id":"0x00124b000b0ea049_switch_center_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b000b0ea049"],"name":"0x00124b000b0ea049","sw_version":"Zigbee2mqtt 1.13.1-dev","model":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)","manufacturer":"Custom devices (DiY)"},"availability_topic":"zigbee2mqtt/bridge/stat [... cut ...] qtt 1.13.1-dev","model":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)","manufacturer":"Custom devices (DiY)"}}'
zigbee2mqtt:info  2020-05-27 08:03:06: MQTT publish: topic 'zigbee2mqtt/0x00124b000b0ea049/action', payload 'test_Wednesday'
```

So now only the ghost switches announced during joining are unexpected artifacts.
Is the double publishing of the LQI (once together the incoming action and once seperately) expected? 

Thanks.

ptvoinfo, 2020-05-27:
@netweaver1970 I don't know anything about HA because I don't use it. I don't know how it gets info about the device configuration. I think, it gets all supported converters from Z2M, but they are not used in your firmware configuration.

netweaver1970, 2020-05-27:
@ptvoinfo Thanks.

@Koenkk , I see below config in devices.js. Is there a way to make the HA devices depending on the device FW config? Or is that impossible with a highly configurable device as this?
```    {
        zigbeeModel: ['ptvo.switch'],
        model: 'ptvo.switch',
        vendor: 'Custom devices (DiY)',
        description: '[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/)',
        supports: 'hold, single, double and triple click, on/off',
        fromZigbee: [fz.ptvo_switch_state, fz.ptvo_switch_buttons],
        toZigbee: [tz.on_off, tz.ptvo_switch_trigger],
        endpoint: (device) => {
            return {'bottom_left': 1, 'bottom_right': 2, 'top_left': 3, 'top_right': 4, 'center': 5};
        },
    }
```
If not, then I just have to disable the unwanted entities in HA. No big deal but it would be nice if there was a smart way of generating the real HA config based on the real FW config, IF that's exposed to Z2M in the 1st place of course. 

Also, is the 2x MQTT publishing of the Zigbee payload (with different syntax) and the 2x publishing of the LQI the normal flow?

Thanks.

closed 2020-07-26 by stale[bot]

pawelGontarz, 2020-08-08:
Hi, i have similar problem with UART on the CC2530 module (custom PTVO firmware). If i want to send payload "11" to topic "zigbee2mqtt/0x00124b000e368eb9/set/action" i receive this error: https://ibb.co/7JTsMfG . My firmware is:
-settings (https://ibb.co/VBMNqcZ)
- expert (https://ibb.co/L6jyS3R)
@Koenkk any idea where is problem? I am using the lates version of zigbee2mqtt and PTVO firmware tool.