Koenkk/zigbee2mqtt #3610: Osram motion sensor sends occupancy false but not true.

MatthiasKohler, 2020-05-25:
#3172 Bug Report
## What happened

I have two Osram motion sensors (AC01353010G), which correctly detect motion as can be seen on their flashing green light. I do not receive any occupancy: "true" message. But I do receive a occupancy: "false"  shortly after triggering them. 

I switched from a CC2531 to a CC1352P-2, because the CC2531 seemed unreliable. But both sensors worked okayish with the CC2531. I repaired all devices and changed the pan-id. Right now all lightbulbs and powerplugs can be switched reliably without any noticeable latency and two Xiaomi sensors report flawlessly. I will report another bug, because there is one lightbulb which wont report its model and hence can not be controlled.

## What did you expect to happen

Receive a message with occupancy: "true" from the triggered motion sensor.

## Debug Info
Zigbee2mqtt version: 1.13.1 (same with 1.12.1)
Adapter hardware:  CC1352P-2
Adapter firmware version: zStack3x0 20200417

Jogobo, 2020-05-26:
I have 3 Osram Motion Sensors and they work, let's say, with some restrictions. First of all, did you check the database.db for "interviewCompleted=true" on the sensors? I found out that I needed several tries to get "interviewCompleted" completed. Until then I only got a "state=on" message.
What do you receive from the sensor when you subscribe to the topic of the sensor?

Koenkk, 2020-05-26:
@Jogobo @MatthiasKohler could you both share your `database.db`?

@MatthiasKohler is the issue solved by changing the entry in `devices.js` to:

```js
{
    zigbeeModel: ['Motion Sensor-A'],
    model: 'AC01353010G',
    vendor: 'OSRAM',
    description: 'SMART+ Motion Sensor',
    supports: 'occupancy, tamper and temperature',
    fromZigbee: [
        fz.temperature,
        fz.ias_occupancy_alarm_1,
    ],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
        await configureReporting.temperature(endpoint);
        await configureReporting.batteryVoltage(endpoint);
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to get to `devices.js`

MatthiasKohler, 2020-05-26:
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/4683938/database.db.txt)

This change did not solve the issue. I am also not getting any message from them anymore.I tried resetting both sensors by pressing the button for at least ten seconds. Few weeks ago this worked and the device indicated the reset by flashing its light. Now I do not get any reaction from neither of them. Maybe they are just broken.

Jogobo, 2020-05-26:
The 10 seconds button press reset does not work on my motion sensors either. You have to do the following steps:

- Remove the battery for at least 6 seconds
- Press the button and keep it pressed
- Insert the battery 
- Release the button when the LED is red

Jogobo, 2020-05-26:
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/4684181/database.db.txt)

MatthiasKohler, 2020-05-26:
The resetting method above results in a device announcement but nothing else. It does not send any message when it detects movement.

Jogobo, 2020-05-26:
You should remove the Motion Sensor from zigbee2mqtt before resetting the device

MatthiasKohler, 2020-05-26:
Is it sufficient to delete the line in configuration.yaml?

Jogobo, 2020-05-26:
You should remove the device from the database too. Publish the device friendly name to bridge/config/remove or shutdown zigbee2mqtt and delete the line from database.db

MatthiasKohler, 2020-05-26:
Both sensor work again. Thank you!

I have `fz.ias_occupancy_alarm_2`, which I think it is right now by default in `devices.js`

It would be useful if this method of resetting above is listed in the documentation.

Koenkk, 2020-05-27:
Good! Please create a PR to update https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/AC01353010G.md (press the pencil icon). Assuming this can be closed now.

closed 2020-05-27 by Koenkk