Koenkk/zigbee2mqtt #3611: Osram lightbulb does not report model, hence can not be controlled.

MatthiasKohler, 2020-05-25:
## What happened

I have four Osram lightbulb (AC03642) three of them work without any problem. One refuses to even after resetting it multiple times to report its model. This prevents if from being controlled. 

I switched from a CC2531 to a CC1352P-2, because the CC2531 seemed unreliable. This paritcular lamp worked okyaish with the CC2531. I repaired all devices and changed the pan-id.  I havel reported another bug, because I have also a problem with two motion sensors.

## What did you expect to happen

Lightbulb should report its model.

## Debug Info
Zigbee2mqtt version: 1.13.1 (same with 1.12.1)
Adapter hardware: CC1352P-2
Adapter firmware version: zStack3x0 20200417

Koenkk, 2020-05-26:
could you share the herdsman debug logging when pairing this device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

MatthiasKohler, 2020-05-26:
[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/4684164/debug.txt)

Switched allow pairing on, reset offending lightbulb waited a few moments and stopped zigbee2mqtt.

MatthiasKohler, 2020-05-26:
I removed the lightbulb by sending its friendly name to `zigbee2mqtt/bridge/config/remove`, this worked for another problem [](https://github.com/Koenkk/zigbee2mqtt/issues/3610). It also worked here and the lightbulb again can be controlled.

Koenkk, 2020-05-27:
Do you have the herdsman debug log of the original pairing attempt (after which it didn't work)?

If you cannot reproduce it anymore, please close this issue.

MatthiasKohler, 2020-05-27:
Is the debug log above (debug.txt) not what you are looking for?

Koenkk, 2020-05-28:
No, the device was already repaired so it's not reinterviewed. First remove the device through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

closed 2020-05-28 by MatthiasKohler