Koenkk/zigbee2mqtt #3614: Cannot start Zigbee2MQTT after update to 1.13.1.1 (friendly_name not allowed error)

Elmardus, 2020-05-26:
After updating from 1.13.0 to 1.13.1.1, I cannot start Zigbee2MQTT. I get the following errors in the log:
```
UnhandledPromiseRejectionWarning: Error: Following friendly_name are not allowed: 'left,right,center,bottom_left,bottom_right,default,top_left,top_right,white,rgb,cct,system,top,bottom,center_left,center_right,ep1,ep2,row_1,row_2,row_3,row_4,relay,l1,l2,l3,l4,l5,l6,l7,l8,button_1,button_2,button_3,button_4,button_5,button_6,button_7,button_8,button_9,button_10,button_11,button_12,button_13,button_14,button_15,button_16,button_17,button_18,button_19,button_20',
```

Full log here: https://pastebin.com/58fJNTZm

## Debug Info
Zigbee2mqtt version: 1.13.1.1
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version: zStack12

sjorge, 2020-05-26:
You have a device with a friendly_name that is also used as an endpoint name so those are not allowed. I think the list got updated a while back because a few are missing. 

So check your friendly_name’s in configuration.yaml for any of the names/values listed in the message.

Elmardus, 2020-05-27:
I had a device named button_1, so this caused the problem and was fixed by renaming it. Thanks.

closed 2020-05-27 by Elmardus