Koenkk/zigbee2mqtt #3615: Data request failed with error: 'No network route' (205))

aureliosaraiva, 2020-05-26:
# Bug Report
sometimes this error occurs when I try to change the state of a device. Sometimes it happens when I send 20 lamps at the same time and some of them occur this problem there.

Sometimes it occurs only when you turn on one, and it goes out of order and then after a while it works again.

What can this be?
## What happened

```
Publish 'set' 'state' to '0xf0d1b80000061b44' failed: 'Error: Command 0xf0d1b80000061b44/1 genOnOff.on({}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-05-26 20:33:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0xf0d1b80000061b44' failed: 'Error: Command 0xf0d1b80000061b44/1 genOnOff.on({}, {\"timeout\":10000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0xf0d1b80000061b44"}}'
```

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: 1.13.1

Koenkk, 2020-05-27:
Please try using the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

aureliosaraiva, 2020-06-20:
I changed the firmware.  The mesh network looks faster now.  I'll take a few days to see if it solves the problem.

closed 2020-07-20 by aureliosaraiva

lymereJ, 2020-08-30:
I'm getting the error all the time and for all my devices since I updated to Zigbee2MQTT version 1.14.3-dev.

Here's an example:
```
Zigbee2MQTT:error 2020-08-30 14:45:37: Publish 'set' 'state' to '0x0017880103efa9d1' failed: 'Error: Command 0x0017880103efa9d1/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
```
They are accompanied by:
```
Zigbee2MQTT:error 2020-08-30 14:45:45: Publish 'set' 'state' to '0x0017880103905910' failed: 'Error: Command 0x0017880103905910/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 57826 - 11 - 5 - 6 - 11 after 10000ms)'
```
I tried using the source routing firmware but I'm still getting these errors. Any suggestions on what I should try next?

cytrinox, 2020-11-26:
@Koenkk I've the same problem and before flashing the CC2531 I've a understanding issue:
I've 36 devices paired (and more to come). The Z-Stack firmware page says the home default firmware supports max 30 normal routes. So I wonder why this works "most" of the time? 
I've found this statement in faq:

> However, by having routers in your network the network size can be extended. Probably all AC powered devices e.g. bulbs serve as a router

24 of my 36 devices are light bulbs by innr, so maybe I already have multiple routers and can use more than 30 devices? If yes, is it automatically determined what device gets routed over another device? What happens if a bulb that acts as router gets physically powered off? Does it results in the reported "no network route" error or should automatically another router choosen?

elbueno222, 2020-11-26:
If you are using Home Assistant and zigbee2mqtt you can use the Zigbee Map option to understand what devices are routers and how they are connected (i.e. https://community.home-assistant.io/t/zigbee2mqtt-show-the-networkmap-in-home-assistant/89116)
Regarding your question about one router being powered off, zigbee creates a mesh network that automatically detects offline routers and route the messages through a different router (during the re-routing process some messages can return the "no network router" error though)

cytrinox, 2020-11-26:
@elbueno222 thanks. I would go with the Source routing firmware. Do I understand it correctly that with this firmware, only 5 bulbs can connect directly to the coordinator and all other devices automatically selects a bulb as a router without any configuration from my side?
I use OpenHAB2, so can't use this nice HA feature.

elbueno222, 2020-11-26:
@cytrinox yes, that is how it should work