Koenkk/zigbee2mqtt #3616: 1.13.1 armv6 image is broken (Cannot find module '@serialport/bindings')

halkeye, 2020-05-27:
# Bug Report
## What happened

1.13.1 armv6 image is broken (Cannot find module '@serialport/bindings')

## What did you expect to happen

It to start up. At least crash because no device was provided
```
root@raspberrypi:/etc/systemd/system# docker run -it --rm --name g1 koenkk/zigbee2mqtt:1.13.0
Using '/app/data' as data directory
Creating configuration file...

> zigbee2mqtt@1.13.0 start /app
> node index.js

zigbee2mqtt:info  2020-05-27 00:51:44: Logging to console and directory: '/app/data/log/2020-05-27.00-51-44' filename: log.txt
zigbee2mqtt:info  2020-05-27 00:51:45: Starting zigbee2mqtt version 1.13.0 (commit #55e6283)
zigbee2mqtt:info  2020-05-27 00:51:45: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-27 00:51:46: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-27 00:51:46: Failed to start zigbee
zigbee2mqtt:error 2020-05-27 00:51:46: Exiting...
zigbee2mqtt:error 2020-05-27 00:51:46: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-05-27T00_51_46_100Z-debug.log
```

## How to reproduce it (minimal and precise)

On a raspberry pi
```
root@raspberrypi:/etc/systemd/system# docker run -it --rm --name g1 koenkk/zigbee2mqtt:1.13.1
Using '/app/data' as data directory
Creating configuration file...

> zigbee2mqtt@1.13.1 start /app
> node index.js

zigbee2mqtt:info  2020-05-27 00:50:09: Logging to console and directory: '/app/data/log/2020-05-27.00-50-09' filename: log.txt
zigbee2mqtt:info  2020-05-27 00:50:10: Starting zigbee2mqtt version 1.13.1 (commit #3046680)
zigbee2mqtt:info  2020-05-27 00:50:10: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-27 00:50:10: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-27 00:50:10: Failed to start zigbee
zigbee2mqtt:error 2020-05-27 00:50:10: Exiting...
zigbee2mqtt:error 2020-05-27 00:50:10: Error: Cannot find module '@serialport/bindings'
Require stack:
- /app/node_modules/zigbee-herdsman/node_modules/serialport/lib/index.js
- /app/node_modules/zigbee-herdsman/dist/adapter/serialPortUtils.js
- /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js
- /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/index.js
- /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js
- /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/index.js
- /app/node_modules/zigbee-herdsman/dist/adapter/adapter.js
- /app/node_modules/zigbee-herdsman/dist/adapter/index.js
- /app/node_modules/zigbee-herdsman/dist/controller/controller.js
- /app/node_modules/zigbee-herdsman/dist/index.js
- /app/lib/zigbee.js
- /app/lib/controller.js
- /app/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:957:15)
    at Function.Module._load (internal/modules/cjs/loader.js:840:27)
    at Module.require (internal/modules/cjs/loader.js:1019:19)
    at require (internal/modules/cjs/helpers.js:77:18)
    at Object.<anonymous> (/app/node_modules/zigbee-herdsman/node_modules/serialport/lib/index.js:2:17)
    at Module._compile (internal/modules/cjs/loader.js:1133:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1153:10)
    at Module.load (internal/modules/cjs/loader.js:977:32)
    at Function.Module._load (internal/modules/cjs/loader.js:877:14)
    at Module.require (internal/modules/cjs/loader.js:1019:19)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.13.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-05-27T00_50_10_425Z-debug.log
```

## Debug Info
Zigbee2mqtt version: 1.13.1

Koenkk, 2020-05-28:
- With what pi version is this?
- Does the latest dev image work? `koenkk/zigbee2mqtt:latest-dev`

halkeye, 2020-05-28:
I'm pretty sure its a rasberry pi 3, but could be the one 3+ I have

uname says:
Linux raspberrypi 4.14.78-v7+ #1156 SMP Tue Oct 23 14:34:39 BST 2018 armv7l GNU/Linux

so i guess its not the armv6 image?

> Does the latest dev image work? koenkk/zigbee2mqtt:latest-dev

Yep, that one runs fine

Koenkk, 2020-05-29:
I've triggered a rebuild of the 1.13.1 image, let see if that fixes it. Can you check again in an hour from now?

halkeye, 2020-05-29:
5 hours later:

```
root@raspberrypi:~# docker pull koenkk/zigbee2mqtt:1.13.1
1.13.1: Pulling from koenkk/zigbee2mqtt
Digest: sha256:850b1232affa2605e08a8662889fe5e32d51bafd11df9d750b6e68647da9b9d1
Status: Image is up to date for koenkk/zigbee2mqtt:1.13.1
```

According to Created it hasn't been updated

```
root@raspberrypi:~# docker inspect koenkk/zigbee2mqtt:1.13.1
[
    {
        "Id": "sha256:b53702c7c6f2e7908f2a130b90b7525b40d53154527ae78eeeb139e172bd96fa",
        "RepoTags": [
            "koenkk/zigbee2mqtt:1.13.1",
            "koenkk/zigbee2mqtt:latest"
        ],
        "RepoDigests": [
            "koenkk/zigbee2mqtt@sha256:850b1232affa2605e08a8662889fe5e32d51bafd11df9d750b6e68647da9b9d1"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2020-05-20T14:57:15.028563863Z",
        "Container": "",
        "ContainerConfig": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": null,
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": null,
            "Labels": null
        },
        "DockerVersion": "",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "NODE_VERSION=12.16.3",
                "YARN_VERSION=1.22.4"
            ],
            "Cmd": null,
            "ArgsEscaped": true,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "/app",
            "Entrypoint": [
                "./run.sh"
            ],
            "OnBuild": null,
            "Labels": null
        },
        "Architecture": "arm",
        "Os": "linux",
        "Size": 101084641,
        "VirtualSize": 101084641,
        "GraphDriver": {
            "Data": {
                "LowerDir": "/var/lib/docker/overlay2/d91e5051d2a1a7160b79ac2cbe50eddcbac0bebbf50b3fc8bdbaf521f13dd1c8/diff:/var/lib/docker/overlay2/fb86e8ebf0222e69460c14808a38ee492fcd9831de1a0170c74b42d091000d02/diff:/var/lib/docker/overlay2/2eea426bfd3a5afab1f20a9eb124bc0a85561401761accc85fce64e772d82c4f/diff:/var/lib/docker/overlay2/b9c8d683b0190a045aa2d5198ee7e82375cd285f3cb010ee73f78864d0beb15f/diff:/var/lib/docker/overlay2/a7e707c0fe77fceaf3457f041ceca08483a9611faa087759dfa725baae0a5338/diff:/var/lib/docker/overlay2/e391a316a261409fd5d2a37a45fd728a33b72c513ff2b385cce0b66fd96bee2e/diff:/var/lib/docker/overlay2/4027de6273e8c92587afcaf100aa4c2485f1dc6f28ab3ee8b87ed1b7f229d576/diff:/var/lib/docker/overlay2/05d831e89efc4626d2b629528d98c9ca5622933957b99da68f46aee259a20e3b/diff:/var/lib/docker/overlay2/ffb88d636ce3091e07f0ee82ebc7dfef05fb843d5af7a22ad82131825fade08b/diff:/var/lib/docker/overlay2/d45e08c019be156cf80f275c8491fa397991d59935daecbfbd96d92d9aa62aa0/diff:/var/lib/docker/overlay2/8d0118675285bbc0dae5bb32599344a8902978679b4eea56ea9b0cbcde4990bc/diff:/var/lib/docker/overlay2/8d16008f9a5df2ea330644dd4944cc3d9fe6310dddeb1202088036933e24c85c/diff",
                "MergedDir": "/var/lib/docker/overlay2/3bd43a750c47d113ba9c151e266745c772ff9d953c564d5d9ff0e4f2fc64c85a/merged",
                "UpperDir": "/var/lib/docker/overlay2/3bd43a750c47d113ba9c151e266745c772ff9d953c564d5d9ff0e4f2fc64c85a/diff",
                "WorkDir": "/var/lib/docker/overlay2/3bd43a750c47d113ba9c151e266745c772ff9d953c564d5d9ff0e4f2fc64c85a/work"
            },
            "Name": "overlay2"
        },
        "RootFS": {
            "Type": "layers",
            "Layers": [
                "sha256:093b0aa59d568b7b9472633c090a409cb2958fc337253680bc88ba591379d4f4",
                "sha256:7d5a7be78aea35e22adf6352e6ea6a0f300bee41c7d442317430261367b7e7df",
                "sha256:534b930f0fdbec7fcdd2a2e6bafe7cad7c935e97da2d8cb445d1a5c44bd5c9ef",
                "sha256:f63b16414d1ad7abe64f05db4d51f6ab6d5ef4d8bdc52ab7d85b451e832b27d5",
                "sha256:4a775d5296eeec7bd049168d6fa624eb78bbf7131259a91588d020ee1d13d380",
                "sha256:4226f66cf88c29f5af23c09ecfc205f96b0538b4be5c7cdbe8003b4cfe6bd6de",
                "sha256:b7230290e4afd75bc752b644acccb5c409b42ce27983fc49ea40dce66df79924",
                "sha256:69c1bd233b34179c38db151cb2d0e93414535dc7acbf3c9c075084b67a651ebc",
                "sha256:63e1d8c4aab733c62de7627ad098c28daefaca0f9a2e362b08bdc288ea38798b",
                "sha256:86400d2ae19d7daa8e976c3cd292de233308625c34137cbbdbb7081cfcba8f9c",
                "sha256:5a9dcd1bd83ef2a473f274b737c8e6c35dbde11ea82a1f1ca0b5625ae2255cd0",
                "sha256:869c18cc3db25d35704ede20aef028a451c416fa60b675acbe0a3f99b7fc737d",
                "sha256:e153466fb05e3246ad6d7df037303c59bdba51db3d56710c05afcdc72b7ba4c4"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
```

Koenkk, 2020-05-30:
Sorry, the new image wasn't published. Just done this, please check if it works.

halkeye, 2020-05-30:
yep! looks good now.

Koenkk, 2020-06-01:
Great!

closed 2020-06-01 by Koenkk