Koenkk/zigbee2mqtt #3619: Yale YMF40 lock: z2mqtt not receiving "commandOperationEventNotification" on fingerprint unlock

BrunoZumba, 2020-05-27:
Hey there!

I was using dev version 1.7.1 until yesterdey, then a decided to update it. After a few tinkering I got to manage almost everything to work with the latest version (1.13.1), but I noticed that unlocking my Yale YMF40 lock with fingerprint i don't get the "commandOperationEventNotification" message anymore.

When I unlock with fingerprint everything I get on the console is the "attributeReport" message (that should come after the commandOperationEventNotification, right?):
Here's everything that pops out on the console:

```
May 27 10:35:04 raspberrypi npm[697]: zigbee2mqtt:debug 2020-05-27 10:35:04: Received Zigbee message from 'fechadura_yale', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
May 27 10:35:04 raspberrypi npm[697]: zigbee2mqtt:info  2020-05-27 10:35:04: MQTT publish: topic 'zigbee2mqtt/fechadura_yale', payload '{"state":"UNLOCK","user":0,"source":1,"linkquality":52,"battery":45,"code":10,"lock_state":"unlocked"}'
```

These messages doesn't show WHO unlocked it (it should be in the "action_user" part of the json).

Here's an example that works perfectly: unlocking with a code:
```
May 27 10:34:20 raspberrypi npm[697]: zigbee2mqtt:debug 2020-05-27 10:34:20: Received Zigbee message from 'fechadura_yale', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"opereventsrc":0,"opereventcode":2,"userid":1,"pin":0,"zigbeelocaltime":18449466,"data":0}' from endpoint 1 with groupID 0
May 27 10:34:20 raspberrypi npm[697]: zigbee2mqtt:info  2020-05-27 10:34:20: MQTT publish: topic 'zigbee2mqtt/fechadura_yale', payload '{"state":"UNLOCK","user":0,"source":1,"linkquality":47,"battery":45,"code":10,"lock_state":"unlocked","action":"unlock","action_user":1,"action_source":0}'
May 27 10:34:20 raspberrypi npm[697]: zigbee2mqtt:info  2020-05-27 10:34:20: MQTT publish: topic 'zigbee2mqtt/fechadura_yale', payload '{"state":"UNLOCK","user":0,"source":1,"linkquality":47,"battery":45,"code":10,"lock_state":"unlocked","action":""}'
May 27 10:34:20 raspberrypi npm[697]: zigbee2mqtt:info  2020-05-27 10:34:20: MQTT publish: topic 'zigbee2mqtt/fechadura_yale/action', payload 'unlock'
May 27 10:34:20 raspberrypi npm[697]: zigbee2mqtt:debug 2020-05-27 10:34:20: Received Zigbee message from 'fechadura_yale', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
May 27 10:34:20 raspberrypi npm[697]: zigbee2mqtt:info  2020-05-27 10:34:20: MQTT publish: topic 'zigbee2mqtt/fechadura_yale', payload '{"state":"UNLOCK","user":0,"source":1,"linkquality":47,"battery":45,"code":10,"lock_state":"unlocked"}'
```

First I get a commandOperationEventNotification (that shows fields like "action_user" and "action_source" on the json message) and then I get the "attributeReport".

It seems that when unlocking with fingerprint I'm missing the commandOperationEventNotification that should come first.

Any thoughts on how to get the full message when unlocking with fingerprint?

## My Config
devices.js (for some reason my device model is 0700000001 instead of iZBModule01

```
{
//        zigbeeModel: ['iZBModule01'],
        zigbeeModel: ['0700000001'],
        model: 'YMF40/YDM4109+',
        vendor: 'Yale',
        description: 'Real living lock / Intelligent biometric digital lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock_operation_event, fz.battery, fz.lock],
        toZigbee: [tz.generic_lock],
        // Increased timeout needed: https://github.com/Koenkk/zigbee2mqtt/issues/3290 for YDM4109+
        meta: {configureKey: 2, timeout: 20000},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
},

```
fromZigbee.js 
```
lock_operation_event: {
        cluster: 'closuresDoorLock',
        type: 'commandOperationEventNotification',
        convert: (model, msg, publish, options, meta) => {
            const lookup = {
                0: 'unknown',
                1: 'lock',
                2: 'unlock',
                3: 'lock_failure_invalid_pin_or_id',
                4: 'lock_failure_invalid_schedule',
                5: 'unlock_failure_invalid_pin_or_id',
                6: 'unlock_failure_invalid_schedule',
                7: 'one_touch_lock',
                8: 'key_lock',
                9: 'key_unlock',
                10: 'auto_lock',
                11: 'schedule_lock',
                12: 'schedule_unlock',
                13: 'manual_lock',
                14: 'manual_unlock',
                15: 'non_access_user_operational_event',
            };

            return {
                action: lookup[msg.data['opereventcode']],
                action_user: msg.data['userid'],
                action_source: msg.data['opereventsrc'],
            };
        },
    },

    lock: {
        cluster: 'closuresDoorLock',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('lockState')) {
                const lookup = {0: 'not_fully_locked', 1: 'locked', 2: 'unlocked'};
                return {
                    state: msg.data.lockState == 1 ? 'LOCK' : 'UNLOCK',
                    lock_state: lookup[msg.data['lockState']],
                };
            }
        },
    },

```

toZigbee.js
```
    generic_lock: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            await entity.command(
                'closuresDoorLock',
                `${value.toLowerCase()}Door`,
                {'pincodevalue': ''},
                getOptions(meta.mapped),
            );

            return {readAfterWriteTime: 200};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('closuresDoorLock', ['lockState']);
        },
    },
```

## Debug Info
Zigbee2mqtt version: `Starting zigbee2mqtt version 1.13.1-dev (commit #b706233)`
Adapter hardware: CC2531
Adapter firmware version: `Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'`

Koenkk, 2020-05-28:
- `commandOperationEventNotification` should indeed be send before the `attributeReport`.
- Did this work in Zigbee2mqtt 1.7.1?
- Could you sniff the traffic when you expect this `commandOperationEventNotification` to be send? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

BrunoZumba, 2020-05-29:
Yes, this part worked perfectly on version 1.7.1.
I stoped using the zigbee to control my lock because i got a lot of those errors: "No route to host", or something about MAC, i dont really remember. Around december/19 I got everything working, but a few days later the errors started. I unpaired the lock, paired again and it was all back to normal. But again, a week later the errors came back... so I just gave up (all my other zigbee stuff was working perfectly).

But this week I decided to update everything and give it a go, to check if the errors were gone with the new version.

Well, here we go: after messing with these sniffers for today I don't think I made a good job.

I have two zigbee sticks: CC2531 and HUSBZB-1 (that is also zwave).
I couldn't get the HUSBZB-1 to sniff any zigbee traffic at all. I am using windows 7, i don't know if that's the reason. I tried on my Ubuntu virtual machine and still nothing. 

So I tried with the CC2531 and it worked well. I got to sniff these bastards.

But since I only have ONE CC2531, I used it as the sniffer, so my network had no coordinator. I don't know if that messes up the analysis.

Below are the packets for both the code unlock (that works and send the commandOperationEventNotification) and for the fingerprint unlock (that doesnt work, only sends the attributeReport).
Both captures starts at around packet 4 (if you need details on any other packet, I saved both captures)
Code unlock:
[image]

Fingerprint unlock:
[image]

By my little package analysis skills, I think the capture shows us the exact same thing:
With code unlock we can see a "ZCL Door Lock" packet (7). Then a ZCL: Report Attributes on packt 9

But with fingerprint unlock all I got was the ZCL: Report Attributes on packet 5 (I didn't open the lock before this)

Does that mean anything to you?
Since the packet sniffer didn't sniff any "commandOperationEventNotification"ish packet, maybe my lock got faulty since December?

 

I'll try to downgrade my Zigbee2mqtt again to 1.7.1 and see if it still works. I'll report back as soon as I can.

BrunoZumba, 2020-05-29:
Yeah... problem here: how do I downgrade to version 1.7.1?

Koenkk, 2020-05-29:
Can you capture the herdsman debug log at the moment this command is send? 

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

BrunoZumba, 2020-05-31:
So...
When I got the CC2531 back into the raspberry my devices didn't work and I had to remove and repair them. No problem for the curtain and the temperature/humidity sensor... but no success from the lock yet.

I moved the raspberry pi to a few inches away from the lock and when I try to pair I get the following log: https://pastebin.com/t1SzdrMn
(I have no idea why my device is now "iZBModule01" - as expected - and no more "0700000001" as I stated above. I made the correction in devcies.js)

(I don't like pastebin because in a few months when someone is having the same problem, they'll get to this thread, try to click the link above to see the log and it maybe removed. The full pairing log is on pastebin, but I'll paste the error here too):

```
May 31 13:34:50 raspberrypi npm[17362]: zigbee2mqtt:error 2020-05-31 13:34:50: Failed to configure '0x000d6f0012bb1db5', attempt 1 (Error: Bind 0x000d6f0012bb1db5/1 closuresDoorLock from '0x00124b001936ac05/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
May 31 13:34:50 raspberrypi npm[17362]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
May 31 13:34:50 raspberrypi npm[17362]:     at Generator.throw (<anonymous>)
May 31 13:34:50 raspberrypi npm[17362]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

After that, I can see my lock on configuration.yaml and on database.db.
But when I restart zigbee2mqtt, the device is removed.
Below is the error log I get when restarting:

```
May 31 13:52:43 raspberrypi npm[923]: zigbee2mqtt:error 2020-05-31 13:52:43: Failed to configure '0x000d6f0012bb1db5', attempt 1 (Error: Bind 0x000d6f0012bb1db5/1 closuresDoorLock from '0x00124b001936ac05/1' failed (TypeError: Cannot read property 'save' of undefined)
May 31 13:52:43 raspberrypi npm[923]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
May 31 13:52:43 raspberrypi npm[923]:     at Generator.next (<anonymous>)
May 31 13:52:43 raspberrypi npm[923]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
May 31 13:52:53 raspberrypi npm[923]: zigbee2mqtt:error 2020-05-31 13:52:53: Failed to remove 0x000d6f0012bb1db5 (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
May 31 13:52:53 raspberrypi npm[923]: zigbee2mqtt:error 2020-05-31 13:52:53: See https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove for more info
May 31 13:52:53 raspberrypi npm[923]: zigbee2mqtt:info  2020-05-31 13:52:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed_failed","message":"0x000d6f0012bb1db5"}'

```

Any idea on how to correct this so I can get the Herdsman Debug log?

Thanks

BrunoZumba, 2020-05-31:
I already tried keeping the lock awake (by pressing the buttons and navigating the menu) while restarting. No success either...

Koenkk, 2020-06-01:
Interesting that the device is removed, do you also have the logging of this?

The herdsman debug logging is logged to stdout after enabling it (https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging)

BrunoZumba, 2020-06-01:
There you go

Log for when I pair the lock: https://pastebin.com/RLyWbZbY
Log for when I restart zigbee2MQTT and the lock get removed: https://pastebin.com/Pr2jmAsy

Thanks

BrunoZumba, 2020-06-01:
One more thing that caught my attetion looking at the logs right now is that EVERYTIME i restart z2mqtt i get the following messages on the log (mqtt message removing the lock)

```
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:info  2020-06-01 11:49:30: Connected to MQTT server
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:info  2020-06-01 11:49:30: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:info  2020-06-01 11:49:30: MQTT publish: topic 'zigbee2mqtt/temp_hum_sensor', payload '{"battery":86,"voltage":2975,"temperature":22.93,"humidity":55.43,"pressure":891.8,"linkquality":115}'
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:info  2020-06-01 11:49:30: MQTT publish: topic 'zigbee2mqtt/cortina', payload '{"position":16,"running":false,"linkquality":105}'
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:debug 2020-06-01 11:49:30: Received MQTT message on 'zigbee2mqtt/fechadura_yale/set' with data 'LOCK'
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:info  2020-06-01 11:49:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"fechadura_yale"}}'
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:error 2020-06-01 11:49:30: Entity 'fechadura_yale' is unknown
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:debug 2020-06-01 11:49:30: Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data '0x000d6f0012bb1db5'
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:debug 2020-06-01 11:49:30: Received MQTT message on 'zigbee2mqtt/bridge/config/force_remove' with data '0x000d6f0012bb1db5'
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:error 2020-06-01 11:49:30: Cannot remove, device '0x000d6f0012bb1db5' does not exist
```

"fechadura_yale" was the friendly name I had for my Yale Lock ("0x000d6f0012bb1db5").
But I don't have this friendly name on my configuration.yaml nor the device "0x000d6f0012bb1db5" on my database.db

Here's my configuration.yaml
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://XXX.XXX.XXX.XXX'
  user: USER
  password: PASSWORD
serial:
  port: /dev/ttyACM0
  disable_led: true
advanced:
  rtscts: true
  log_level: debug
devices:
  '0x00158d0002b5472b':
    friendly_name: temp_hum_sensor
  '0x00158d0002edb3a1':
    friendly_name: cortina
```

Here are the begining of each line on my database.db (coordinator, temp/hum sensor, curtain, respectvely)
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001936ac05","nwkAddr":0
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b5472b","nwkAddr":
{"id":3,"type":"Router","ieeeAddr":"0x00158d0002edb3a1","nwkAddr"

```

This message is probably sent from Home Assistant 
`Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:debug 2020-06-01 11:49:30: Received MQTT message on 'zigbee2mqtt/fechadura_yale/set' with data 'LOCK'`

But I don't think Home Assistant or Mosquitto Broker itself are sending the remove messages
```
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:debug 2020-06-01 11:49:30: Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data '0x000d6f0012bb1db5'
Jun 01 11:49:30 raspberrypi npm[8067]: zigbee2mqtt:debug 2020-06-01 11:49:30: Received MQTT message on 'zigbee2mqtt/bridge/config/force_remove' with data '0x000d6f0012bb1db5'
```

Koenkk, 2020-06-01:
The force remove messages is probably a retained message. Clear the retained messages from.your MQTT broker to fix this

BrunoZumba, 2020-06-01:
Oh no... can't believe how stupid I am. Sorry about that. I must have checked the retain flag on MQTTLens addon when removing the lock before >.<

Well, OK. After fixing this, i managed to get my lock paired again.
I still get the some "No converters available" while pairing.
Even though I kept pressing buttons and making the lock stay "awake" along the whole pairing process, I also still get the error below at the end: 
```
zigbee2mqtt:error 2020-06-01 19:40:29: Failed to configure '0x000d6f0012bb1db5', attempt 1 (Error: Bind 0x000d6f0012bb1db5/1 closuresDoorLock from '0x00124b001936ac05/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

```
The full herdman debug for the pairing is here: https://pastebin.com/vnPLNPxN

Even with the errors above, the lock still gets paired and I can lock and unlock it using MQTT.

But the initial error persist and got worse.
First, when I opened this thread what happened was:
* When unlocking with code, everything worked correctly. First I got the "commandOperationEventNotification" (that indicates wich user/code unlocked the lock) and then I got the "attributeReport" message.
* When unlocking with fingerprint, I didn't receive the "commandOperationEventNotification". Only the "attributeReport"

What happens now is:
* When unlocking with code I still get everything just like before (heres the herdsman debug log: https://pastebin.com/KxdZDqt2 )
* But when unlocking with fingerprint I get nothing. Now, not even the "attributeReport" message gets sent. Absolutely nothing gets printed to herdsman debug log

Could this be happening because any of the errors I get while pairing?
(by the way, the "Failed to configure '0x000d6f0012bb1db5', attempt X" keeps happening every now and then. It happened even when I was operating the lock for these tests)

Thank you so much for the time you take to answer and help us

BrunoZumba, 2020-06-02:
Here one more test I made now: since my device id used to be "0700000001" instead of "iZBModule01" (as stated in #2456), I decided to get it back to "0700000001" on my devices.js and repair the lock.

As expected, the pairing had a lot of errors: https://pastebin.com/ftbdjb8E

After "pairing", i changed the id back to "iZBModule01" on my devices.js, restarted and everything seems to be like my first post: code unlock works ok; fingerprint unlock sends just attributeReport.

Here is the herdsman debug for the code unlock: https://pastebin.com/Jaefuzrn
And the herdman debug for fingerprint unlock: https://pastebin.com/eshtLKsH

Does it really seems that the problem is with my lock? Is it not sending the commandOperationEventNotification for fingerprint unlock as it was supposed to?
I'm absolutely sure that in version 1.7 it did. I even have some automations on Home Assistant for when it was unlocked with mine or my girlfriends fingerprint.

Koenkk, 2020-06-02:
I have no clue why it doesn't send it. 

To switch to 1.7.1 you can follow this guide: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html#bare-metal  but instead of `git checkout dev` do `git checkout tags/1.7.1`

BrunoZumba, 2020-06-03:
Ok, got it!!
After moving back to 1.7.1 the problem was still there (no commandOperationEventNotification for fingerprint unlock).

I decided to check if the batteries were OK. I checked their voltage and they were almost full.
Put then back and voila! Everything was back to normal.

The good and old: if you don't know whats happening, reset the damn thing!
(should have done it in first hand)
Now I get the commandOperationEventNotification messages.

Again, thank you so much for your time, sir!

closed 2020-06-03 by BrunoZumba