Koenkk/zigbee2mqtt #3620: Threshold based precision generates invalid MQTT autodiscovery template for Home Assistant

crowbarz, 2020-05-27:
# Bug Report
## What happened
After upgrading to zigbee2mqtt v1.13.1, sensors that were configured with threshold based precision (Koenkk/zigbee-herdsman-converters#1135)  became unavailable. This was due to invalid MQTT autodiscovery template code being generated, for example for attribute `illuminance_lux`:
```
  "value_template": "{{ value_json.illuminance_lux | float) | round(object Object) }}"
```

## What did you expect to happen
Previously, the additional precision rounding code in the template shown above was not injected into `value_template` when threshold based precision was configured under `device_options`. However it looks to me that https://github.com/Koenkk/zigbee2mqtt/commit/9da1a35bbde405e5b20a206e849c4edd63704855 changed this to add the precision rounding code when threshold based precision was configured under `device_options` as well.

The precision rounding code was originally introduced in the MQTT autodiscovery template in 
https://github.com/Koenkk/zigbee2mqtt/commit/f6c01a02c98191517afc1c1f3cfa51d5cc1271e7 and discussed briefly in Koenkk/zigbee2mqtt#787. The code that modifies `value_template` where `precision` is configured is [here](https://github.com/Koenkk/zigbee2mqtt/blame/master/lib/extension/homeassistant.js#L1901).

Should precision be applied when the converter initially receives the value via Zigbee, or should it be applied at the point that the data is consumed, such as the Home Assistant entity template. If the former, then rounding in the Home Assistant MQTT autodiscovery template seems superfluous. If the latter, then other systems that integrate with zigbee2mqtt via MQTT (eg. Node-RED) would not receive values with the configured precision applied.

I can work around this issue by overriding the generated `value_template` in `devices` for each of my light sensors. And a suggested quick fix is to check whether the precision is a number before modifying `value_template` to add the precision rounding code.

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: 20190619

Koenkk, 2020-05-28:
I don't see a reason for the precision rounding to be added to the `value_template`. Therefore I think the solution would be to remove it (and just leave it up to the converters).

Koenkk, 2020-05-29:
Closing this as PR is merged.

closed 2020-05-29 by Koenkk